Ralim/IronOS #194: Wrong GUI with disconnected tip

Mrkvozrout, 2018-01-09:
When I start with disconnected tip (or disconnect tip on idle screen), display blinks with TIP HOT warning and shows temperature of imutable 490°C.

- front button (soldering mode) - shows BAD TIP
- back button - shows idle screen with TIP DISCONNECTED

Ralim, 2018-01-13:
This is because of the settling time that is present on the temperature readings. 
You will notice the temperature slowly rises up to out of bounds when you have a tip removed.
It would require a re-structure to the code to work around this, and I haven't (yet) come up with a nice solution that doesn't delay the warning on a hot tip.

fishingpro, 2018-04-03:
> TIP HOT warning and shows temperature of imutable 490°C.

> I haven't (yet) come up with a nice solution

maybe it will do? The indication is always higher than the maximum allowable for both Fahrenheit and Celsius - and this is a good signal.

```
if ((systemSettings.temperatureInC && Temp > 480) || (systemSettings.temperatureInF && Temp > 900)){
	//BAD TIP
}else{
	//HOT TIP
}
```

closed 2018-04-27 by Ralim