Ralim/IronOS #199: Issues with heating

Chosaint, 2018-01-11:
When I press the button closest to the iron tip to heat up, the display shows for a second then it shuts off, restarts, and back at the main menu where I can press the two buttons for settings or heating. I have tried several different firmwares to no success. Anyone else have this issue?

Ralim, 2018-01-11:
Hi,
The first suspect when this occurs is the power supply that you are running the iron off, or a hardware problem. 

What power supply are you using?

Chosaint, 2018-01-11:
was just coming back to update everyone, it was the crappy power supply that came with the TS100, I used the one that I use for my laptop and its working perfectly, same voltage rating as the crap one they sent, 19v

closed 2018-01-13 by Ralim