Ralim/IronOS #200: Firmware flashing - no space

ksperr, 2018-01-13:
I'm tring to install the last firmware (spanish language) on my new ts100. 
When i try to copy de .hex file to the usb drive that appears when i connect the ts100 to the computer it tells me i have no enough space (free space: 5,5kb total space: 8kb needed: 110kb).
When i connect the ts100 to my computer (windows 10) the lcs screen shows "CONFIG"  when i press the button near to the screen it shows ver:2.12 and when i press the other button it shows a MIN logo (miniware logo). When i press the two buttons it comes back to CONFIG. I tried it on a ubuntu pc but its the same.

Rockman18, 2018-01-13:
Unplug your ts100. Press the button near the top and then plug the USB cable. It should show something like "DFU..." on the screen. Now you are in update mode and you can copy the hex file.

closed 2018-01-13 by ksperr

Scritz68, 2023-05-10:
somthing so simple ..had me searching for an hour ..thank you again!