Ralim/IronOS #201: Low threshold for gui_showTipTempWarning() is a little excessive

tbaumann, 2018-01-16:
The "warning tip hot" screen shows until the tip has reached < 30 celsius.
I think that is a little excessive.
This is below body temperature and on a hot day it may very well never cool down to < 30 because ambient temperatures are higher.

I would say <= 40 is a reasonable threshold. 40 Celsius is safe and not even unpleasant to touch.

According to this source even higher temperatures would make sense. Since nobody is going to hold the iron on the tip for a long time.
https://aavidgenie.com/maximum-touch-temperature/

Referring to code line
https://github.com/Ralim/ts100/blob/93869b979e6a498eb210ae44b0a4f26ee73f356e/workspace/TS100/src/main.cpp#L423

I can fork and create a pull request if you like. But since it is a one-liner I suspect this is easier.

Firebie, 2018-01-16:
Agree, in my country summer temperature is over 30 all the time.

tbaumann, 2018-01-16:
Aligning the warning with the blink threshold of 50 degrees doesn't even sound unreasonable. 50 degrees is like slightly hot tap water.

jonnieZG, 2018-01-16:
I don't want to hijack this topic and make it off-topic ;), but @tbaumann, 50°C is the highest temperature of an object that an average adult can hold with bare hands, without burning them.

JohnEdwa, 2018-01-17:
@jonnieZG @tbaumann You'll develop a burn in around six seconds if you hold on to an object that's at 51C:

>The minimum temperature that can cause a burn in a finite amount of time is 44 °C (111 °F). From 44° to 51 °C (111° to 124 °F), the rate of burn increases by a factor of approximately four with each Celsius degree risen or twice per Fahrenheit degree risen, from six hours down to six seconds. The burn would develop in less than a second if the exposure temperature is at least 70 °C (160 °F).
[- Wikipedia](https://en.wikipedia.org/wiki/Thermal_burn#Factors)

If we could go for two temperatures, over 70C should blink with the "TIP HOT" warning, and under that, display "Tip warm" until it hits 40C and goes back to idle.

tbaumann, 2018-01-17:
:+1: except that there are already two thresholds. One for blink and one for hot. Let's just change them to sensible values.
The blink threshold is right now 50 Celsius. I would say up that to 70 Celsius because that is where it hurts. Extra warning makes sense. (Only does that if the setting is set to on)
And cut the 'tip hot' message at 50 Celsius.

Nobody needs a warm warning IMHO.

closed 2018-04-11 by Ralim