Ralim/IronOS #202: Menu locks up - whole device is unresponsive after a couple of seconds.

edmanning555, 2018-01-18:
As above - device stops responding after a few seconds, after a while the screen flashes and it works again for couple of seconds. Rinse and repeat!

I've tried using different firmware with the TS100 and they work fine. Is there anything I can do to help diagnose this issue?

Ralim, 2018-01-18:
Hi,
Can you provide any further information to try and diagnose this?
Is there any particular menu prompt it happens on?
Does it happen when in the soldering menu?
Does it happen on the idle screen?

Can you try an few times and note when it happens?

Sadly I can't diagnose without more information :)

edmanning555, 2018-01-18:
To answer your questions:

It happens in any instance regardless of menu, buttons pressed, or whether it is connected via USB or DC power brick, or any other state as far as I can tell.

It happens every time, there is not a single instance where the device functions correctly for an extended period of time.

freeman0869, 2018-01-19:
So did the same thing for me last night, Bricked my TS100 after upgrading with .hex file...no response what ever you do, after you plug power or usb wire back in, it works for 2 seconds and no response after that for at least a minute...total freeze of menu. The upload went fine and I did get .RDY after updating but no response from TS100 any more !

Ralim, 2018-01-19:
Can you please specify which firmware files (language and version) that are having these issues?

In the mean time you can just use DFU mode to roll back to the previous version for now. 

If you use a language different to English, can you please try the English one as well to confirm if it is a language specific bug or general bug?

edmanning555, 2018-01-19:
Hi,

I am currently using the previous version which is fine. I personally use
the English version and haven't tried any other. Would you like me to?

freeman0869, 2018-01-20:
I used TS100_EN.hex latest version and the older version but nothing worked now I did try another file I found here and totaly lost it..now it says ("BL FLASH"....CONT?...CHK CRC...FC6227E8...Done!) and that is By pressing the button closest to the tip...now I can not see it when pluged to the computer on Win 10...even when pressing the buttons. Any clue ?

MathisDELOGE, 2018-01-22:
I just received my ts100 and I have the same problem. I tried with all versions from 2.0 (those before do not even boot).
In any menu, whan i plug a power supply, the screen freeze after about 2 seconds.

freeman0869, 2018-01-22:
I tried all the files here and nothing works the only ones that do work are the originals ! So if anybody having same problems flash either one and your TS100 will be back online !

[Version 2.12and 2.17.zip](https://github.com/Ralim/ts100/files/1653860/Version.2.12and.2.17.zip)

MathisDELOGE, 2018-01-22:
Thank you for the originals, I did not find them.
I just tried it, it works !

Ralim, 2018-01-22:
Can you please confirm where you have purchased your TS100 units from? And roughly when?

Are these all new units?

@freeman0869  you flashed the conversion firmware to replace your stock bootloader with the dfu-util model instead (Im assuming you took that firmware from the thread with the discussion on alternate bootloaders).

I have reconfirmed correct operation on all of the units I have access to and have not had any issues with the firmware locking up on these units.

What you are experiencing with the reset after a few seconds is the watchdog timing out.
The watchdog is there to catch errors where one of the threads in the scheduler locks up.

The most likely cause of this is an issue with the I2C communications locking up the peripheral if there are transient errors on the bus (as these have an exclusivity around them as its a shared bus).

Are your units new or is this error new? 

This is most likely yet another new manufacturing change that is causing issues. The original firmware does not actually use any of the hardware acceleration in the STM32 and instead bashes everything out using software which is why its performance can be sub-par in some instances.
This also means that because its software runs really slowly (comparatively) it is less likely to run into issues with the hardware.

freeman0869, 2018-01-22:
Thank you for the explanation of the  TS100...which I bought in December and received last week on the 16 of Jan.
 https://www.tmart.com/TS100-65W-Temperature-Adjustment-Built-in-STM32-Chip-Digital-OLED-Programmable-Soldering-Iron-Set-TS-I-Black_p341108.html...it was the cheapest combo for me.

 It is working with version 2.17 for now, and tried win 10, win 7, and linux Ubuntu to upload the same .hex file (no Apple) and figured it is the new TS100 models that I received that is the problem.

And I am pretty sure when I first received it last week it did say ver 2.18 on my Iron but, I'm yet to find that version anywhere !!!

MathisDELOGE, 2018-01-23:
For me it's the same, i bought the ts 100 un december and received last
week so it's à new unit..
(Thanks for the nice explanation ;) )

Ralim, 2018-01-23:
Yeah okay, this could be an issue with the newer units coming out.
I have also noticed a trend in some of the cheaper seller's units having more issues than some others.
Not sure why that is though :/

The firmware that came on your Iron is *not* this repository. This firmware has just eclipsed up to the numbers that the official firmware uses. [This](http://www.minidso.com/forum.php?mod=viewthread&tid=892&extra=page%3D1) is the location of the *official* firmware.

I'm not sure what is causing the lockups on your units to be honest, as hopefully the hardware is the same as the official and there are not clones yet.

When I next have time Ill try and get a better debug firmware up for you. But Its really odd that the newer models are having this issue.

Boyeen, 2018-01-25:
If it's of any use, I just purchased a TS-100 ("updated", brown, with 2 tips and power supply) from Banggood... It came with v2.12 MiniDSO firmware installed.

I updated to the latest firmware (v2.02) from this site with zero problems. The iron works correctly and the menus all work smoothly. Perhaps there are  "clones" being sold under the MiniDSO brand...   

BTW, awesome job Ralim. I LOVE the tiny font Settings descriptions, totally readable, super professional.  Way to go :-)  Thanks for your efforts.

electroplated, 2018-01-25:
I have this exact same issue - bought the iron ob ebay from the link (https://www.ebay.co.uk/itm/12-24V-Mini-65W-Programmable-TS100-Electric-Solder-Soldering-Iron-LCD-Digital-UK/253273635474?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) on the 17th Jan and it arrived today. Only the stock firmware works, every other version I've tried works fine for about 2-3 seconds after power is applied then the buttons become unresponsive until reboot. Iron also got so hot when I had the latest release installed the entire tip was glowing bright red!!

Ralim, 2018-01-26:
Hi,
Can you try this firmware on your units?
I have slowed down the I2C bus by 50%.
I'm guessing this is the most likely code to be locking up to cause the hanging issue.
[TS100.hex.zip](https://github.com/Ralim/ts100/files/1666308/TS100.hex.zip)

xPsIXx, 2018-01-26:
This might help a bit. I received a unit and the barrel plug connector broke first time i plugged it in. I claimed a second unit and when it arrived it would freeeze after a couple of seconds with Ralim's firmware, the first unit never did. I now had two units so i swapped the daughter board of the new TS100 with the old TS100 and it worked like a charm... Note the silkscreen quality.

It takes about 5 seconds to freeze.
Sorry Ralim, that firmware doesnt work either.
 
[image]
[image]

freeman0869, 2018-01-26:
You slowed down the software and it gives me 3 seconds more to surf the menu so 5 seconds in all until it freezes ! Sorry that software still doesn't work !

Ralim, 2018-01-26:
@xPsIXx thank you! That confirms my fears that there has been a design change to the hardware.

Any chance you can get some high resolution photos of the front and back of the board that doesnt work?
Also if you can get the part number on the same square chip if it's changed compared to the other one :)

I'm guessing there is a redesign and something has changed subtly. 

Sorry about the new firmware not working yet, didnt know at the time that there was a hardware change.

jonnieZG, 2018-01-26:
I would say from the photo that they just changed the layout of the components, so the control pins must have changed. We just have to figure which ones.

electroplated, 2018-01-26:
Here's a quick video showing the behaviour with the firmware you posted up to try

https://www.youtube.com/watch?v=vCduzDBHiVQ&feature=youtu.be

As you can see it not only locks up but also starts heating up without updating the temperature readout which can lead to burnt fingers or worse perhaps?!

Here's a shot of the chip inside
[image]

Let me know if there's any other info, photos or tests I can do to help :)

xPsIXx, 2018-01-27:
freeman0869 indeed has the same board rev. 
For completion sake, better quality pictures.

[image]
[image]
[image]

xbeefpv, 2018-02-02:
Hi Ralim! i have the same problem, the TS100 freezes. My product is this:
https://www.getfpv.com/ts100-digital-oled-programmable-interface-mini-soldering-iron.html

Did you have a solution? Thanks!

Ralim, 2018-02-02:
Hi,
It appears that either there is a second revision of the mainboard out there or someone has changed something in these units.

As not all units are affected, I'm not sure if it's an old/new stock issue or some other manufacturer out there.

What would be ideal, is if someone who has one of these units could use a multimeter with a fine point tip and trace out the connections from the sides of the STM32 to the bottom connector.

I'm also going to try and make an I2C probe firmware to check if the accelerometer has been changed in this version.

freeman0869, 2018-02-02:
I've been using the Iron a lot and the accelerometer does not seem to work I set my time out at 600 and when I pick it up after not using it in between soldering it seems to not wake up even if I shake it a lot...so I constantly have to repress the button close to the tip to restart the iron !

Ralim, 2018-02-02:
Interesting, which firmware are you running ?

xbeefpv, 2018-02-02:
im trying with many, 2.0 2.2
but only the original firmware works fine.
Strange, but untill the oldest firmware not working. thanks for your reply @Ralim

SquiglyD, 2018-02-02:
Same issues here, purchased last week from getfpv, when I first received the iron I noticed it had firmware v2.18 (cant seem to find this anywhere) after trying yours I reverted back to V2.17. Just tested to see if the ACC is working and it appears not, torch will only come out of sleep mode by pressing the front button, motion has no effect.

electroplated, 2018-02-03:
Ralim - I've just tried to trace out the connections between the STM32 and the 16 pin connector on the opposite side of the board. I managed the first 8 pins ok but the second set were almost impossible to check reliably with the tools I have to hand right now - I have fine tips on my multimeter but its; still very tricky without an extra pair of hands!

As best I can tell, the first 8 connectors are connected as follows, using the attached images for reference.

1 - 29
2 - 30
3 - 32
4 - 33 & 34?
5 - 33 & 34?
6 - 32
7 - 6
8 - 7

[image]
[image]

I'll try and double check these and do the other 8 tomorrow if I can find a better way of doing such small scale work without better magnification!

1upRacing, 2018-02-03:
Ralim - If you are unable to get this sorted out I can send you an iron that has the issue.  Just received an order of 5 that won't work with your firmware upgrade.  

You're welcome to keep it afterwards to work on future upgrades.  Just get in touch if you need.

alexanderkozhinov, 2018-02-08:
Ralim, same to me. New device from Ali with 2.18, DFU: 3.43
Freeze after 2 seconds > 2.17 > ACC doesn't work.

sascha108, 2018-02-08:
Hi Ralim,
same problem, ts100 from banggood with 2.17
pcb is same as freeman0869

Ralim, 2018-02-08:
@1upRacing Where in the world are you located?
I'm tempted to take you up on this offer, however, I would like to make up an I2C tester firmware first.

@sascha108 @alexanderkozhinov 
Can you confirm if the accelerometer function works for you in the original firmware?

1upRacing, 2018-02-09:
I'm in the US but do a lot of international shipping so it wouldn't be any inconvenience to send one of these out to you.

Ralim, 2018-02-09:
@1upRacing Would be wonderful if you could.
Just hit me up on ralim@ralimtek.com to discuss.

alexanderkozhinov, 2018-02-09:
@Ralim on 2.17 at the moment TS100 doesn't react on shake in sleep. I need to push button to wake it up.
I didn't tested if it will go to sleep even in use (in motion).

SilverSho0t, 2018-02-09:
They maybe removed the accelerometer on new ones.

Ralim, 2018-02-09:
I don't think so, there is still an ic on the board, but I'm wondering if the model has changed and I just need to write a new driver for it (hopefully).

They may have changed the pinout as well subtly so ill be checking it as well :)

Mostly just need time :(

sascha108, 2018-02-09:
Hi Ralim,
there are no function with accelerometer @V2.17, TS100 go's to sleep no matter its moved or not.

kormyen, 2018-02-13:
@Ralim I just got a new TS100 from [Ali](https://www.aliexpress.com/item/1pc-Free-Shipping-Brand-New-1-Set-Mini-Portable-65W-Programmable-TS100-Electric-Soldering-Iron-Digital/32689219813.html) with the same 2-3 second freeze with your hex. Works fine on stock 2.12 and 2.17. Happy to help or test anything if you explain steps. Cheers!

Ralim, 2018-02-13:
Hi,
This definitely looks like a new batch of the units coming out. 

@1upracing has offered to send an affected unit to me, so when I pick up that one I will aim to get this up and running asap after that.

Not sure on an exact date sadly, but will let you all know :)

Sorry for the delay :(

kormyen, 2018-02-13:
Sweet. Good luck! :D

undirezt, 2018-02-15:
Hello. I also ran into this problem. The differences are not only in the PCB, but also in the absence of the stub hole for the sting and the other side of the packaging.
But, I still have not figured out, is this change in the original design or began to produce a fake?

AcidBurns, 2018-02-15:
[image]
[image]
[image]
[image]
[image]
[image]
I also made some pictures, high res with my magnification camera. I hope it helps. Needless to say that I got the same problem :D

TomK, 2018-02-16:
Same issue here, using 2.02.
My brand new TS100 received a few days ago locks up after a few seconds. The same firmware works fine on an older TS100.

Ralim, 2018-02-18:
Hi,
As a quick check, could someone please run this firmware and tell me the two numbers it prints out on your non-working unit ?
[TS100.zip](https://github.com/Ralim/ts100/files/1734510/TS100.zip)

kormyen, 2018-02-18:
025 060

TomK, 2018-02-18:
same here - 025 060
[image]

Ralim, 2018-02-18:
Thank you both.
This confirms that the accelerometer in your units is definitely a totally different unit.
Can you also try this firmware please :) (Sorry about this)
This tries to read the device ID register from the 3 most common accelerometers with that device ID.

Hopefully one of these matches up, if not I will definitely have to wait for the unit in the post to me to arrive. If one lines up Ill try and start working on a library for the new accelerometer and test once the unit on the way to me arrives.

Sorry about this :/
[TS100.zip](https://github.com/Ralim/ts100/files/1734558/TS100.zip)

TomK, 2018-02-18:
It's not a problem, your work is much appreciated!
000-013-015.  :)

[image]

kormyen, 2018-02-18:
@Ralim 
Happy to help. No worries at all to test further.
Same here with the 2nd zip.
000-013-015.

[image]

AcidBurns, 2018-02-18:
for me it's also 000-013-015.
those chinese guys, simply changing hardware without notice 👎

Ralim, 2018-02-18:
Bummer, those dont match any of the common ones :(
I'll hopefully be back soon with more information when the next one turns up (Or if i find more likely candidates to test)

xavkno, 2018-02-18:
Wouldn't it be easier to change out the component for the one on the previous hardware version?

badm1n, 2018-02-18:
2nd zip. Bought from Banggood on 2018-01-29.
[image]

Ralim, 2018-02-25:
Hi,
Sorry for the delay in response. I have been spending the last two days trying to figure out what make that damn IC is and haven't gotten very far. I have also reached out to miniware to see if they will help.

In the meantime, the below firmware will work but it disables sleep mode as it just doesnt talk to the accelerometer. This will still provide all the rest of the firmware for you hopefully until we can identify this ic.

[TS100.zip](https://github.com/Ralim/ts100/files/1755701/TS100.zip)

bubblefish12, 2018-02-25:
Yes, it works

sascha108, 2018-02-25:
It works fine - in original V2.17 its always go sleep after max sleeptime :(
I hope that you can identify the chip soon.

xbeefpv, 2018-02-25:
Works good! thanks Ralim! nice Work! i will be awaiting for the the complete functions

electroplated, 2018-02-25:
Also working fine for me - amazing job! 

Can anyone else confirm if it's normal at any temperature for the tip of the iron to glow bright red?!
With the older non compatible firmware loaded my iron got that hot which I've never seen an iron do before, just repeating that aspect here for anyone who might inadvertantly try the older firmware on one of these irons.

[30 more replies]

PR #216 (2018-03-03): LIS2DH - Adding support for the new accelerometer
+ Creates a new driver for the LIS2DH accelerometer
+ Fixes timing issues since we're already touching a chunk of code

The LIS2DH driver should output similar numbers to the old MMA accelerometer.

Fixes #202
Fixes #189
commit aef8f5f5 (2018-03-03): LIS2DH - Adding support for the new accelerometer (#216)

+ Creates a new driver for the LIS2DH accelerometer
+ Fixes timing issues since we're already touching a chunk of code

The LIS2DH driver should output similar numbers to the old MMA accelerometer.

Fixes #202
Fixes #189
closed 2018-03-03 by Ralim