Ralim/IronOS #203: Can't update FW from Windows on new TS100

apazga, 2018-01-21:
Hi,

DFU 3.43, just received it from Banggood.

Tried to update it using v2.01 & v2.02, by dragging file using Explorer and also tried from PowerShell copying it manually. Result: ts100.err file.

Any suggestion?
Thanks in advance.

Ralim, 2018-01-21:
Hi,
Make sure the file is downloaded to you computer first (not in an archive or first from Web browser).
Secondly when the first copy fails copy the file again. 

I prefer to just download the hex to the downloads folder and then just click and drag it in Explorer. 

Works fine for me on Windows 10 pro :/

apazga, 2018-01-21:
Hi @Ralim 
Thanks for your reply!
Just re-tried again, ensured that the hex was right, but same happened.
Anyway, **I found the solution**... and it was... easier... I just MOVED (not copied) the file and it worked.

I repeated the process to confirm that strange thing so... I deleted it again, and retried, and I really don't know why, but file permissions and that dialog only appear if you copy the file, and TS100 didn't like it.

Thanks anyway and hope it helps if someone has this same weird issue.

closed 2018-01-21 by apazga

mazdak18, 2019-12-20:
I am having the same issue here. I am using STM32CubeIDE to rebuild the hex code from the source code. The hex code is generated in the Debug folder. But when I try to drag-and-drop the hex file, I get the same message. I was wondering if I am doing something wrong with building the hex code as the generated TS100.hex is 107KB while the one in the latest release is 93KB.

Ralim, 2019-12-21:
If the file you are generating is that large, it may be too big and thus why its being rejected.
What compiler settings are you using when you compile? Do you have optimizations turned on ?

mazdak18, 2019-12-21:
I am using the default setting after installation. Are they any parameters/settings that needs to be changed?

whitehoose, 2019-12-21:
They all do (pretty much) what they say on the tin - what you need to alter depends on what you need it to do.
I seem to remember on the ts80 there were (unsubstantiated as far as I know) rumours of tip meltdown due to some setting.  Without sobering up I can't remember the details - and I got the impression that whatever meltdown occurred (if it did) was down to a non related faulty tip.
With all the banging on I've been doing about temperature - I might be tempted to limit max temperatures in turbo - but I've never bothered or used it in anger.  I'm a ts80 user and I'm just pathetically grateful if my power source is compatible with the iron.  I like the detailed view - It's not something that influences me greatly - but its interesting to see the effect of a big groundplane or aluminium backing on tip temp.

mazdak18, 2019-12-21:
Thanks @whitehoose. I just trying to get started on playing around with the source code. I did an extensive search to find a step by step instruction to produce the hex file from source code but I have not been able to locate any.  Would appreciate any guidance in that regard.

whitehoose, 2019-12-21:
I maybe mis-answered then - only saw your post in mail, didn't read the whole thread so missed the context - I assumed you were just starting out as a user, was talking about the finished article not the cooking arrangements.
Not dabbled with the dark arts for many moons I wouldn't resume to offer advice