Ralim/IronOS #208: mono app on Mac does not work

myhd, 2018-02-16:
Great project! I have one minor issue though. The logo editor page on the wiki says "the logo editor should run under mono on Mac, similar to Linux". I am on a Mac, and this happens:

I have downloaded the 370+MB Mac [installer](http://www.mono-project.com/download/stable/) for mono (MonoFramework-MDK-5.8.0.108.macos10.xamarin.universal.pkg), installed, and get a long error output when doing `mono TS100.Logo.Editor.exe`.

What is this, and how can I simply edit a bitmap file for a soldering iron? Any help is greatly appreciated. Thank you very much.

```
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all
Fontconfig warning: ignoring utf-8: not a valid region tag
Fontconfig warning: ignoring utf-8: not a valid region tag
Fontconfig warning: ignoring utf-8: not a valid region tag
Fontconfig warning: ignoring utf-8: not a valid region tag
Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) System.Windows.Forms.XplatUICarbon.CGDisplayBounds (intptr) [0x00002] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.XplatUICarbon.get_WorkingArea () [0x00005] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.XplatUICarbon.get_VirtualScreen () [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.XplatUI.get_VirtualScreen () [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Screen..cctor () [0x00034] in <1141838629e74c21a3b96d0a06337bad>:0
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) [0x0001e] in <ede5cf6930dc4b9a8366f15f6c89bbd0>:0
  at <unknown> <0xffffffff>
  at System.Windows.Forms.Form.get_CreateParams () [0x00157] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.UpdateBounds (int,int,int,int) [0x0000b] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.SetBoundsCoreInternal (int,int,int,int,System.Windows.Forms.BoundsSpecified) [0x00110] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.SetBoundsCore (int,int,int,int,System.Windows.Forms.BoundsSpecified) [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Form.SetBoundsCore (int,int,int,int,System.Windows.Forms.BoundsSpecified) [0x00074] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.SetBoundsInternal (int,int,int,int,System.Windows.Forms.BoundsSpecified) [0x0008a] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.SetBounds (int,int,int,int,System.Windows.Forms.BoundsSpecified) [0x00038] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.ScaleControl (System.Drawing.SizeF,System.Windows.Forms.BoundsSpecified) [0x0002c] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.ScrollableControl.ScaleControl (System.Drawing.SizeF,System.Windows.Forms.BoundsSpecified) [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Form.ScaleControl (System.Drawing.SizeF,System.Windows.Forms.BoundsSpecified) [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.Scale (System.Drawing.SizeF) [0x00033] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.ContainerControl.PerformAutoScale (bool) [0x00073] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.ContainerControl.PerformAutoScale () [0x00000] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.ContainerControl.PerformDelayedAutoScale () [0x00008] in <1141838629e74c21a3b96d0a06337bad>:0
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.ContainerControl.PerformDelayedAutoScale () [0x00031] in <1141838629e74c21a3b96d0a06337bad>:0
  at System.Windows.Forms.Control.ResumeLayout (bool) [0x00027] in <1141838629e74c21a3b96d0a06337bad>:0
  at TS100_Logo_Editor.Form1.InitializeComponent () [0x007e4] in <a83ec818c1c344aa9f09db88312645fc>:0
  at TS100_Logo_Editor.Form1..ctor () [0x00006] in <a83ec818c1c344aa9f09db88312645fc>:0
  at (wrapper remoting-invoke-with-check) TS100_Logo_Editor.Form1..ctor () [0x00018] in <a83ec818c1c344aa9f09db88312645fc>:0
  at TS100_Logo_Editor.Program.Main () [0x0000b] in <a83ec818c1c344aa9f09db88312645fc>:0
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) [0x0004c] in <ede5cf6930dc4b9a8366f15f6c89bbd0>:0

Native stacktrace:

	0   mono                                0x00000001046e8c01 mono_handle_native_crash + 257
	1   mono                                0x0000000104753236 altstack_handle_and_restore + 70
	2   SkyLight                            0x00007fff4cc8bfa8 SLDisplayBounds + 83
	3   ???                                 0x000000010b14ea1e 0x0 + 4480887326
	4   ???                                 0x0000000104bc2396 0x0 + 4374406038
	5   mono                                0x000000010463f4c7 mono_jit_runtime_invoke + 1383
	6   mono                                0x00000001047ffed4 do_runtime_invoke + 84
	7   mono                                0x00000001047fc024 mono_runtime_class_init_full + 996
	8   mono                                0x00000001046387ce mono_jit_compile_method_inner + 2478
	9   mono                                0x000000010463b683 mono_jit_compile_method_with_opt + 1363
	10  mono                                0x00000001046eb0c1 common_call_trampoline + 1217
	11  mono                                0x00000001046eabbc mono_magic_trampoline + 60
	12  ???                                 0x0000000104afa39e 0x0 + 4373586846
	13  ???                                 0x000000010ae4351b 0x0 + 4477695259
	14  ???                                 0x000000010ae4feaf 0x0 + 4477746863
	15  ???                                 0x000000010ae4fcb3 0x0 + 4477746355
	16  ???                                 0x000000010ae4f8be 0x0 + 4477745342

Debug info from gdb:

(lldb) command source -s 0 '/tmp/mono-gdb-commands.EQc13X'
Executing commands in '/tmp/mono-gdb-commands.EQc13X'.
(lldb) process attach --pid 27485
warning: (x86_64) /Library/Frameworks/Mono.framework/Versions/5.8.0/lib/mono/4.5/mscorlib.dll.dylib empty dSYM file detected, dSYM was created with an executable with no debug info.
[... 59 lines cut ...]
    frame #7: 0x00007fff52d6756d libsystem_pthread.dylib`_pthread_start + 377
    frame #8: 0x00007fff52d66c5d libsystem_pthread.dylib`thread_start + 13
  thread #4
    frame #0: 0x00007fff52c2c562 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff52d6706a libsystem_pthread.dylib`_pthread_wqthread + 1035
    frame #2: 0x00007fff52d66c4d libsystem_pthread.dylib`start_wqthread + 13
  thread #5
    frame #0: 0x00007fff52c2c562 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff52d6726f libsystem_pthread.dylib`_pthread_wqthread + 1552
    frame #2: 0x00007fff52d66c4d libsystem_pthread.dylib`start_wqthread + 13
  thread #6
    frame #0: 0x00007fff52d66c40 libsystem_pthread.dylib`start_wqthread
(lldb) detach

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Process 27485 detached
(lldb) quit
Abort trap: 6

```

Ralim, 2018-02-23:
Honestly you might be the first one to try this.
Looks like Mono just doesnt support this.
Does the python script work for you?

Also happy to convert any images you want if you post them here or email them to me.

myhd, 2018-02-23:
### Converting with mono graphical interface application on Mac / High Sierra

I found a solution: Starting mono on MacOS High Sierra in 32 bit mode. Like this:  `mono --arch=32 TS100.Logo.Editor.exe`

### Converting with python script

Ow! I did not realize / find the python script until your posting. The mono .exe is in the release files only, so that's where I looked. But I have Python 2.7.10 and had to install `pil`(or, Pillow in this case) for this:

```
sudo pip install Pillow 
./img2ts100.py infile.png outfile.hex
```
This indeed did work. Thank you for the offer.

Ralim, 2018-03-03:
Hi, thank you for the investigation.
Any chance you feel like adding your notes to the Wiki (anyone can edit it basically).
That way the next person to come across knows whats up?

If not I can add notes, but easiest if you add them since you know how you would have liked for it to be written :)