Ralim/IronOS #211: Power indicator missing

DragonHunter274, 2018-02-24:
after the update the power indicator was missing

Ralim, 2018-02-26:
Hi,
Can you please provide more details?

Ralim, 2018-03-03:
Can you please clarify if this is an issue with the battery indicator not showing up for you?
Or is there an issue with the power indication iron ?

DragonHunter274, 2018-03-08:
it is the power indocator that shows the power the iron is using in a bar like in previous versions instead of the chevrons

Ralim, 2018-03-10:
Hi,
The power bar from 1.x was removed in the 2.x series as it was poorly calibrated and didn't accurately reflect the power level going to the tip.

In the newest 2 alpha you can enable detailed soldering screens that will show the actual duty cycle that the tip pwm is implementing if you want.

Adding a power bar is a potential option, but it has been 6 months since it was removed with no one complaining and no one asking for any indication either, so it's probably not a massively asked for item?

JohnEdwa, 2018-04-13:
@Ralim 

One way would be to split the 'whisps of smoke' into a bar graph:

Vertical : [image]

Complex/segmented: [image]

While the segmented could have as many as 24 states if it's pixel by pixel, or 7 if it's divided by 'smoke segments' and is easier to read accurately, I personally feel the vertical to be better looking and much easier to implement as well - either clip the graphic, or just draw a black box on top of it.

Add in *tiny* bit of averaging, and you have power indicator to match the much-improved temperature you added in 2.04.

Ashdinalt, 2018-04-13:
I don't think one really needs a power bar, but to have some animation to this icon would be nice

Ralim, 2018-04-14:
@JohnEdwa I prefer the first option was it maps easier to 0-100% range for the PWM output.
Could you create a set of 12x16 (12 pixels wide) graphics for these ?

JohnEdwa, 2018-04-15:
@Ralim 
Sure, but you could save some flash space if you implemented this by drawing a black box on top of the full graphic that hides the part of the smoke whisps that aren't needed.

Combined:
[image]
[image]

Separate files:
[image] [image] [image] [image] [image] [image] [image] [image] [image]
[image] [image] [image] [image] [image] [image] [image] [image] [image]

Ralim, 2018-04-27:
Confirming I did implement this with drawing over in the end :)

commit 42db57da (2018-04-27): Update power indicator

Fixes and closes #211
Final 2.04 release coming up :)
closed 2018-04-27 by Ralim