Ralim/IronOS #213: Adding an option to swap the buttons for increase & decrease temperature.

JohnEdwa, 2018-02-27:
I've been using this iron for a few months now, and no matter how much I use it I *always* change the temperature the wrong direction. To my mind, 'Start Soldering', 'Boost' and 'Increase temperature' should be the same button, but right now that button decreases the temperature which throws me off completely.

Would it be possible to swap them, or add an option for it?
I'd also suggest swapping the UI from "< 350C >" to "+ 350C -" or "- 350C +" depending on which way the buttons are set up as.

Ralim, 2018-02-27:
Hi,
The buttons do indeed swap depending on the orientation of the screen, I can happily change to the "+/-" instead of "<>".

The buttons are designed to be logical from the view of the temperature on the screen, which is slightly annoying as it means for right handed folks that decrease temperature is on the front button.
However, this makes more sense the hard coding the button positions and having it only be logical for left handed folks. (An aside here, if you go back to *really* early in this project, I originally had the buttons fixed in function and people complained)

I can change to the "+/-" in the next release and see if that helps you ?

JohnEdwa, 2018-02-27:
Ah, I see where you are coming from.  
You look at it as being "Left/Right" which changes depending on the orientation of the iron, while I see it as being "Up/Down" where up is always towards the tip.
Adding the +/- will most likely help me a lot already though, so let's start with that.
 
If you ever get so many settings implemented you are forced to add sub-menus to keep it sensible, you can add it in there then :)

Ralim, 2018-02-27:
Yeah, to some its logical and to some its highly illogical 😀 
Ill add the "+/-" to the next build :)

Heh, I'm really trying to avoid needing to have more menu options :/

JohnEdwa, 2018-02-28:
> Heh, I'm really trying to avoid needing to have more menu options :/

They say pictures tell a thousand words, so you'll get my stand on this by taking a look at [the config page for the Pebble watchface I developed.](https://user-images.githubusercontent.com/5417183/36452150-0970ea22-169c-11e8-87d6-d36125b380ff.png) (There can never be too much customization, as every user is unique)

As you've already hit the spot where you need to really think if a setting is 'worth' adding, splitting them to sub-menus would make sense, to me at least.
Quick grouping I came up with:

Power (or just as a single setting like it's now)
-Power Source
-Auto Start (Possibly under 'Advanced'?)
	
Motion & Sleep 
-Sleep Time
-Sleep Timeout
-Shutdown Timeout
-Motion Sensitivity
	
Soldering (Or 'Boost Settings', 'Boost Mode', 'Boost')
-Boost Mode Enabled
-Boost Temperature
	
User Interface
-Temperature Unit
-Display Orientation
-Button Orientation <- This suggestion
-Cooldown Blink
-Detailed Idle Screen
-Detailed Solder Screen
-Settings Style <- For people who want them all in one line, no matter how many there are

Calibration (or 'Advanced')
-Calibrate Temperature
-Calibrate Input Voltage
-Factory Reset

Ralim, 2018-02-28:
That's a nice UI 😄 
I wish I had a smartwatch like the pebble :/

The big issue with going for multiple menus is that people get lost because of the tiny screen, and with multiple languages its a pain to have enough room for everything to fit neatly.
(Though I'm totally open to ideas from people!!)

I have been tossing up going to a two step menu, with common ones showing up and just hiding others under an advanced submenu? Simpler for people to use (less going in and out).

Thinking it could just be the last entry ("Advanced")
And when you select it, it just shows more options before exiting, or if you dont then you just exit the menu on the next press.

Open to ideas though?

closed 2018-03-15 by Ralim