Ralim/IronOS #214: Discussion about redesigning the settings menu.

JohnEdwa, 2018-02-28:
Continuing the discussion from #213 in its own issue: grouping settings items into submenus to allow for more options without having to scroll through them all.

----

> The big issue with going for multiple menus is that people get lost because of the tiny screen, and with multiple languages its a pain to have enough room for everything to fit neatly.  
-Ralim

So, here's a mockup of my idea: (it loops once to show the long-press exit) 

[image]

Mark menu headers with a submenu icon, and when you are in a sub-menu, add a single line on the edge of the screen (could be a 'scroll bar' to show how many items there are, if one wants to be fancy, and the main menu could have one at the top, if there is space for it)
Once you scroll the sub menu items once, it kicks you back into the main menu to continue to the next item, so you don't need to specifically exit the sub menu. (or it could have a 'Go back' as the first item)
Long pressing the 'next' button would kick you out of the menu instead of scrolling to the next one , and doing so in the 'main' menu will bring you back to the idle screen. This removes the hold-scrolling though, but would be a more useful feature imho.

Quick grouping I came up with:

**Power** (or just as a single setting like it's now)
-Power Source
-Auto Start (Possibly under 'Advanced'?)

**Motion & Sleep**
-Sleep Time
-Sleep Timeout
-Shutdown Timeout
-Motion Sensitivity

**Soldering** (Or 'Boost Settings', 'Boost Mode', 'Boost')
-Boost Mode Enabled
-Boost Temperature

**User Interface**
-Temperature Unit
-Display Orientation
-Button Orientation <- issue #213
-Cooldown Blink
-Detailed Idle Screen
-Detailed Solder Screen
-Settings Style <- For people who want them all in one line, no matter how many there are

**Calibration** (or 'Advanced')
-Calibrate Temperature
-Calibrate Input Voltage
-Factory Reset

---

Power Source could be it's own item instead of being in a sub menu (with Auto Start being in an advanced submenu), as it seems like the kind of a setting people would want to change often and quickly.

Ralim, 2018-03-03:
I really like this suggestion.
It actually should be doable to implement without too many changes to the current code base, with just a few helper functions to be made up.

I would prefer something like this : 

**Power Source**

**Motion & Sleep**
-Sleep Time
-Sleep Timeout
-Shutdown Timeout
-Motion Sensitivity

**Soldering** (Or 'Boost Settings', 'Boost Mode', 'Boost')
-Boost Mode Enabled
-Boost Temperature
-Auto Start

**User Interface**
-Temperature Unit
-Display Orientation
-Button Orientation <- issue #213
-Cooldown Blink
-Detailed Idle Screen
-Detailed Solder Screen
-Settings Style <- For people who want them all in one line, no matter how many there are _(This may cause a chunk of duplicated code, so might only implement if needed)_

**Advanced**
-Calibrate Temperature
-Calibrate Input Voltage
-Factory Reset

---

Mostly taking power source out of a sub-menu, since I think 2 options is pointless (basically works out more clicks than no menu). And i prefer the idea of moving the auto start into the soldering menu.

Otherwise I like this layout, will cause translations to have a fair bit of catching up to do though 😄 

I like your preview image, can you send me a bitmap of your menu icon you used ? (if it fits I'll go with it)

JohnEdwa, 2018-03-04:
Here you go: [image] 
It's my artistic masterpiece that I carefully crafted in roughly 15 seconds.

Like I said myself, having the Power as a singular setting makes sense as well, and the Settings Style was kinda just an idea if it's easy to implement. The submenu system, if the controls are implemented smoothly enough, should be superior to the single list style for pretty much everyone.

Ralim, 2018-03-07:
It work work as a good graphic until someone pipes up with a better one :)

Yeah I agree it would be much nicer to use than a single menu. I'm currently working on rough ideas for the implementation of this now :) (As I get time :| )

jonnieZG, 2018-03-07:
Guys, what do you think of this structure?

Power and Economy
- Power Source
- Auto Start
- Sleep Temperature
- Sleep Timeout
- Motion Sensitivity
- Auto Start

Display and Units
- Temperature Unit
- Display Orientation
- Detailed Idle Screen
- Detailed Solder Screen
- Cooldown Blink
- Scrolling Speed

Soldering and Controls
- Boost Mode
- Boost Temperature
- Button Orientation

Advanced and Reset
- Settings Style <flat/grouped>
- Calibrate Temperature
- Calibrate Input Voltage
- Factory Reset

federck, 2018-03-08:
My proposal follows Ralim's one, but with shorter sub-menus name in perpective of translatisons and specific icon per item.
Here it is:

**Power Source** [image] (or even noting)

**Soldering** [image]
-Boost mode enabled
-Boost temperature
-Auto start

**Sleep** [image]
-Sleep time
-Sleep timeout
-Shutdown timeout
-Motion sensitivity

**Calibration** [image] 
-Adjust scrolling speed
-Calibrate temperature
-Calibrate input voltage

**User Interface** [image]
-Temperature unit
-Display orientation
-Button orientation 
-Cooldown blink

**Advanced** [image]
-Settings style
-Detailed idle screen
-Detailed solder screen
-Factory reset

Also animated icons would be a great thing if not in conflict with memory space.

Ralim, 2018-03-11:
Hi,
I have pushed up the branch of my implementation of the layered menu (https://github.com/Ralim/ts100/tree/LayeredMenu).

This will require all translations to be updated, but curious to hear your thoughts before I merge this into mainline?

Here is a build (English)
[TS100.zip](https://github.com/Ralim/ts100/files/1800198/TS100.zip)

Maartenwut, 2018-03-11:
So much better than it has ever been, it's perfect!

federck, 2018-03-11:
Hi, I've observed that:
-Icons results somewhat distorted to me, should I redesign them taking into consideration something in particular?
-Descriptions of menu items seems to be redundant, would be better to take them off?
-Unluckily; long descriptions in sub-menu items interrupts before the end of the phrase, whereas, shorter descriptions take _an eternity_ to reappear after running off.
Nevertheless, work carried out is very very good!

JohnEdwa, 2018-03-11:
> -Icons results somewhat distorted to me, should I redesign them taking into consideration something in particular?

They've been converted from 17x16 pixels to 16x16 by skipping the middle column of pixels, instead of the very first blank one for some reason:
[image] -> [image]

---

>-Unluckily; long descriptions in sub-menu items interrupts before the end of the phrase, whereas, shorter descriptions take an eternity to reappear after running off.

Adding examples of these: 'Sleep modes' menu desciption seems to be 
"Power Saving Settings&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;", while Motion Sensitivity is just "Motion Sensitivity <0.0 Off 1."

---

Would there be space for the single pixel line indicating the it's a sub menu item? It would help recognizing submenu items from the headers and I think it would help.

--- 

The 'Hold to scroll' timer doesn't reset when you get back to the idle screen, so you immediately go to the firmware version info screen.

federck, 2018-03-11:
Here resized logos: 
[image] [image] [image] [image] [image]

Ralim, 2018-03-12:
@JohnEdwa 

Indeed the icons were shrunk to 16x16, I just used software I had locally to perform the conversion, will update these shortly. I wasnt concerned massively over these as I figured they would probably be changed soon after I make a proper release as everyone seems to have their own ideas on icons :)

I'll look into the descriptions now, This code was recently re-written by others so I havent looked at it recently, will look into it next though 😄 

Yep I can add a single pixel line, however it was thinking the icons work to show its a menu ?

Oh derp, I normally use two button exit, ill touch that up now.

@federck Thanks for new icons :)

commit b0b51203 (2018-05-11): Fix up menu icons

Sorry this got lost :angry:
Fixed now :)
Fixes #214 finally.
closed 2018-05-11 by Ralim