Ralim/IronOS #215: Question: When there will be a new release?

jonnieZG, 2018-03-01:
I am well aware of the issues with new hardware revision that is incompatible with the motion sensor, but when do you plan to make a new release? At least for the original hardware revision?

It's been 2 months since v2.02...

Ralim, 2018-03-01:
Currently working on the new hardware support as my focus :/
I was away on holidays for part of the last period of time as well.

It will be the same firmware for both hardware versions etc, they will all support the same stuff, work the same etc. (Not splitting the code base).
There are currently 5 issues that I would like to fix in 2.02 that I just havent had time to really look into properly, and the new hardware chewed up a chunk of time too.

So I figured it would be worth fixing those issues so I have some content to push in 2.03 :)

My rough plan is to try and get the new hardware tested and finalized as soon as possible, and then push a 2.03 that supports that hardware fully and closes those 5 issues as soon as I can following that.
In a perfect world that might be this weekend, in reality that will naturally move as life gets in the way.

Ralim, 2018-03-03:
Alpha's been pushed out.
Slowly working on the other issues as I test

JohnEdwa, 2018-03-04:
@Ralim 
I'll add this here as it *kinda* relates, but wouldn't it make sense to add the version numbering to the .hex files as well to help keep track of the files?
I would tag this one as T100_203_RC1_EN.HEX as I personally like to use Release Candidate instead of Alpha (as it implies it's supposed to be ready, but maybe isn't yet), but something like T100_v203_A1_EN.HEX could work too.

Ralim, 2018-03-07:
@JohnEdwa 
I'm open to changing the file names, however some TS100+OS combinations don't like it when you have a filename that's too long for 8.3 format when you write to the TS100 DFU bootloader, which is why I was (trying) to keep them shorter?

closed 2018-04-09 by Ralim