Ralim/IronOS #222: A single Multi-Lingual firmware?

jonnieZG, 2018-03-07:
Since we are already talking about broadening the setup menu, should we consider making a multi-lingual firmware, where the default language would be English, and a user could choose his favorite language from the Setup?

That would make distribution of compiled firmware much easier, since there would be just a single HEX file, instead one HEX for each language.

Ralim, 2018-03-07:
The big question for this will mostly be flash space, if after the new menu there is enough space to build all the languages in and we split up the font table to account for the overlapping Russian and Latin (I think?) sections then I'm happy to make it multi lingual. Another thing is that we still don't even have Korean and Chinese or japanese in the firmware yet either and they all need more font space.

We could look into build time calculating the font table and stripping unused characters as well, but I'm not that familiar with the best way to do that with c++? (all ears if you know)

jonnieZG, 2018-03-07:
If the problem is the available memory for all the fonts, we could have several groups of firmwares (all of them would have a common English option):

* One for Latin-script languages
* One for Cyrillic languages
* One for Chinese, Japanese and Korean.

The other option might be, as you suggested, making a pre-compile script that would cleanup the Font.c from the unnecessary characters, but I haven't yet considered all the pitfalls of such a solution...

jonnieZG, 2018-03-08:
Here's my rough idea for making all of the languages supported in the same FW version:

1. Instead of using a fixed pointer that is defined by #defined language, each pointer would be an index of pointers, with a language ID as a key. 
2. Instead of the current Translation.c we will have a RawTranslation.c with all of the language strings loaded in the same time.
3. All the strings that appear for printing will HAVE to be externalized in Translation.c file.
4. Instead of Font.c we will have a RawFont.c with both the big and the small font defined, with *all* the characters from *all* the languages that are supported. That would be a *huge* file, but it won't be compiled entirely, because...
5. There will be a pre-compile script that will read and modify the RawTranslation.c and RawFont.c and automatically generate Translation.c and Font.c files from them, in the following way:
* It will first scan all the characters that appear in RawTranslation.c strings
* For each character it will lookup in a newly generated Font.c if that particular character was already used.
* For a new (unused) character, it will create two new Font.c entries taken from RawFont.c (one for small and one for big bitmap) with a current index. Character's indexes will be assigned in order in order of character's appearances, e.g. if the first character that appeared in the translation strings was 'T', then the character 'T' would have index 0, and all the appearances of 'T' in the new Translation.c file would be replaced by uint16_t value of 0. If the second char to appear is 'h', the same will be done, but the 'h' would now be indexed as 1. Consequently, the Translation.c would result in a human-unreadable file, but that is not important, because we will be able to read them on the iron screen, because...
6. The print routine will be modified to display bitmaps according to the indexes from the automatically Translation.c and Font.c, instead of the old UTF-8 encoding.

Basically, that new script of ours will be changing UTF-8 encoding into a dynamic 16-bit encoding of characters, by their order of appearance.

Good?

jonnieZG, 2018-03-12:
I have some further improvements to the above idea, and I am willing to make them myself one of these days.

Instead of the RawTranslation.c we will have one **translation_*XX*.properties** for each language (XX being the language code). It will be a file with set key-value pairs, key being a code of each string. These files would be edited from a JavaScript application, and after saving, the application would make the final Translation.c and Font.h files.

Ralim, 2018-03-13:
Hi,
We have about 15kb to play with for font data on the unit to leave room for the rest of the firmware etc, as we have 48k after the DFU bootloader, then less the program size, plus the 2kb consumed by the settings and boot up screen images (as these are page aligned on their own pages).

I'm all for having a binary searchable lookup section for the font data. 

Also any scripts that need to be run to generate these files need to be callable by make on Linux, so probably JavaScript is out as a language for doing the processing in at this point on time. Cross platform support should be a priority so that users who are using the system workbench for stm32 can still use the files that are generated.

Otherwise i agree with your concept. Would make for a nice big clunky pre processor but otherwise would be nice.

I was thinking of a smallish c program that could be compiled in the first stage before the main program but need to check for integration with the ide as well. 

Don't really want to check in the processed files as they may change randomly and bloat the repository..

jonnieZG, 2018-03-13:
As much as for the strings editor goes, I thought of making a user-friendly(sh) GUI in HTML5 that would produce the "source" property or Json files for the fonts and the text strings. The pre-processor can be written in C++ so it could be called by make, to make the target C and H files.

15kB is a bit less than I expected, but we don't have the Korean, Chinese and Japanese yet, so it should be fine for the languages we have by now. We also need to count in the space taken by the strings, so I am thinking of compressing them in a some way. E.g. having encoded characters from the range 0x00-0x7F taking just one byte, while those starting with 0x8--- taking two bytes (similar to the UTF-8 principle. 

If hit the 15kB limit when the Asian languages kick in, I can work out a font compression algorithm as well, but all in good time. BTW, **do we have any volunteers (preferably native speakers) for translating the messages to Japanese, Chinese and Korean**? 

I think this whole thing is worth the effort, because having a single multi-lingual firmware instead of having a different version for each language, adds a substantial value to the device.

Ralim, 2018-03-13:
Ah, right I get what you mean now.
That is certainly a possibility, I'm currently in toying with seeing if I can make the c++ evaluation at compile time work to building the data at compile time so they can be constants in the firmware. Not sure if this will work, need to toy with it.

At this point, should we just stick with utf-8 encoding and then optimise the encoding later on when we run out of space. 

I'm not massively concerned with space just yet, since we already encode the entire font table and we still have a couple of k of flash space left. 

I'm thinking a plan of attack of :

1. Add a language menu and change to compiling all the current languages into the firmware
2. Optimise the font table at compile time to reduce firmware size
3. Add in Japanese, Chinese and Korean.

I think the first stage might *just* fit, and would let us proof of concept the idea.

I think if we are only compiling in the needed fonts, and we just use a b-tree to sort from utf8 to font index we should be okay for room.

I agree it's definitely worth the effort!

I'm sure if we convert over the fonts and post for interest people will show up, people seem to keep showing up around here that love the firmware so I'm sure people will :)

jonnieZG, 2018-03-13:
If we stick with UTF-8, then we'd have to have an additional mapping between charCode and bitmapData, and that map might get quite big. So if we encode the chars as I described, we eliminate the need of having a translation map. Instead of that, we'll just have 8-16 bit char being mapped to a bitmap directly in order of appearance. We will just have to ensure that space ' ' and digits '0'-'9' always have the same code (for example codes 0-9 for digits, and 10 for space) so you could more easily print numeric values.

Ralim, 2018-03-14:
@jonnieZG 
I understand, was thinking UTF-8 even with the extra lookup might be less code to implement -> Faster release is mostly why :) 

Looks like this could be compute able using the `constexpr` keyword in normal C++, but I haven't tested this in the microcontroller build as of yet. Planning to make a standalone c++ app to test this and then if that works back-port this to the firmware.

Alternately we *could* have a javascript tool to manage the editing and output pre-processed files for this, and just have those particular files stored in git-lfs. Then just perform the calculation before building release builds. This probably depends on how good you are with javascript too (I'm not a web person)

With the allocation to uint16_t I agree we could keep the UTF-8 style of using the 0x80 bit as a marker for extended format. We could then just sort all chars by usage counts to decide on the <0x80 values as you suggest.

jonnieZG, 2018-03-14:
@Ralim, I didn't work with STM32, but according to my experience with Arduino, I believe that using ```constexpr``` alone is not enough. If you want some data to be stored in flash memory, you will also have to write a method that fetches that data back from the FLASH into the RAM, when you want to work with it (e.g. to draw a character bitmap). See [this](https://stackoverflow.com/questions/37513118/reading-internal-flash-memory-contens). I think that mere marking an array with constexpr and reading that data using array[...] just won't work.

Ralim, 2018-03-14:
Constexpr and reading data are different things.

On the STM32 when the data array is marked as being a constant It will be left in flash and not copied to ram. So you can work with it directly. There are other ways to force this to happen, but the easiest way to verifify is to take a pointer to the content and print that to the LCD screen and see if it starts with 0x08.....

Which is where the internal flash is mapped to. 
Compilers are also getting alot better at spotting read only variables and leaving them in flash. 

The settings structure on the other hand is an example where we operate with a ram copy and read/write it back and forth between the flash.

I can go into more on this if you want. 

With constexpr we are telling the compiler it can evaluate this particular function at compile time and just store the results. This is often used for things such as calculating constants that don't change in the firmware but might change on new versions. Such as the sizeof command.

So the array would be a compile time constant with its contents defined by calling a constexpr function.

Doridian, 2018-03-21:
You can squeeze some size out if needed by swapping `-O2` for `-Os`
```
-Os
Linking TS100_EN.elf
arm-none-eabi-size -x Hexfile/TS100_EN.elf
   text    data     bss     dec     hex filename
 0x9f0c   0x6e8  0x3400   55796    d9f4 Hexfile/TS100_EN.elf
arm-none-eabi-objcopy Hexfile/TS100_EN.elf -O ihex Hexfile/TS100_EN.hex

-O2
Linking TS100_EN.elf
arm-none-eabi-size -x Hexfile/TS100_EN.elf
   text    data     bss     dec     hex filename
 0x94bc   0x6e8  0x3400   53156    cfa4 Hexfile/TS100_EN.elf
arm-none-eabi-objcopy Hexfile/TS100_EN.elf -O ihex Hexfile/TS100_EN.hex
```

I have also tried enabling LTO, but it seems the generated assembly doesn't quite work then (the assembler fails)...

Ralim, 2018-03-22:
@Doridian 
I'm aware I can swap to Os, but I would rather not until I have verified the timing characteristics for all of the ISR's under that compiling option. :)

jonnieZG, 2018-03-22:
I've made a translation editor #249 that produces separate JSON files for each language, and now I'm working on Python script that will build a Translation.cpp *from* these JSON files.

Ralim, 2018-03-22:
Thanks @jonnieZG.
I'll bug fix the 2.03 RC and then look into this for making a new release :)

(I'm not ignoring you, I really appreciate the work, just busy)

jonnieZG, 2018-03-23:
For the beginning, I've made a Python script that builds the Translation.cpp from JSON files for the single-language version file, just to see if the language files are ok.

I am writing now a proof-of-concept that will use multi-lingual messages.

jonnieZG, 2018-03-26:
@Ralim, this is the concept that I had in mind. See example below. It can write a corresponding weekday name in one of the defined languages, but the same concept will be applied to all the required strings.

It is only that instead of the characters array, we'll be having an array of bitmaps for each character. I am just not sure whether the compiler will put that const char arrays into the flash memory... I've tested the concept [in this online compiler](https://www.tutorialspoint.com/compile_cpp_online.php)

```C++
#include <iostream>

using namespace std;

// START Auto-Generated Block

#define LANG_EN 0
#define LANG_HR 1
#define LANG_COUNT 2

// Generic Strings
#define STR_sunday 0
#define STR_monday 1
#define STR_tuesday 2

// Instead of having ASCII codes, we'll be having an array of bitmaps for each character
// 0=END, S=1, u=2, n=3, d=4, a=5, y=6, M=7, o=8, T=9, e=10, s=11, N=12, j=13, l=14, P=15, k=16, U=17, t=18, r=19
const char chars[] = {'S','u','n','d','a','y','M','o','T','e','s','N','j','l','P','k','U','t','r'};

const char sunday_en[] = {1,2,3,4,5,6,0}; // Sunday
const char monday_en[] = {7,8,3,4,5,6,0}; // Monday
const char tuesday_en[] = {9,2,10,11,4,5,6,0}; // Tuesday

const char sunday_hr[] = {12,10,4,13,10,14,13,5,0}; // Nedjelja
const char monday_hr[] = {15,8,3,10,4,13,10,14,13,5,16,0}; // Ponedjeljak
const char tuesday_hr[] = {17,18,8,19,5,16,0}; // Utorak

const char* strings[3][LANG_COUNT] = {
    {sunday_en, sunday_hr}, 
    {monday_en, monday_hr}, 
    {tuesday_en, tuesday_hr}, 
};

// END Auto-Generated Block

int print(const char *str[][LANG_COUNT], int stringId, int languageId) {
	const char *pc = str[stringId][languageId];
	while (true) {
		const char c = *pc;
		if (c == 0)
			break;
		// Instead of printing an ASCII char, we'll be rendering a corresponding bitmap
		int charIndex = (int)c - 1; // we'll also have to handle two-byte encoded chars for bytes > 127
		cout << chars[charIndex];
		pc++;
	}
    cout << endl;
}
int main()
{	
    print(strings, STR_monday, LANG_HR);
    return 0;
}
```

jonnieZG, 2018-03-31:
Ok, I've made a much more accurate calculation, and with the total of the current 17 languages, and the described encoding, we have the following size:

* Both Big and Small Fonts: ~6500 bytes
* Array of arrays: ~2200 bytes
* Strings: ~30000 bytes
**TOTAL: 38700 bytes**

This is definitely much more than we expected, but as I see it, we have 2 options:

a. We cut down the descriptions to bare minimum;
b. We split the languages into 2 or 3 groups by regions, each group having English as a common language. That way we would  have a multi-lingual firmwares, but there would be 2 or 3 of them with different language groups.

My script reuses strings, rather then making multiple definitions of a same string (using ```define```). It keeps the existing ASCII range unchaged, but for chars above 127, it first writes lowest 7 bits with 8th bit set to 1, and then the rest of the bits in the next string. For example, a 16-bit value of ```00010010 01101011``` will be encoded as two bytes: ```11101011``` (least significant byte first, with bit7 set to 1) and ```00100100```

This is the file that my script generated for the current 17 languages.

```C++
/********** AUTOMATICALLY GENERATED - DO NOT EDIT **********/

// ================ COMMON CONSTANTS ================

#define LANG_COUNT 17    /* Number of languages */

// ---- Special Character Codes ----

#define CHAR_EOS    0    /* End-of-String */

// ---- Language IDs ----

#define LANG_EN     0   /* English */
#define LANG_CS     1   /* Český */
#define LANG_DA     2   /* Dansk */
#define LANG_DE     3   /* Deutsch */
#define LANG_ES     4   /* Español */
#define LANG_FR     5   /* Français */
#define LANG_HR     6   /* Hrvatski */
#define LANG_HU     7   /* Magyar */
#define LANG_IT     8   /* Italiano */
#define LANG_LT     9   /* Lietuvių */
#define LANG_PL    10   /* Polski */
#define LANG_PT    11   /* Português */
#define LANG_SK    12   /* Slovenčina */
#define LANG_SV    13   /* Svenska */
#define LANG_TR    14   /* Türkçe */
#define LANG_RU    15   /* Русский */
#define LANG_BG    16   /* български */

// ================ STRING INDEXES ================

// ---- Settings Groups ----

#define LOC_SG_SOLDERINGMENU                       0
#define LOC_SG_POWERSAVINGMENU                     1
#define LOC_SG_UIMENU                              2
#define LOC_SG_ADVANCEDMENU                        3

// ---- Settings Names ----

#define LOC_SN_POWERSOURCE                         0
#define LOC_SN_SLEEPTEMPERATURE                    1
#define LOC_SN_SLEEPTIMEOUT                        2
#define LOC_SN_SHUTDOWNTIMEOUT                     3
#define LOC_SN_MOTIONSENSITIVITY                   4
#define LOC_SN_TEMPERATUREUNIT                     5
#define LOC_SN_ADVANCEDIDLE                        6
#define LOC_SN_DISPLAYROTATION                     7
#define LOC_SN_BOOSTENABLED                        8
#define LOC_SN_BOOSTTEMPERATURE                    9
#define LOC_SN_AUTOSTART                           10
#define LOC_SN_COOLDOWNBLINK                       11
[... 2313 lines cut ...]
0x00,0x00,0xc0,0xe0,0x30,0x10,0x10,0x30,0xe0,0xc0,0x00,0x00,0x00,0x00,0xff,0xff,0x21,0x21,0x21,0x33,0x3f,0x1e,0x00,0x00, //   #148  ß  c3 9f
0x00,0x80,0xc0,0xe0,0x68,0x6c,0x66,0x62,0xe0,0xc0,0x80,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x38,0x1f,0x0f,0x00, //   #149  ó  c3 b3
0x00,0x00,0xe0,0xe8,0x6c,0x64,0x6c,0x68,0xec,0xc4,0x80,0x00,0x00,0x00,0x3f,0x3f,0x00,0x00,0x00,0x00,0x00,0x3f,0x3f,0x00, //   #150  ñ  c3 b1
0x00,0x00,0x00,0x00,0x80,0xf3,0xf3,0x80,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x0f,0x3f,0x3f,0x0f,0x00,0x00,0x00,0x00, //   #151  ¡  c2 a1
0x00,0xe0,0xe0,0x00,0x08,0x0c,0x06,0x02,0x00,0xe0,0xe0,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x18,0x3f,0x3f,0x00, //   #152  ú  c3 ba
0x00,0x00,0x00,0x80,0xc0,0xfb,0x7b,0x00,0x00,0x00,0x00,0x00,0x00,0x0e,0x1f,0x3b,0x31,0x30,0x30,0x30,0x38,0x1e,0x0e,0x00, //   #153  ¿  c2 bf
0x00,0x80,0xc0,0xe0,0x62,0x66,0x6c,0x68,0x60,0xc0,0x80,0x00,0x00,0x0f,0x1f,0x33,0x33,0x33,0x33,0x33,0x33,0x13,0x03,0x00, //   #154  è  c3 a8
0x00,0x00,0x40,0x60,0x62,0x66,0x6c,0x68,0x60,0xe0,0xc0,0x00,0x00,0x1c,0x3e,0x33,0x33,0x33,0x33,0x33,0x33,0x3f,0x3f,0x00, //   #155  à  c3 a0
0x00,0x80,0xc0,0xe0,0x60,0x60,0x60,0xe4,0xc4,0xff,0xff,0x04,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x30,0x3f,0x3f,0x00, //   #156  đ  c4 91
0x00,0x80,0xc0,0xe0,0x60,0x60,0x68,0x6c,0x64,0xc0,0x80,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x30,0x18,0x08,0x00, //   #157  ć  c4 87
0x00,0x60,0xf0,0xf8,0x99,0x9b,0x9a,0x9b,0x99,0x30,0x20,0x00,0x00,0x0c,0x1c,0x39,0x31,0x31,0x31,0x31,0x3b,0x1f,0x0e,0x00, //   #158  Š  c5 a0
0x00,0x80,0xe0,0x70,0x38,0x18,0x1a,0x1b,0x39,0x70,0x60,0x00,0x00,0x03,0x0f,0x1c,0x38,0x30,0x30,0x30,0x38,0x1c,0x0c,0x00, //   #159  Ć  c4 86
0x00,0x00,0x00,0x80,0xe4,0x7e,0x7b,0xe1,0x80,0x00,0x00,0x00,0x00,0x38,0x3e,0x0f,0x0d,0x0c,0x0c,0x0d,0x0f,0x3e,0x38,0x00, //   #160  Á  c3 81
0x00,0x80,0xc0,0xe0,0x68,0x64,0x68,0x64,0xe0,0xc0,0x80,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x38,0x1f,0x0f,0x00, //   #161  ő  c5 91
0x00,0xe0,0xe0,0x00,0x08,0x04,0x08,0x04,0x00,0xe0,0xe0,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x18,0x3f,0x3f,0x00, //   #162  ű  c5 b1
0x00,0xe0,0xe0,0x00,0x02,0x06,0x0c,0x08,0x00,0xe0,0xe0,0x00,0x00,0x0f,0x1f,0x38,0x30,0x30,0x30,0x30,0x18,0x3f,0x3f,0x00, //   #163  ù  c3 b9
0x00,0x00,0x00,0x1e,0x3f,0x33,0x33,0x3f,0x1e,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00, //   #164  °  c2 b0
[... cut ...]
0x00,0x00,0x00,0x00,0x60,0xec,0xec,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x30,0x30,0x7f,0xbf,0xb0,0x30,0x00,0x00,0x00, //   #168  į  c4 af
0x00,0xe0,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0xe0,0xe0,0x00,0x00,0x0f,0x1f,0x38,0x30,0xf0,0xb0,0xb0,0x18,0x3f,0x3f,0x00, //   #169  ų  c5 b3
0x00,0x80,0xc0,0xe0,0x60,0x60,0x60,0x60,0x60,0xc0,0x80,0x00,0x00,0x0f,0x1f,0x3b,0x33,0x33,0x73,0xb3,0xb3,0x13,0x01,0x00, //   #170  ę  c4 99
0x00,0x00,0x00,0x03,0x03,0xff,0xff,0x03,0x03,0x00,0x00,0x00,0x00,0x00,0x00,0x30,0x30,0x7f,0xbf,0xb0,0x30,0x00,0x00,0x00, //   #171  Į  c4 ae
0x00,0x18,0x18,0x18,0x19,0x1b,0x9a,0xdb,0xf9,0x78,0x38,0x00,0x00,0x30,0x3

Ralim, 2018-04-04:
Thank you for your continued work on this :) 

Bummer that it wont fit. Out of curiosity, is there a way to group the languages so that they have the maximum overlap of font symbols to reduce font table size possibly?

I'm happy to go with your best feeling on this really, since you are making great progress :)
Is there also an encoding system we could look into using to split strings if they have overlap? so some things are composed of joins of sub strings? (No idea if that will work).

It may be possible to find a trivial compression algorithm we could use on the fonts possibly?

jonnieZG, 2018-04-11:
I have compiled my code, and it seems that we have just around 12000 bytes for fonts and strings data, not 15000 as you presumed. It is enough just for 4 languages at a time, or 3 languages if one of them uses Cyrillic script.

I will see how much would we gain by using Huffman coding (i.e. Deflate compression).

Ralim, 2018-04-12:
Hmm, Huffman might reduce some encoding but not a huge amount i dont think.

The firmware has continued to grow as we work on this, which is not helping with the flash space utilization :cry:

jonnieZG, 2018-04-13:
@Ralim, **IT WORKS!!!** [This](https://youtu.be/Y64NMb-55hg) is the first multilingual version that does not use any compression. This specific package is compiled for English, Croatian and Russian. If there was another Latin-script language instead of Russian, there would be place for 4 instead of 3 languages. 

Now I'll make a version that uses Huffman compression for text and fonts, so I expect it to be able to squeeze in 6-7 languages per package. Huffman gives about 45% compression on text files. Not all-in-one, but still much better than having just one language per build.

If we want to squeeze in more languages, the translators should consider shortening descriptions to the bare minimum, and use them only in places where really needed.

jonnieZG, 2018-04-16:
@Ralim, I am doing an overall test with Latin and CyriIlic and all the current translations, and I am getting ~40% compression of the font data (4217 instead of 6690 bytes), but only if I unpack it after boot directly into the RAM and use it from there. Dynamic unpacking (reading and unpacking each character before displaying) would drop the compression rate to ~25%. 

So I was wondering if you could estimate how much extra RAM do we currently have, because it would make things easier if I could first unpack both fonts into RAM, rather than doing on-the-fly decompression.

jonnieZG, 2018-05-03:
Just reporting that I've been quite busy on some other projects the last couple of weeks, but I **will** complete the implementation of this feature in the next month or so.

Mrkvozrout, 2018-05-03:
_I already asked this in other issue, but this is the better place - copied from #273_ 

@jonnieZG @Ralim 
Please take this as my opinion that triggers discussion, don't take it bad or personally.

Is it still a good idea to strive for multilingual FW when there is not enough memory to have all in 1 (or at least whole language familly in 1) comfortably?

It would be a nice feature, BUT:

- it is not that useful (at least in my opinion) - feature is used only one time an the rest of device life it just **occupies most of memory without real usage**
- it limits the space for translators
- it limits space for new features (e.g. features with menu options, animations...) - features that add more value to every time usage
- user must go to settings and change the language to desired one - not comfortable
- user still needs to flash this FW so why not to just flash the correct one (with target language) - saves tons of space, no need to dig in the menu to set correct language before first use

#222 Multilingual feature would be nice with enough space, but not with all these downsides. It is a nice excercise to optimize the code and memory usage but when it goes that far so description lengths are limited or removed (effectivelly meaning new features with menu options are forbidden) - I don't like the idea any more.

jonnieZG, 2018-05-03:
@Mrkvozrout, I am glad you are stating your opinion. I am not taking this personally. :) This feature is probably a matter of taste, so I will state my opinions, but it is up to @Ralim to give a final "go" or "no go".

- Some prefer their local language, some prefer English. I live in Croatia, and I know many more people who prefer English on their gadgets over Croatian. It is a matter of preferring English as a "technical" language, especially if the "native" translation lacks in good or consistent technical terms. If I want to borrow or sell the device to someone, he would most certainly exclusively prefer one or the other. Having the possibility to choose between the English and a local language adds a great value.
- This is a soldering iron, not a spacecraft. Descriptions were making sense back in times when the option names were just acronyms. Now you have verbose two-line names, and even a completely fresh user seldom requires further explanations of each. There are user manuals for those.
- Additional features will not suffer. When a good, *functional* idea arises, we will implement that feature, and we would always have a possibility to play with number of languages in a build. I don't think that @Ralim would agree to implement novelty features like "an oscilloscope" or "a portable accelerometer", or "tetris", so I don't expect the firmware to significantly grow further. Having all the languages in a single build is possible only if we drop out the descriptions. Still with the descriptions, we can pack them in groups by regions, plus English as a common language for each build. Having a possibility to choose between at least two is still *much* better than having none. 
- Animations over multilinguality? This issue could start a religious war... ;) I would always choose the second over the first one, because I prefer functional over aesthetic improvements in tools, but that's just me.
- Mind-reading is hard to implement, so I guess we can not spare users from the annoyance of changing it through the Settings. ;)
- As I said, packing languages in groups by region, or dropping out the descriptions and having them all in one.

Mrkvozrout, 2018-05-03:
- mostly I like english more too, regarding the technical terms translations
- descriptions are closer to user than manual
- ok, but you are talking about _must have_ features, I think also some _nice to have_ features could add value to the FW (sure not oscilloscope :D )
- I also like functionality over aestetics more, but this functionality is really memory-hungry O:)

From what you said, the best compromise could be to just add the English to every language (just two langs in each bundle). I might switch between Czech and English, but I will surely not switch to Ukraine, Croatian or even Slovak (even if I understand Slovak out of box :D ). It would add value but not eat all memory :)

jonnieZG, 2018-05-03:
Decompression has a very small footprint, so it would also pay to implement it if we just had only one language. It also saves a lot of memory on the fonts. 

@Mrkvozrout,  the two of us basically agree on the matter, except that I would like to push the concept of packing the languages in groups, *as long as the memory allows us to*. Once we approach the memory limit, we keep the English + one language per build, since it would be pretty much the same as having one language and font with no compression at all.

Mrkvozrout, 2018-05-03:
Since you count with adjusting package size in future, I am OK with it ;)

Ralim, 2018-05-05:
Hi,
I'm still looking forward to packing this feature in. 
Current builds are low on ROM (code) room at the moment (3k approx), but I'm hoping we will gain more with the reduce font data etc as well. There is currently at least 5k of RAM free for on the fly use.

I'm currently working through sections of the code base still trying to find the lockup bug, and cleaning up sections as I go too, so might be able to free up some more flash space hopefully as well.

I like the sound of packing English+ at least one extra language in each build. 
Is it worth having the system try and find pairings that maximize compression at compile time?

I'm happy to use a github-pages site to host the editor or throw it on my site (ralimtek.com) to keep it easy to find.

federck, 2018-05-08:
Maybe a web configuration tool capable of building a firmware depending on users needs would find a compromise between discordant opinions.
E.g.: @jonnieZG's concept of the translation tool would reamin, but it will be readjusted in order to include a page with check boxes and a bar indicating memory space; in order to choose which feature to have on firmware, such as animated icons or multilanguality, depending on residual resources.

jonnieZG, 2018-05-08:
@federck , that's one brilliantly sweet idea!!!

Mrkvozrout, 2018-05-08:
This is what I have in my mind for some time :) something like lua FW builder for ESP8266 https://nodemcu-build.com

closed 2020-04-04 by Ralim