Ralim/IronOS #223: Display doesn't time out V2.03-alpha

csdpm, 2018-03-09:
I'm surprised nobody else has posted this one... with V2.03-alpha installed (original h/w), the display doesn't time out (blank) when sitting idle for a while.

It behaved fine on the previous release, I've tried a factory reset but it doesn't make any difference.

Given the worries about OLED lifetime, the previous behavior would be helpful.

Ralim, 2018-03-10:
Thanks for spotting this.
It's hard to manually do coverage tests for all the features these days, which is partly why I push out the alphas as it's amazing how fast other people can find problems in your code tag you can't.

Someone else has already submitted a pull request to fix the typo, so I'll push out another alpha soon.

SmayK, 2018-03-11:
hello, small contribution on translation for French, here they are:
```
const char* SettingsCalibrationWarning = "Assurez-vous que la panne soit à température ambiante avant de continuer!";
const char* SettingsResetWarning = "Voulez-vous vraiment réinitialiser les paramètres aux valeurs d'usine?";
const char* UVLOWarningString = "DC FAIBLE";            // <=8 chars
const char* UndervoltageString = "Sous-tension";     // <=16 chars
const char* InputVoltageString = "V d'entrée: ";        // <=11 chars, preferably end with a space
const char* WarningTipTempString = "Temp. Panne: ";     // <=12 chars, preferably end with a space
const char* BadTipString = "PANNE HS";                // <=8 chars
const char* SleepingSimpleString = "Zzzz";              // Must be <= 4 chars
const char* SleepingAdvancedString = "En veille...";    // <=16 chars
const char* WarningSimpleString = "HOT!";               // Must be <= 4 chars
const char* WarningAdvancedString = "ATTENTION! CHAUD"; // Must be <= 16 chars
const char* SleepingTipAdvancedString = "PANNE:";      // <=6 chars
const char* IdleTipString = "PANNE:";     // IdleTipString+IdleSetString <= 10
const char* IdleSetString = " Set:";    // preferably start with a space; IdleTipString+IdleSetString <= 10
const char* TipDisconnectedString = "PAS DE PANNE!"; // <=16 chars
const char* SolderingAdvancedPowerPrompt = "Puissance: ";
```

csdpm, 2018-03-16:
Unfortunately not quite fixed in v2.03-RC2  :(

The display now times out correctly but after a while it becomes impossible to start the TS100 up again, either by movement or button press. It is necessary to disconnect the power in order to get the iron to work again.

JohnEdwa, 2018-03-16:
I had this happen today as well. First rev hardware, 19V psu.

SmayK, 2018-03-16:
I confirm the problem with 24v

JohnEdwa, 2018-03-19:
@Ralim I just noticed that after the iron gets to this unresponsive mode, it'll eventually (10ish minutes, hard to say?) wake up on its own again with no movement at all. 

Maybe a faulty watchdog wakeup?

Ralim, 2018-03-23:
This should hopefully be fixed in RC3 I have just uploaded, let me know if its still bugging up for anyone.

Not sure on the 10 minute wakeup, could be a rollover artifact. :|

csdpm, 2018-03-23:
Sorry Ben, RC3 is still locking up in the same way, times out to blank display then can't be roused after a while.

I've also noticed that the Sleep temp bug is back - or at least a display bug. Sleep is set at 180'C but display hovers between about 250 and 230 varying rapidly over a couple of seconds so I don't think the actual tip temperature can be tracking that fast. Tip BC2 supply 24V.

SmayK, 2018-03-23:
the RC3 does not have the latest version of the normal French language?

Ralim, 2018-03-25:
@SmayK Please keep Language issues in their own issue.

@csdpm Can you confirm a rough value for how long it needs to be off until it cannot be woken up by movement?
The sleep temp is not a display bug but actually what is being measured inside the tip (not the outside is probably a lot more stable). This occurs whenever you have a high supply voltage but a small temperature. I'm still trying to see if its possible to have a nice PID tuning that works well for 24V and 12V, with the different tip sizes.

csdpm, 2018-03-25:
Sorry, for the delay, it took a bit of tying down. From power applied (no heating)... From detailed idle screen going off, at 5:30 it still wakes up, at 6 minutes it doesn't. I hope that helps narrow it down a bit.

Understood on the sleep temperature PID tuning(though it still looks far too fast for actual temperature variation), shame, you had it nailed on 2.02.

Ralim, 2018-03-26:
No problem at all on the delay.

That helps narrow it down, and gives me a test case to investigate further :)

Temperature is read from the tip ~ 10 times per second which is used to adjust the PID. With some of the newer units that have come out, they seem to have a slightly different tolerance on some of the components, so I have adjusted the timings for the PWM loop. I'm still tweaking the new PID to try and bring it back into how it performed on 2.02. (2.03 is currently a bit too aggressive). I should have more news on this when I next get time to test :)

Ralim, 2018-03-31:
Hi, Can you confirm if the 2.03 release fixes this or not.
If it doesnt I'll get on it next :)

JohnEdwa, 2018-03-31:
Nope, didn't fix it for me at least.

csdpm, 2018-04-04:
> Hi, Can you confirm if the 2.03 release fixes this or not.
If it doesnt I'll get on it next :)

Hi, Sorry for the delay, I've been away for a few days. Yes, it seems to be waking up normally for me now, I've tried leaving it for the previous periods and much longer and it wakes up every time I pick it up. 

The temperature display is still rather unstable when hot. BC2 24V 320'C, I know you're still working on that, but you seem to have magically fixed the sleep temperature back to my 180'C set-point.

Many thanks. :)

csdpm, 2018-04-04:
Correction, it does still eventually fail to wake up - hard to pin down but I think it's probably at least an hour or so.

Maybe a slow memory leak? (typical hardware engineer suggestion when he has no idea!).

Ralim, 2018-04-04:
Thanks for the correction.

I'm basically wondering the same thing after staring down the code for a while. Going to need to spend a while hunting. 

Yep I'm going to be tuning the pid code different soon too to try and fix the temps and update the display filter too

JohnEdwa, 2018-04-07:
I think that fixed it, at least I've had it on my table the whole day and it hasn't had any issues waking up.  
I'll leave it plugged for the night and poke it in the morning just to make sure, but I think it's fine.

This is on PCB1, if it makes any difference.

csdpm, 2018-04-07:
Hi Ben,
Yes I concur, after an extended run, I think you've fixed it. Thanks.

I don't know if you've tried to tweak the temperature stability in this image, it seems a tiny bit more stable but could well be my imagination, still pretty jumpy.

Ralim, 2018-04-07:
That's fantastic to hear.

And yeah, I'm trying to rev a new version asap to bug fix the 2.03 version

JohnEdwa, 2018-04-07:
Huh, might not be completely fixed yet, mine failed to wake up in the morning.

Ralim, 2018-04-09:
@csdpm yeah it looks like its either a memory leak somewhere, or something going awry in the FreeRToS setup.

I need to try setting up some kind of tracing for this :/

Ralim, 2018-04-14:
I think I have found the source for this and have patched it in the latest builds im doing.
Going to wait for #211 's new graphics and ill push up another RC.

closed 2018-04-20 by Ralim

csdpm, 2018-04-20:
Sorry Ben,
I saw your note and also downloaded V2.04RC2 last weekend. I thought that your note was after the release so was waiting for the next one to come out.

I've re-downloaded RC2 again today, compared and re-loaded. Unfortunately my iron has become unresponsive again after about three hours (I'm really not sure if the failure time is constant because I'm sure it has lasted much longer previously). Sorry for the bad news.

Just an aside, it's having trouble getting down to my 180'C sleep temp again (24V).

If it's really just me seeing this, then I still have the option of dropping back to V2.02 as I have the original h/w. No worries there, :)

weav2k4, 2018-04-20:
@csdpm You're not the only one.  I've noticed as well it will not wake after extended idle.  Also have issue of sleep temp too high and rapid fluctuation.  I just adjust sleep -50C to what I actually want and it gets close.  I also use a 24v supply and a bc2 if matters.  I appreciate all the effort Ben is taking on this little iron!  Worked on a small project yesterday and love how it performs.

Ralim, 2018-05-07:
@freetushkan No room. Since to add logging it would require at least ~5k of flash free.
I have added extra readouts for debugging in the 2.04.1 bugfix to help address this.

freetushkan, 2018-05-07:
@Ralim Okay. Will test now... Do we need to know or provide something to you or just blindly test it?

Ralim, 2018-05-07:
Nothing really of note, since the issue shows up over time randomly

freetushkan, 2018-05-07:
@Ralim First thing - cyrillic symbols desapeared... English version display characters normally. Should I create separate issue? May be it related to my commits. If it's true, i'm sorry, but have no idea why.

Ralim, 2018-05-07:
@freetushkan Move to its own issue. *Any* changes in translations can cause issues if encoding changes etc as well. One reason to move to the more unified process that will eventually come along.

csdpm, 2018-05-09:
After a couple of days of sleep and occasional nudging,  I'll stick my neck out and say that I think you've nailed it this time. Longest time between nudges ~12 hours.

Well done and thanks. :)