Ralim/IronOS #225: Erratic Temperature Display while heating up

otinex, 2018-03-11:
The display of the temperature in the heating mode is somewhat erratic and "overshoots" by quite a bit.
See Video: [https://youtu.be/6x6O_vK5X-Y](https://youtu.be/6x6O_vK5X-Y)

Ralim, 2018-03-12:
Hi,
Can you please provide details on which tip you are using as well as your power supply?

otinex, 2018-03-12:
Hello,
I use the 2.02 Version of the Firmware. As Tip I use the TS-I that came with it. As a power supply I use a 24 V 1.5 A SMPS.

Ralim, 2018-03-13:
Ah okay,
I have found the same issue with my small tips for the iron as well. It's down to the current pid tuning which is done for the larger tips. This changed on the 1.x to 2.x transition as the pid was completely rewritten. Can you confirm the last firmware that worked for your tip so I can compare with the one I have here.

Also that supply is probably getting a bit pushed as the iron cam pull up to 2.5A at times (but probably is fine).

I'll back to this soon hopefully with more information

otinex, 2018-03-13:
It must have been V 1.16 because I got the TS100 in July of 2017.

Ralim, 2018-03-23:
Could you please verifiy if the latest RC3 helps with this issue?

marcotocco, 2018-03-23:
Same problem here, im using the v 2.03 (PCB 2) with BC2 iron tip.

Havent got a problem with the previous versione (the first one fully compatible with the PCB 2 without new menus)