Ralim/IronOS #227: Thankyou! #metoo

phedders, 2018-03-13:
My TS100 arrived yesterday.

"Wow it really is tiny"
"Is that really all the firmware can do?"

So of course within 10 minutes your firmware was on my iron and it is fantastic.
Many thanks.

(Sorry for the bug noise, but maybe this should be a sticky!

nerocide, 2018-05-30:
@Ralim  Thanks a LOT for your work.
Default FW is so much features weak compared to yours, there is an OCEAN. As @phedders said, 10mins after unboxing FW needed to be changed.
Again thanks.

lrusnac, 2018-05-30:
Issues like these should be made more often, really thanks for your hard work on this project @Ralim

m0lmk, 2021-01-28:
Another thank you from me! Same story. Arrived today and 10 minutes later had been flashed with the Ralim firmware. I've used it alongside my Hakko FX-888D for a few hours and I think the Hakko is going to spend quite some time on the shelf as a spare. Fantastic work!

ackzsel, 2021-10-01:
Got my SQ-001 in the mail yesterday. Flashed it with IronOS, fiddled with the settings, calibrated the voltage, made a boot logo and it's nothing less than awesome!

Thanks to all developers of this fantastic project!

builder555, 2023-02-07:
Just wanted to chime in with another "Thank You". Speaking as a user and a developer, you've been doing excellent work! 👏