Ralim/IronOS #230: Please russian language. Also can I ask you about version with max info on-screen with mini-font?

bakerenko, 2018-03-14:
* **I'm submitting a ...**

  - [X] Feature request. 
Can I ask you a version with max info on-screen. I always like, be-in-know how temperature is and what voltage on battery, time left to cooling mode (maybe graphical line in top as battery level on android). and maybe some quick tips about amperage drain? if possible, of course.
  - [X ] Translation russian as was in previous version.

Ralim, 2018-03-14:
Hi,
Have you tried turning on the advanced screens? These show the current voltage and temperature all the time.

The iron does not have means to measure current so I don't show any current values on the screen. 

In regards to Russian I'll push it up shortly, sorry about that. Once it's up, would you be open to updating its translation?

Ralim, 2018-03-14:
@bakerenko 
I have added the missing Russian hex file to the latest RC2 release.

bakerenko, 2018-03-15:
Thank you!
So, how about indicator of timeout? Like time indicator on japanese traffic lights? A decreasing one pixel line indicating how much time isleft to cooling mode.

bakerenko, 2018-03-15:
I have tested Russian and it is mixed with english (: Can I help you with translating?

Undervoltage Input V = Низкое входное напряжение
Soldering Settings = Режимы паяльника
Sleep Modes = Настройки энергосбережения
User Interface = Пользовательский интерфейс
    DESCSP = СКОРПР (Скорость прокрутки текста ), F = Б , S = М
Advanced options = Расширенные настройки

main screen multi info small font text:
Tip = Темп
Set = Уст
Input V = Вх Напр 

ALSO as we are using small text - could we improve those cutted russian descriptions to full?

ИстП = Источник питания
ТТур = Температура турборежима
Астарт = Режим автостарта
Тожд = Температура  в реж ожидания
Вожд = Время перех в реж ожидания
Тоткл = Время до отключения
ЧувсДв = Чувств датчика движения 
ЕдТемп = Единица измерения температуры
ПовЭкр = Поворот экрана
Охлажд = Отобр темп при охлажд
ИнфОжд = Расш инфо в ожидании
ИнфПай = Расш инфо при пайке
КалибрТ = Калибровка темодатчика
КалибрU = Калибровка вх напряжения

Ralim, 2018-03-23:
Hi,
Can you please just edit this block and post back and I will update the translation:
```
#ifdef LANG_RU
const char* SettingsDescriptions[17] = {
	// These are all the help text for all the settings.
	// No requirements on spacing or length.
	/* Power source (DC or batt)          */ "Источник питания. Установка напряжения отключения. <DC 10V> <S 3.3 V на батарею>",
	/* Sleep temperature                  */ "Температура режима ожидания <С>",
	/* Sleep timeout                      */ "Время до перехода в режим ожидания <Минуты>",
	/* Shutdown timeout                   */ "Время до отключения <Минуты>",
	/* Motion sensitivity level           */ "Акселерометр <0. Выкл. 1. мин. чувствительный 9. макс. чувствительный>",
	/* Temperature in F and C             */ "В чем измерять температуру",
	/* Advanced idle display mode enabled */ "Показывать детальную информацию маленьким шрифтом на домашнем экране",
	/* Display rotation mode              */ "Ориентация дисплея <A. Автоматический, Л. Левая рука, П. Правая рука>",
	/* Boost enabled                      */ "Турбо-режим при удержании кнопки А при пайке ",
	/* Boost temperature                  */ "Температура в турбо-режиме",
	/* Automatic start mode               */ "Автоматический запуск паяльника при включении питания. T=Нагрев, S=Режим ожидания,F=Выкл.",
	/* Cooldown blink                     */ "Показывать температуру на экране охлаждения, пока жало остается горячим.",
	/* Temperature calibration enter menu */ "Калибровка термодатчика.",
	/* Settings reset command             */ "Сброс всех настроек.",
	/* Calibrate input voltage            */ "Калибровка напряжения входа. Настройка кнопками, нажать и удержать чтобы завершить.",
	/* Advanced soldering screen enabled  */ "Показывать детальную информацию при пайке.",
	/* Description Scroll Speed           */ "Speed this text scrolls past at",
};

const char* SettingsCalibrationWarning = "Убедитесь, что жало остыло до комнатной температуры, прежде чем продолжать!";
const char* SettingsResetWarning = "Are you sure to reset settings to default values?";
const char* UVLOWarningString = "БАТ РАЗР";              // <=8 chars
const char* UndervoltageString = "Undervoltage";         // <=16 chars
const char* InputVoltageString = "Input V: ";            // <=11 chars, preferably end with a space
const char* WarningTipTempString = "Tip Temp: ";         // <=12 chars, preferably end with a space
const char* BadTipString = "BAD TIP";                    // <=8 chars
const char* SleepingSimpleString = "Хррр";               // Must be <= 4 chars
const char* SleepingAdvancedString = "Ожидание...";      // <=16 chars
const char* WarningSimpleString = " АЙ!";                // Must be <= 4 chars
const char* WarningAdvancedString = "ОСТОРОЖНО! ГОРЯЧО"; // <=16 chars
cconst char* SleepingTipAdvancedString = "Tip:";         // <=6 chars
const char* IdleTipString = "Tip:";                      // IdleTipString+IdleSetString <= 10
const char* IdleSetString = " Set:";                     // preferably start with a space; IdleTipString+IdleSetString <= 10
const char* TipDisconnectedString = "TIP DISCONNECTED";  // <=16 chars
const char* SolderingAdvancedPowerPrompt = "Power: ";    // <=12 chars

/*
 * #TODO change support for multibyte constants here
 const char SettingRightChar = 'П';
 const char SettingLeftChar = 'Л';
 const char SettingAutoChar = 'A';*/

const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';

const char SettingFastChar = 'F';
const char SettingSlowChar = 'S';

const enum ShortNameType SettingsShortNameType = SHORT_NAME_DOUBLE_LINE;
const char* SettingsShortNames[17][2] = {
  /* (<= 11) Power source (DC or batt)          */ {"Power", "source"},
  /* (<=  9) Sleep temperature                  */ {"Sleep", "temp"},
  /* (<=  9) Sleep timeout                      */ {"Sleep", "timeout"},
  /* (<= 11) Shutdown timeout                   */ {"Shutdown", "timeout"},
  /* (<= 13) Motion sensitivity level           */ {"Motion", "sensitivity"},
  /* (<= 13) Temperature in F and C             */ {"Temperature", "units"},
  /* (<= 13) Advanced idle display mode enabled */ {"Detailed", "idle screen"},
  /* (<= 13) Display rotation mode              */ {"Display", "orientation"},
  /* (<= 13) Boost enabled                      */ {"Boost mode", "enabled"},
  /* (<=  9) Boost temperature                  */ {"Boost", "temp"},
  /* (<= 13) Automatic start mode               */ {"Auto", "start"},
  /* (<= 13) Cooldown blink                     */ {"Cooldown", "blink"},
  /* (<= 16) Temperature calibration enter menu */ {"Calibrate", "temperature?"},
  /* (<= 16) Settings reset command             */ {"Factory", "reset?"},
  /* (<= 16) Calibrate input voltage            */ {"Calibrate", "input voltage?"},
  /* (<= 13) Advanced soldering screen enabled  */ {"Detailed", "solder screen"},
  /* (<= 11) Display Help Text Scroll Speed     */ {"Description","Scroll Speed"},
};

const char* SettingsMenuEntries[4] = {
/*Soldering Menu*/"Soldering\nSettings",
/* Power Saving Menu*/"Sleep\nModes",
/* UI Menu*/"User\nInterface",
/* Advanced Menu*/"Advanced\nOptions", };
const char* SettingsMenuEntriesDescriptions[4] ={
"Soldering settings",
"Power Saving Settings",
"User Interface settings",
"Advanced options"
};
#endif
```

closed 2018-04-09 by Ralim