Ralim/IronOS #233: Translation string length?

The-M1k3y, 2018-03-15:
- [X] Question

Hi, I really like your Firmware and am working on the german translation at the moment and ran into something I did not understand.

For some texts (like UndervoltageString) there is a comment with the length of the strings. Are those mandatory values? Some comments are "Must be <= x" and some only "<=". Some clarification would help.

jonnieZG, 2018-03-15:
It's the same, and means that the text length **must** be less or equal than the specified number.

The-M1k3y, 2018-03-15:
ok, thanks for the clarification.

The-M1k3y, 2018-03-15:
ok, I missed something. Is there a limit on SolderingAdvancedPowerPrompt" ?

Mrkvozrout, 2018-03-18:
@The-M1k3y hi, length of that string is <= 12

Max chars on display for big font is 8, for small font is 16. Here the value takes 3 digits plus percentage sign so 16 - 3 - 1 = 12 ;)

closed 2018-03-23 by Ralim