Ralim/IronOS #235: bug 2.03cr2

SmayK, 2018-03-16:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
hello I come to test version 2.03cr2, I have a problem with the ignition the temperature rises to more than 500 and this place out of service. the official version "2017.5.12 Update: TS100 APP 2.17" no problem

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.03cr2
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :24v

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

SmayK, 2018-03-16:
again, then after reinstalling the official version tests and reinstalling the 2.03cr2 the bug is gone. may be due to the basic management of the fireware provided with the purchase. the update of the latest version of the official followed by yours has worked properly. test to do maybe.

federck, 2018-03-18:
There is also an error in string 534: /* (<= 11) Display Help Text Scroll Speed     */ {"Description "," Vitesse de défilement"}.
The characters exceed the maximum number permitted.

SmayK, 2018-03-18:
yes I have changed since. seeing I got the iron i saw the worry see my link

federck, 2018-03-18:
Pardonne-moi :)

SmayK, 2018-03-18:
pas de problème :)

closed 2018-03-23 by Ralim