Ralim/IronOS #240: Bogus translation LANG_SE?

jvitkauskas, 2018-03-17:
It seems that LANG_SE is a half russian half english translation. There's proper one called LANG_RU. So I suppose LANG_SE can be deleted.

Ralim, 2018-03-17:
The language _se was contributed by a community member. The English is because whenever a new thing is added, English is added to all the languages until translators can come and replace that with the correct text (better to have English than no feature)

When I'm home I'll see if i can tag the original creator of the _se language

jvitkauskas, 2018-03-17:
It seems that Swedish translation was lost in commit 64dbd7d32ec175207e43a62c8249331b19c4764c and then commit 7cc0fd7c84571e8d7c54c17923e21d93ded38638 brought in incorrect one.

Ralim, 2018-03-19:
Ah right, yeah my bad on the 2.x translations import. Ill push a fix for this when im home next.

closed 2018-04-12 by Ralim