Ralim/IronOS #243: Graphics

federck, 2018-03-18:
**I'm submitting a ...**
  - [x] Feature request

 **Do you want to request a *feature* or report a *bug*?**
Insert GIFs instead of static images in the main menu (if possible)

**Steps to reproduce:**
1. Access main menu by pressing the nearest button to the micro-USB port

 **What are you running:**
  - Firmware Version: 2.03
  - PCB Version: 1

 **Other information** 

[image][image][image][image][image] (one time running)
[image][image][image][image][image] (in perpetual running)
[image][image][image][image][image] (revisited better-looking PNGs)

Mrkvozrout, 2018-03-18:
Great effort and interesting request. I just want to say that I personally like the static pictures better, but of course depends on majority or repo owner ;)

JohnEdwa, 2018-03-19:
Animated GIFs as a filetype won't work, but if you combine the frames for each animation to [a sprite sheet](https://i.imgur.com/Q1OQM.png), the animation can be done by drawing a part of the image at a time.

Also, is that supposed to be a [Monkey Wrench](https://images-na.ssl-images-amazon.com/images/I/81uf87nZnrL._SX355_.jpg) or [Calipers](https://traceable.com/media/catalog/product/cache/2/image/9df78eab33525d08d6e5fb8d27136e95/3/4/3415mainweb_6.jpg)?  
Because I've always thought that icon is the wrench, but the animation (and well, calibration/calipers) makes me feel like I've been lied to my whole life.

Ralim, 2018-03-19:
@JohnEdwa I think its meant to be Calipers :smile: 

I like this idea, there has also been calls for custom idle screens to be supported.
I think both of these things are doable, however I would like to first : 

1. Get 2.03 out with bug fixes
2. Look into #222 
3. Evaluate the left over flash space to pack in extra icons to allow for support for animated icons.

Also I feel most people will want the icons to *not* animate to make them easier to see and less confusing if your eyesight is poor. Not too sure on that one though.

JohnEdwa, 2018-03-19:
> Also I feel most people will want the icons to not animate to make them easier to see and less confusing if your eyesight is poor. Not too sure on that one though.

Mmh, one thing that popped to my mind is that they should default to the full icon at the start. With the provided examples, the first few frames of the spanner and sleep animations are almost empty and very similar to each other.

federck, 2018-03-19:
@JohnEdwa I confirm, it's a vernier type caliper, at least in my intent :smile: 

Designing icons was not too big of an effort; I'm open to modify styles as many times as needed. 
However, one-time-running icons would probably be a good compromise between static and animated.

playday3008, 2021-03-08:
And so, I added intermediate frames, and it will even look like animation, only I don't know how to implement it.
https://github.com/playday3008/IronOS/tree/animated-icons
and maybe important info, `TS100_EN.hex` size 98 KB → 99 KB, I guess its probably nothing

P.S. yep, time to write the code that was asked for 3 years ago

Ralim, 2021-03-09:
Hex file increasing is not as much of a concern; if we run out of flash space the CI will fail which will catch this. 
I believe TS80P is the device with the least flash at the moment.

playday3008, 2021-03-09:
> Hex file increasing is not as much of a concern; if we run out of flash space the CI will fail which will catch this.
> I believe TS80P is the device with the least flash at the moment.

Then what's the problem?

Ralim, 2021-03-09:
Check if the changes work for TS80P.
At the time this was shelved we had 0 spare flash space. Since then there was a lot of re-work of languages to increase the available flash space. My previous answer on this was from 2018 and as such is related to the state of things in 2018.

playday3008, 2021-03-09:
Well
`TS80P_EN.hex` size 113 KB → 114 KB

Ralim, 2021-03-09:
What is the object size that is printed out?
Hex size is extremely misleading
Or the binary file size?

If you run the build all script do all languages for all 4 irons build ?

playday3008, 2021-03-09:
> If you run the build all script do all languages for all 4 irons build ?

Yep
[image]

> Or the binary file size?

`TS80P_EN.bin` size 39.8 KB → 40.1 KB

playday3008, 2021-03-09:
gcc `-Os` flag magic

Ralim, 2021-03-09:
I see no reason not to do it then

playday3008, 2021-03-09:
> I see no reason not to do it then

So, I have frames, you have brain, lets go mb

PR #866 (2021-03-15): Animated icons
* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**
(Feature)

Close #243
closed 2021-03-15 by Ralim