Ralim/IronOS #244: TS100 self-resetting constantly or locking up

Doridian, 2018-03-19:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
On the two newest releases (the two 2.03 ones): My TS100 self-resets constantly (every ~ 10 seconds)
On the release before that (2.02): My TS100 locks up after ~ 30 seconds and becomes unresponsive entirely for a bit until it resets
On commit 3ce0cc0d996cd3a59368f040993d66f83b44f12f (which was the most recent one when I got my iron), it works perfectly without any problems whatsoever

This means that https://github.com/Ralim/ts100/commit/aef8f5f5bb358c1531859ab687dff9c975d7e434 (as it is the only commit between the working build and the broken build for my iron) must be the culprit.

* **What is the expected behavior?**
It should not reset or hang

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: As described above
  - PCB Version: PCB2
  - Power Supply (Voltage and Current Rating) : Happens when just in the settings screen off of USB power, or with a 19V PSU that came with the unit.

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

JohnEdwa, 2018-03-19:
Release 2.02 and prior lock up because they don't have support for the second PCB revision with the LIS2DH accelerometer. aef8f5f added support for that.  
If yours freezes on both, there is the remote possibility there is a third revision floating around as well.

Back when @Ralim was debugging this, he released [a test firmware](https://github.com/Ralim/ts100/files/1734510/TS100.zip) to query the accelerometer ID. It should return '02**5** 060' if your unit has the LIS2DH, and '02**9** 060' if it has the original.

Doridian, 2018-03-19:
It shows "060" (photo attached, sorry for the darkness, but otherwise you can't read the display at all)
[image]

Doridian, 2018-03-20:
@Ralim Any ideas? Is my unit's gyro just broken maybe?

Doridian, 2018-03-21:
Ok the replacement arrived and it shows a second number and now stuff seems to work. Is it normal that the orientation (Left/Right) takes a bit to change?

Ralim, 2018-03-21:
Yeah that's normal,  the accelerometers take a little bit to stabilize so there is a slight delay until they kick in. I'm planning to reduce this delay in the full release, but there is a little bit of one so that it doesn't flip randomly while it stabilizes.

Ralim, 2018-03-21:
Oh also, did you try the newest rc1 release? The older stable doesn't work on the newer accelerometer models.

You can check if you hold the rear button on the idle screen it will show PCB: and then a number

Doridian, 2018-03-21:
@Ralim 
Yeah, on the "broken" iron I was trying the newest from git (built myself) and the newest RC. (As you can see above, with your test firmware, it didn't even see any second device on the I2C)
On this iron now I have the latest git installed, too. (Both irons show PCB2)

JohnEdwa, 2018-03-21:
@Ralim  
I'm not sure how you currently do the accelerometer detection, but I guess it wouldn't hurt to add a failsafe for when it doesn't work so the iron doesn't freeze or get stuck in a boot-loop.

Ralim, 2018-03-23:
I have added a lockout for if neither accelerometer is detected to prevent this being horrendous in future. 
@Doridian Could you try the latest RC that I've just uploaded?

Doridian, 2018-03-23:
@Ralim Sadly, since I did RMA the iron, I am no longer in possession of the broken one, but instead one with an actually working accelerometer, so I can't help you there.

Ralim, 2018-03-25:
@Doridian No problem.

closed 2018-03-25 by Ralim