Ralim/IronOS #245: Allow change Temperature steps

ohiyo777, 2018-03-19:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Request a feature

* **What is the current behavior?**
Temperature steps, default is 10 °c step

* **What is the expected behavior?**
Is that possible add an configure that allow change temperature steps by 1~30 °c?

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Add an configure that allow change temperature steps by 1~30 °c

Ralim, 2018-03-23:
Hi,
Can I ask why you are looking for smaller step sizes ?

ohiyo777, 2018-03-23:
@Ralim Yeah~ I was using to cut film, the best temp was around 55 degree, even the temp in lcd does not that consist with tip, but smaller step could help me to calibration tip temp more closer I want.
I have to say, Thank You for Your Hard Work. :)

Ralim, 2018-03-23:
Hi,
Yeah there are a few things that can affect the accuracy of the system, especially when you are using it at such a low temperature, what power supply were you using ?

The system is roughly +-5 C before calibration, but more accurate afterwards. But I don't do any calibration for the bottom half of the curve, so nothing is calibrated below about 250C.

ohiyo777, 2018-03-23:
@Ralim 
Right that’s what I meant. So I had a Digital Thermometer to get real tip temp, and the problem is temp bump too much by default step. :D
Oh,  I use 12v 1.5a  input.

charun-za, 2018-11-16:
Would love this too, 1~10c steps would be great, just more control right? seen as default firmware does support this, why remove it :P?

Ralim, 2018-11-23:
@alexstrassheim "Exact Temperature" ? You are awake that the tip oscillates in temperature due to the nature of it being a soldering iron?

@charun-za I agree finer steps would be nice, I'm going to try and add this but its less of a priority than the new PID and TS80 support. Running low on flash space so I'll have to see how much space it will take up in the flash as well.

Its not a case of "remove it", this firmware had a menu system before theirs did xD Most of the features in their firmware came from here.... I did have temp change steps in the older firmwares (1.x) but everyone I talked to just left it on 10 deg steps since its still finer control than most irons actually give you.

Ralim, 2020-09-14:
This is now done 😄

closed 2020-09-14 by Ralim