Ralim/IronOS #246: v2.03-stable release date?

teejay22, 2018-03-20:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

Just wondering when there will be a stable release of v2.03?

Ralim, 2018-03-20:
When life leaves me enough time to sit down and fix the last outstanding bugs?

teejay22, 2018-03-20:
Didn't mean to sound rude. I just received my ts100 and was wondering if I should stick with the official firmware. 
Are there any major issues with the release candidate?

Ralim, 2018-03-20:
Its okay :smile:, Just its one of the most popular questions about this project.

Depends how happy you are to update the iron, the current stable is quite good, and im aiming to have a new stable soon :)

teejay22, 2018-03-20:
I tried the last stable and had the menu freezing problems after powering on 🙁 I'll keep an eye out for next stable. Cheers!

Ralim, 2018-03-20:
Ah right, your on the new PCB. I would feel free to use the current RC1 release, just note that there is currently a bug with it not being able to be woken from sleep when the screen powers down. Ill be aiming to push out a fix for this shortly though :)

closed 2018-04-05 by Ralim