Ralim/IronOS #248: Sleep issue

CeeLowe, 2018-03-21:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a Bug

* **What is the current behavior?**
When the iron is set to sleep and then shutdown. The sleep appears to work fine and the shutdown powers the unit off. However after the shutdown there is no way to power the unit back up without removing the power cord and reinserting it. No button presses or long presses seem to do anything. Also the shutdown does not always follow the time limit set. i.e. I have it set for 8 minutes sleep, and 30 minutes shutdown. It will sleep just fine, but the screen will go black at a random time after the sleep starts. Sometimes 30 seconds, sometimes 15 minutes. 

Once it goes black there is no way to make the iron work without removing power and starting over. 

* **What is the expected behavior?**
I believe that once it times out completely you should be able to press a button and bring it back up. If that is not the case then making the shutdown time work properly would be acceptable. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x 2.03
  - PCB Version: (1/2) pcb1
  - Power Supply (Voltage and Current Rating) : 120v to 20v ac/dc power supply

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

smntsn, 2018-03-22:
My device has this bug too.

I'm using v2.03-RC2

federck, 2018-03-22:
https://github.com/Ralim/ts100/issues/246#issuecomment-374554618

AMHenriques, 2018-03-22:
Same problem here with v2.03 PCB1

Ralim, 2018-03-22:
I have confirmed this issue. Should be pushing out fix shortly.

commit 6c346750 (2018-03-22): Sleep screen check bug

Should fix #248
closed 2018-03-22 by Ralim