Ralim/IronOS #252: Produce binaries for all languages

jvitkauskas, 2018-03-24:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
I am requesting a feature

* **What is the current behavior?**
Binaries for not all languages are produced in releases tab on GitHub

* **What is the expected behavior?**
Binaries produced for all languages in releases tab on GitHub

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **Other information** 
BG, RU, SK, HUN, BR, LT translations are missing from v2.03-RC3.

jvitkauskas, 2018-03-24:
I don't know how you are producing these builds. Maybe I/we could help you by setting up CI pipeline.

Ralim, 2018-03-24:
Ah good catch, I'll double check the build script.
It will be rewritten after the languages are all packed into one firmware so I'm not going to stress about polishing it too much. 

At the moment it's the `build.sh` in the workspace that is run automatically on my server and I just grab the hex files.

commit c4230d79 (2018-03-25): Update Build Script

Fixes #252
closed 2018-03-25 by Ralim