Ralim/IronOS #254: v2.03 don't start with 12VDC

nimoru, 2018-03-25:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

***Steps to reproduce:***

1. Use factory firmware and 12 VDC - no problem (solder work).
2. Use 2.03-RC3 firmware and USB power - all settings work
3. Use 2.03-RC3 firmware and 12 VDC  - nothing on screen
4. Use 2.03-RC3 firmware and 12 VDC - on press START button (near soldering iron tip) on screen "DFU:3.43"
5. Use 2.03-RC3 firmware and 12 VDC - on press MENU button (near DC connector) - nothing on screen

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.03-RC3
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : 12VDC

* **Other information**

alxsob, 2018-03-25:
V2.03 PCB1
DFU:3.43
Works for me with 12VDC.
Even with the settings 6S.

Ralim, 2018-03-26:
All of my units work fine down to 10V supply input for soldering, and will power down to much lower than that for just changing settings.

Are you able to confirm the current rating of your 12V supply? Can you test another 12V supply ?

nimoru, 2018-03-26:
Try another PC power supply 12V 13Amps - it works on factory firmware and don't work (don't start) on 2.03-RC3.

I try regulated power supply - screen start show in 18.06V DC.

Actually I found 19VDC power supply unit - so 18V is not big problem at all. But I can't use solder from car power (12VDC).

nimoru, 2018-03-26:
Found some intresting: When I set 20VDC power supply solder show "Input: 14.1V"

rako233, 2018-03-27:
I have the same issue. I've used a 12V 4A and a 16V 6A power supply.

hundemann, 2018-03-27:
Same for me.
Neither my 16V DC PS, nor any LIPO (3S,4S) will power up my TS100 up, after I disconnect USB.
The only reaction is the DFU Screen afteroressing A Button. 
Also USB is not correctly recongnized by my WIN10, on the original FW this works fine (I can open the CONFIG)
With USB and DC source (Lipo in that case) connected at the same time, I made it once start, and I could also use functions, like Boost etc.

Ralim, 2018-03-28:
Have any of you tried using the input voltage calibration in the menu?

Can you please turn on the detailed idle screen, and check what the iron is reading your input voltage as?

Where were your irons purchased from?

This firmware will not power up if its reading an input voltage of less than 10V.

@hundemann For the USB, please read the note that is on the main page of this firmware. It does *NOT* impliment the config USB configuration as it has the nice menu onboard instead.

hundemann, 2018-03-28:
@Ralim I did the calibration, and this was always just slightly under the measured voltage from from DC PS or 3S or 4S  LiPo, so it was alwas above 10V. 
And for the USB Config: Yes I know and I love this config in the iron :-). I just thought that not showing USB connected properly could be an issue connected, if not, I'm fine.
I got mine from BG (with the blue case).

Ralim, 2018-03-28:
Ah,
Can you confirm if the voltage is read correctly on the detailed screens?

hundemann, 2018-03-28:
Yes, I can confirm it is read correctly. Before calibration almost correct and also afterwards.

hannesfischer, 2018-03-28:
I've got the same issue. The iron seems to be happy at 20V. The Display flashes on and off at 19.5V and everyting below the display stays off.
With this firmware from Ralim it works just fine. It just hasn't the Fncy submenues.
[TS100Firmware v2 working.zip](https://github.com/Ralim/ts100/files/1854951/TS100.Firmware.zip)

xalius, 2018-03-28:
I have a similar issue with the new TS100 PCB, software works fine with only USB power, automatic display flipping seems to work ok too. If I power ther TS100 from a DC supply (16V LiFePo atm) the software hangs during startup - sometimes the tip is heated indefinitely in this state until it starts to glow :)

nimoru, 2018-03-28:
@Ralim 
I calibrate voltage in menu.
Before calibration display show 14.1V when real voltage is 20V
After calibration menu show correct voltage (from 18.5 to 24V) but in 18VDC shut down as earlier.

Soldering iron bought in this store: https://ru.aliexpress.com/item/2016-NEW-TS100-Pen-type-MINI-Programmable-Smart-Adjustable-Digital-LCD-Electric-soldering-iron-Soldering-station/32652186852.html?spm=a2g0v.10010108.1000016/B.1.41c8e9364EFM5v&isOrigTitle=true

JonnyFPV, 2018-03-28:
Same here, dfu 3.43 showing with power from lipo. With usb it works. And if i plug in usb and DC input i can solder... but thats not so nice :D Voltage reading is ok.

Ralim, 2018-03-29:
Hi,
Can you please try this firmware and confirm if the issues are still present?

I haven't been able to replicate any of these issues on the selection of units I have here at the moment.

[TS100.zip](https://github.com/Ralim/ts100/files/1859731/TS100.zip)

hundemann, 2018-03-29:
Firmware update will not work, it is resulting in an *.ERR on mine

subzer0x, 2018-03-29:
doesn't work here either, still the same dfu 3.43 showing

nimoru, 2018-03-29:
TS100.zip 
new firmware show voltage correct on screen but start at 18VDC.
Below 18VDC screen doesn't work

hannesfischer, 2018-03-29:
@Ralim still doesnt work below 20V ):. Voltage reading is correct.

hundemann, 2018-03-30:
@JohnEdwa Thanks for the hint, managed to flash sucessfully, but no change.
Still showing up also just DFU on DC power.

Ralim, 2018-03-30:
Okay,
I *think* one of my irons is affected by this.
This firmware reverse the change to the code that fixes it for mine.
Can you confirm this one for yours ?
[TS100.zip](https://github.com/Ralim/ts100/files/1862988/TS100.zip)

hannesfischer, 2018-03-30:
@Ralim This one works down to 12 V! :D 
## *THANK YOU :D*

JonnyFPV, 2018-03-30:
for me this is working :-)  coltage need to be calibratet its way off. but then it woorks :-)

hundemann, 2018-03-30:
working on 16V DC

xalius, 2018-03-30:
Hi Ralim, the above firmware works for me, issue seems fixed.

nimoru, 2018-03-30:
It works fine from 10V to 24V DC. 
Thank you!

closed 2018-03-31 by Ralim

SolderSoldier, 2019-02-27:
In my case the issue was caused by cable. **Cable's high resistance** (almost 3R) caused that lowest possible voltage for my TS100 was 13V. After replacement of the cable I can work from even 10.0V !
_Εύρηκα !_