Ralim/IronOS #256: V.2.0.3 PCB2 - off

fishingpro, 2018-03-28:
- Firmware Version: 2.0.3 Mar25 2018
- PCB Version: 2
- Power Supply: 22.2V 5.9A - power supply from laptop.

Soldering iron bought in this store: [https://ru.aliexpress.com/item/Mini-Portable-65W-Electric-Solder-Iron-Digital-LCD-12-24V-Station-Welding-Soldering-Stations-Heating-Element/32580177477.html?spm=a2g0s.9042311.0.0.5Z9Le8](url)

It seems to be not a fake. On the firmware from the manufacturer it works normally.

On the current firmware: 2.0.3 Mar25 2018 there are problems:
1) A few minutes after turning on the soldering iron turns off, it starts to heat up a little in the back and does not move the accelerometer, but the buttons do not help - just distort the cable.
2) If you set SleepTemp 60, the soldering iron stops at 105-110C after going to sleep, and then does not leave it. Further in this place turns off after 2-3 minutes, as written in paragraph 1
3) Disconnected when in version display mode after 2-3 minutes, as described in paragraph 1
4) I tried to put ShutdownTime large (for example 59 min) but the soldering iron is still turned off, as in paragraph 1.
In general, the soldering iron hangs regularly and I have not found any working configuration. Just in case, I reset the current settings, there may be something in them that is conflicting from the "Known problems"

SleepTemp = 150
SleepTime = 6 
SolderingTemp = 350 
cutoutSetting = 0
detailedSoldering = 0
detailedIDLE=1
OrientationMode = 2
sensitivity = 5
voltageDiv = 222
ShutdownTime = 10
boostModeEnabled = 1
BoostTemp = 450
autoStartMode = 0
coolingTempBlink = 1
descriptionScrollSpeed=1

Ralim, 2018-03-31:
Hi,
Please use the issue template when filing an issue, it is provided for a reason.
You are missing the exact steps to re-create the issues. 

Also, I'm not quite sure what your issue is.
1&4 seem to be a misunderstanding between sleep/shutdown and the screen turning off when its idle?
For 2 please test the latest release and confirm if you are still having this issue. This is sometimes an issue with small tips + high supply voltages.

fishingpro, 2018-03-31:
> For 2 please test the latest release and confirm if you are still having this issue. This is sometimes an issue with small tips + high supply voltages.
corrected. Thank you. so thanks for the Russian translation.

> 1&4 seem to be a misunderstanding between sleep/shutdown and the screen turning off when its idle?
Maybe you're right. Explain please - if the soldering iron goes into "Shutdown", should it not wake up while it is moving? Those. that it would be necessary to unplug the wire and re-insert it? If so, I really misunderstood how it should work. But this is not very convenient because:
1) The soldering iron in the "Off" mode continues to bask near the power connector (apparently it is a DC-DC converter).
2) Constantly disconnecting and turning on the connector may damage it over time.

fishingpro, 2018-03-31:
You have added the ability to Off(sleep/shutdown), just noticed this! Thank you!

JohnEdwa, 2018-04-01:
The shutdown with the iron not waking up when moved and requiring a power cycle is known. Issue #223

closed 2018-04-09 by Ralim