Ralim/IronOS #258: Optional sleep settings

fishingpro, 2018-03-29:
I wanted to propose to do in the "Sleep settings" section, the options "Sleep(On/Off)", "Shutdown(On/Off)". What would be if you want to remove care of sleep and shutdown.

jonnieZG, 2018-03-30:
Is there a specific use case where one would need to turn the sleep off? IMHO, sleep option is the main feature why people buy this iron in the first place, because forgetting to turn off the iron is the best way to ruin it. You could always set the sleep time to maximum...

Ralim, 2018-03-30:
I have pushed up a commit to allow turning sleep off because it used to be in the old firmware and there was already support for it in the code. 

*However*, I would really, really push for no-one to turn it off.

fishingpro, 2018-03-30:
Now the firmware is not working properly, leaving the soldering iron goes to sleep before turning the power cord back on. If it were possible to turn off everything related to sleep, it would be possible to use the firmware further.

Ralim, 2018-04-02:
Your above comment is #223. And the original issue has been added to the firmware 😄

closed 2018-04-02 by Ralim