Ralim/IronOS #259: v2.03 custom bootup image not supported

hannesfischer, 2018-03-31:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

I uploaded the .hex file that was created using the TS100 Logo editor and on powerup the Logo is not displayed

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Create a .hex file using the Logo editor
2. upload the .hex file in DFU mode
3. power up the iron.

  - Firmware Version: 2.03
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 18.5V LIPO 20A

Ralim, 2018-03-31:
Hi,
Can you confirm if the logo flashed correctly (ie extension changed to RDY)?

hannesfischer, 2018-03-31:
Yes, i copied the file to the iron after that the window closed, opened
again and the file was renaimed to [filename].RDY

hannesfischer, 2018-03-31:
I also noticed that the font is kinda weird after I uploaded the logo.hex

Ralim, 2018-04-02:
Your logo was corrupted / over-ridden the font table.
It is quite possible your logo file is from the first generation of tools for making the logo files and not the latest ones. Could you try making the logo.hex file again with the newest tools and try again. (Reflash the firmware first to fix the font).

hannesfischer, 2018-04-02:
I tried it again using the TS100.Logo.Editor.exe that was provided in v2.02 and I still get the same. Can you upload your version of the Logo editor?

hannesfischer, 2018-04-02:
Nevermind. Used the wrong Logo file.

closed 2018-04-02 by hannesfischer

Ralim, 2018-04-04:
Thank you for reporting back :) Sorry about that.