Ralim/IronOS #261: Temperature indicator is flickering

rako233, 2018-03-31:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
As soon the set point is reached the temperature indicator is flickering wild with up to more than 20C

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.  Upload the firmware downloaded from issue 254 on the TS100
2.  Set the temperature to 270C
3.  Wait for the TS100 has reached the 270C 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.03 PCB2 30th march
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : 12V 5A

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

wildekek, 2018-03-31:
Confirmed, same issue here with the exact same parameters. I did calibrate the temperate before I tested btw.

Firebie, 2018-04-01:
Same here on PCB Version 1

hannesfischer, 2018-04-01:
Same with mine.
PCB version 2.
Set to 330°C jumps sometimes up to 370°C

martin1454, 2018-04-02:
Same here - jumps +- 30 degrees

Ralim, 2018-04-04:
Thanks for your report, I'm stuck working away from my home system at the moment, but will hopefully have an update here shortly.

magyarl89, 2018-04-04:
Same here.
PCB v2
There is no this problem for your quick solution for PCB v2 [what you posted here](https://github.com/Ralim/ts100/issues/202#issuecomment-369560535)

Ralim, 2018-04-04:
@magyarl89 
Yes, because that firmware didn’t have the adjusted PWM timings. In the 2.03 I changed the PWM timings to bring the values into alignment with what they should be ideally.

Ralim, 2018-04-05:
[TS100.zip](https://github.com/Ralim/ts100/files/1882171/TS100.zip)

Hi, can you test to see if the above firmware improves the performance for you ?

magyarl89, 2018-04-05:
Hello,
Unfortunately not, it's still flickering +/- 10 °C when it adjust the heating output around the setpoint.

Ralim, 2018-04-06:
@magyarl89 
Can I ask what setpoint temp, power supply and tip you are using ?
Originally the issue was very large temperature jumps, and I will work on narrowing down the jumps.

magyarl89, 2018-04-06:
DC laptop power supply 19.5V 4.7A
TS-B2 tip
PCB v2
@ 120°C: ±10°C
@ 300°C: ±5°C flickering

fishingpro, 2018-04-08:
@Ralim + -5 and + -10 are obvious trivia. Now there are no abnormal throws at +50 (and more), which met before. the fix works.

@magyarl89 a slight increase in temperature with the supply of stabilizing impulses - is inevitable. To make the temperature completely stable, it is always necessary to use a constant supply of short PWM pulses (as on the native firmware), but this will inevitably lead to heating of the soldering iron case (as on the native firmware).

2gnospam, 2018-04-11:
I have two irons now.  DC 18.5V.  My first unit with firmware 2.03 downloaded on 3/20/2018 at 7pm EST worked well with stable temps.  New unit with same firmware as first unit behaves the same (stable display with +/- 2 deg F).   

firmware 2.03 downloaded on 4/11/2018 at 7pm EST was flashed to both units and demonstrates large fluctuations on the order of +/- 30 deg F.  

Guessing the PID parameters are different between the two date releases of 2.03.  Would say that the older firmware was slower to ramp up to temp and tended to slow down ramp as it got close to the set point but more stable.  The newer firmware seems to get on temp a little faster especially at the point where the other firmware was "slowing down", but it show much larger indicated temp.  

Gut feeling is the P is higher.  For now I have both using the 3/20 download.

Ralim, 2018-04-12:
The `kd` value was changed after looking at the PID's response after timing changes.
The timings for the ADC readings and the PID were changed to prevent some of the wind up issues that some users had been experiencing, which unfortunately meanas that the PID needs to be retuned again from scratch.

2gnospam, 2018-04-12:
Although I am not familiar with this programming language nor super familiar with PID control (but I did stay at a Holiday Inn last night :) ).  I was wondering if the supplied voltage (and also the board type) has an effect on the PID performance. One could envision the PID parameters being a function of those variables if required.  It does look like the code is "aware" of these parameters, but it would not be "aware" of the tip used (thermal mass, heating element size...).   Hum....  Just some random thoughts trying to help.  Regards Gary

JohnEdwa, 2018-04-12:
> ...if the supplied voltage (and also the board type) has an effect on the PID performance.

Yes, and by quite a lot. At 12V, it takes almost a minute to hit 300C, and at 26V it's around 10 seconds. You need to compensate or you either slow down to a crawl and never reach the goal at 12V, or absolutely massively overshoot at 26V.

Wiht the tips, there simply are no PID values that would work with both the tiny TS-I as well as the massive TS-K. One possibility is to add a few different PID configs based on the thermal mass of the tips and let us choose - to me, I, ILS, C1, KU look small, B2, BC2 and D24 are medium and K and C4 are large, so maybe just "Tip Size: S/M/L" and tune an okay PID for each size.

2gnospam, 2018-04-12:
Thanks for the info.  I did a quick test with all 9 tips today, @19V,  and am getting a feel for the net ramp speed/overshoot for the various tips, etc.  I have not tried the March 22 firmware.  I can see the potential for a FW update to select tip size (and thus adjust PID coefs accordingly ).  

I assume there is no easy way to "data log" the input/output performance of the controlled system during a test.

Ralim, 2018-04-13:
@2gnospam The board type does not affect the performance, however everything else does basically :)
I would love to implement a proper PID that is setup based on input voltage and provide a menu with options for different tips.

If anyone has access to most of these tips I would happily accept PID tuning values for them all ?

2gnospam, 2018-04-13:
I have all of the tips and now only have PCB2.  I can do testing and try to get a feel for what PID parameters work well for each tip.  I need to install the dev system and figure out the compile process in order to help.  I will try this.  In my day job, I do coding, measurement, correlation with models of a system, etc. every day.  This particular system is new to me, etc.

Ralim, 2018-04-13:
@2gnospam 
Tackling PID setup is really the last thing left in terms of pain points in this software. Would happily take any help.

I could also build a version that exposes the PID constants to let you find "tunes" for all the tips or something ?

Ashdinalt, 2018-04-13:
I don't know, what topic to add this:
I wanted to add Ukrainian language, but then checked UK version. And it appeared to be Ukrainian.
It is just a small adjustment - UK means United Kingdom. Ukraine has the label of UA.
Just a small adjustment ))

Ralim, 2018-04-13:
@Ashdinalt Fixed in code, will happen for all future builds I do.

JohnEdwa, 2018-04-13:
@Ralim 
Just throwing my idea at the wall, but:

An advanced setting that swaps 'Hold for Boost mode' for a PID tune mode toggle: back button swaps what PID variable you edit, holding back goes back to regular soldering mode (or just shuts down), front enables +- adjustment for current variable, and holding front temporarily disables the heater for faster testing of the rise time and overshoot.

For display, something like this I guess: [image]

2gnospam, 2018-04-13:
> I could also build a version that exposes the PID constants to let you find "tunes" for all the tips or something ?

That would be great.  I have a variable voltage supply that I also could use.

2gnospam, 2018-04-14:
2.04 RC2 seems to respond well (no bad temp flickering) tried it with the large knife tip, the medium truncated cone tip, and the smallest cone they make.  All trials were using 19V.

2gnospam, 2018-04-14:
The very small tip did fluctuate about 6 deg F, but that is not bad at all.  When I have time I need to make a plug adapter such that I can use my monster/old HP DC power supply.  Where I should be able to test the full range of voltages with the full range of tips.

Ashdinalt, 2018-04-14:
Yeah, I observe pretty much the same behaviour as weav2k4. "During sleep will randomly have screen go blank and not responsive to movement or keypress. If press button closest to tip actually puts it into DFU mode sometimes when screen is blank. I have to power cycle to get iron to go back to normal"
Feeding from 18V battery.

Ralim, 2018-04-14:
[TS100.zip](https://github.com/Ralim/ts100/files/1909974/TS100.zip)

Can you test the above to see if you can reproduce the bug with screen going black?

> Also, if power cycle while tip is hot, it would have been nice to see again the warning message blinking, as on cooldown.

@Ashdinalt Could you spin that out into its own separate issue ?

Ashdinalt, 2018-04-14:
Nope, I don't see the bug now. About 20 minutes in Sleep, still ok. Will test it more.

Ashdinalt, 2018-04-14:
Sleep temp is set to 60C. Bounces around 110-120C.

weav2k4, 2018-04-14:
Confirmed as well the black screen issue is gone.  Still have sleep temp is 50C higher than what is set.

firereverie, 2018-04-14:
Is it just me, or are we talking multiple bugs in this thread? In regards to the screen going to black I will test the beta as well tomorrow when I have some time, as I was experiencing that as well. (Should we open a new thread for this?) As for determining the thermal load to set the pid to correct for rapidly fluctuations, could this be accomplished by implementing a calibration function that would run the tip up to a fixed temp and set the pid based on the amount of time taken to reach said temp? It would need to be run when tips are swapped. It is also possible that we are fighting more than on variable here, even if the mass and composition, and therefore the thermal load of two identical tips are the same we could still have variations in the heating element.

blitzcode, 2018-04-15:
I'm having the same issue. Tip smoothly rises to the set temperature in 1-2C increments, then temperature fluctuates rapdily in a -10 / +30 C range. I tried a few of the recent firmware versions, they all show this problem.

One thing I noticed with my TS100 is that AC leakage of the PSU and grounding of the iron can apparently mess with the temperature measurement. Most cheap switch mode power supplies leak about half mains VAC. This means you can measure something like ~100VAC to mains earth on the tip of the iron. Current is of course <1mA, likely <100uA. Now, that's one of the reasons the TS100 has an earthing point near the handle. I noticed that actually doesn't work, though. If I use a switching PSU with AC leakage that's not mains earth referenced and ground the tip to mains earth with the screw, temperature measurement on the stock firmware goes crazy and jumps around. When using a grounded / mains earth referenced PSU that problem goes away as the tip is connected to ground through a 100nF cap already, and hence there is very little AC potential between the iron tip and mains earth.

No idea if this is relevant here, just wanted to point out that PSU AC leakage and iron grounding seemingly has an impact on temperature sensing with the stock firmware. What kind of grounding and/or AC leakage your PSU has and how/if you use the ground screw on the iron might have an impact on this bug.

Ralim, 2018-04-20:
@firereverie You are correct, and some of this is actually for #223 but it was easier to keep the conversation inline.

@blitzcode I have recently heard about this earthing issue. I would be inclined to think that this is most likely down to the design of the actual iron rather than a software sensitivity.

Glad to hear the latest RC fixes the issue, as this was the version that should have :) 
Ill be making it into a full release shortly.

---
At this point in time, as everyone seems happy with the latest RC I'm going to close this.

closed 2018-04-20 by Ralim

blitzcode, 2018-04-20:
@Ralim Exactly, that was the reason I brought it up. People still seeing issues with fluctuating temperature might be a victim of a poor PSU / TS100 hardware flaw instead of experiencing a bug with your firmware. Thanks for your work & recent fixes! After some further use it seems that the current RC performs very well on my iron.