Ralim/IronOS #263: Fluctuations at the time of the impulse to maintain the temperature.

fishingpro, 2018-04-01:
There is an unpleasant visual artifact. At the time of reaching the set temperature, a temperature stabilization signal (5-35%) is given, at which time the actual temperature of the soldering iron tip rises to 3-8 ° C, but during the first second the readings may be completely different from the real ones, for example at 150 ° C be throws up to 190-220 ° C, then the temperature stabilizes and starts to decrease smoothly from the value of 155-160 ° C. The lower the established temperature, the more significant the error in fluctuations in the first second. Can it be within 1-1.5 seconds after the voltage is applied to not show the real temperature and display the set value (in my example, 150 ° C)? This will avoid these unpleasant visual effects. Of course this is exclusively IMHO :)

Ralim, 2018-04-02:
Can you please use the issue template to break up your issue into the seperate headings to make it easier to answer / reference to when replying?
You have not provided some of the information asked for in the template.

fishingpro, 2018-04-02:
description fixed

Ralim, 2018-04-04:
Aha, thank you. Sorry to ask you to update this, but much easier to confirm this is the same issue as #261. :)

fishingpro, 2018-04-04:
2 themes are created in 1 day. we can assume that the failure lies in the last update.

Ralim, 2018-04-04:
Yeah, the timing on the(  PWM -> PWM ) <-> ADC linking was updated, but on different tips and different units these times can vary. Its a trade off because the tighter the timing the better the performance of the iron, but looser timings are more tolerant of the tip variations. The 2.03 works on all 8 irons I can get access to easily, with 16 tips between them. But there are always outliers.

fishingpro, 2018-04-04:
it seems to me these emissions are associated with the imperfection of the tip. with instant power supply for some time there is an expansion of the heater and at this time there are problems with the measurement, after a while the spiral begins to show more predictably. just a small inductive effect is possible, the coil after all has some kind of inductance.

Ralim, 2018-04-04:
This is less to do with the tip, and the recovery time of the entire system including the op-amp. It is a delay of around 4.5 - 5 milliseconds after the heat is turned off. So as of current the system waits for 5 milliseconds before measurement. I'm going to push this out slightly to see if it helps and upload a test firmware.

fishingpro, 2018-04-04:
If you need pre-test testing, you can send me an assembly to an e-mail: x@x-id.ru

Ralim, 2018-04-05:
[TS100.zip](https://github.com/Ralim/ts100/files/1882171/TS100.zip)

Hi, can you test to see if the above firmware improves the performance for you ?

fishingpro, 2018-04-06:
It helped. The readings became more predictable, the upward shots lost anomalous indicators.

Ralim, 2018-04-09:
Going to close this and just keep  #261

closed 2018-04-09 by Ralim