Ralim/IronOS #267: Add new language

mikko70, 2018-04-12:
Finnish language translation. Please add it.

* **I'm submitting a ...**
  - [ ] Bug report
  - [ ] Feature request
  - [ * ] Translation

* **Other information** 

Finnish language use same alphabets than swedish language

[Translation_FI.zip](https://github.com/Ralim/ts100/files/1901687/Translation_FI.zip)

JohnEdwa, 2018-04-13:
You are overflowing many of the maximum string lenghts in SettingsShortNames, which is actually exactly the reason I gave up on my own translation, as I simply couldn't find anything that fit while making sense.  
 
I'm also quite perplexed why you use "Kalibroidaan" and "Palautetaan" instead of "Kalibroi" and "Palauta" and so on in the descriptions. It just sounds very odd to me.

Anyhow, here's my translation suggestions just edited on top of yours (I left them as comments for easier comparsion.) The SettingsShortNames are still too long though:

```C
#ifdef LANG_FI

const char* SettingsDescriptions[17] = {
  // These are all the help text for all the settings.
  // No requirements on spacing or length.
  /* Power source (DC or batt)          */ "Käytettävä virtalähde. Asettaa katkaisujänniteen. <DC 10V, 3S=9.9V, 4S=13.2V, 5S=16.5V, 6S=19.8V>",
  /* Sleep temperature                  */ "Lepotilan lämpötila <C>", //"Lepolämpötila <C>",
  /* Sleep timeout                      */ "Lepotilan viive <minuuttia/sekuntia>",//"Lepoviive <minuuttia/sekuntia>",
  /* Shutdown timeout                   */ "Sammutuksen aikaviive <minuuttia>",
  /* Motion sensitivity level           */ "Liikkeentunnistuksen herkkyys <0=pois, 1=epäherkin, 9=herkin>", //"Liikeherkkyys <0=pois, 1=epäherkin, 9=herkin>",
  /* Temperature in F and C             */ "Lämpötilan yksikkö <C=celsius, F=fahrenheit>",
  /* Advanced idle display mode enabled */ "Näytä yksityiskohtaiset tiedot lepotilassa", //"Näyttää tarkemmat tiedot leponäytössä",
  /* Display rotation mode              */ "Näytön kierto <A=automaattinen O=oikeakätinen V=vasenkätinen>", //"Näytön suunta <A=automaattinen O=oikeakätinen V=vasenkätinen>",
  /* Boost enabled                      */ "Etupainikeella siirrytään juotettaessa tehostustilaan ",
  /* Boost temperature                  */ "Tehostustilan lämpötila", //"Tehostuksen lämpötila kun käytetään tehostustilaa",
  /* Automatic start mode               */ "Aloittaa lämmityksen automaattisesti kun virrat kytketään. T=juotostila, S=Lepotila, F=Ei käytössä", //"Lämmitys käynnistyy automaattisesti kun virrat kytketään. T=juottamiseen, S=lepotilaan, F=ei käytössä",
  /* Cooldown blink                     */ "Vilkutetaan näytössä lämpötilaa kun juotoskärki jäähtyy, mutta on yhä kuuma",
  /* Temperature calibration enter menu */ "Kalibroi kärjen lämpötilaero", //"Kalibroidaan juotoskärjen lämpötilaero",
  /* Settings reset command             */ "Palauta kaikki asetukset oletusarvoihin" //"Palautetaan kaikki asetukset alkuperäisiksi",
  /* Calibrate input voltage            */ "Tulojännitten kalibrointi (VIN). Painikkeilla säädetään ja pitkään painamalla poistutaan",
  /* Advanced soldering screen enabled  */ "Näytä yksityiskohtaiset tiedot juotostilassa", //"Näytetään tarkemmat tiedot juottamisen aikana",
  /* Description Scroll Speed           */ "Näiden ohjetekstien vieritysnopeus" ,//"Tekstin vieritysnopeus näytöllä",
};

const char* SettingsCalibrationWarning = "Varmista että kärki on huoneenlämpöinen ennen jatkamista!"; //"Älä jatka ennen kuin kärki on jäähtynyt!";
const char* SettingsResetWarning = "Haluatko varmasti palauttaa oletusarvot?"; //"Haluatko palauttaa oletusarvot?";
const char* UVLOWarningString = "DC LOW";               // <=8 chars
const char* UndervoltageString = "Alijännite";        // <=16 chars
const char* InputVoltageString = "Jännite: ";           // <=11 chars, preferably end with a space
const char* WarningTipTempString = "Lämpötila: ";        // <=12 chars, preferably end with a space
const char* BadTipString = "VIKATILA";                   // <=8 chars
const char* SleepingSimpleString = "Zzz";              // Must be <= 4 chars
const char* SleepingAdvancedString = "Lepotila...";     // <=16 chars
const char* WarningSimpleString = "HOT";               // Must be <= 4 chars
const char* WarningAdvancedString = " ! KÄRKI KUUMA !"; //"KUUMA KÄRKI";  // <=16 chars
const char* SleepingTipAdvancedString = "Kärki:";         // <=6 chars
const char* IdleTipString = "Kärki:";                     // IdleTipString+IdleSetString <= 10
const char* IdleSetString = " Asetus:";                    // preferably start with a space; IdleTipString+IdleSetString <= 10
const char* TipDisconnectedString = "KÄRKI ON IRTI"; // <=16 chars
const char* SolderingAdvancedPowerPrompt = "Teho: ";   // <=12 chars
const char* OffString ="OFF";                           // 3 chars max

const char SettingRightChar = 'O';
const char SettingLeftChar = 'V';
const char SettingAutoChar = 'A';

const char SettingFastChar = 'N';   //'F';
const char SettingSlowChar = 'H';   //'S';

const enum ShortNameType SettingsShortNameType = SHORT_NAME_DOUBLE_LINE;
const char* SettingsShortNames[17][2] = {
  /* (<= 11) Power source (DC or batt)          */ {"Virtalähde", "DC"}, 
  /* (<=  9) Sleep temperature                  */ {"Lepotilan", "lämpötila"},
  /* (<=  9) Sleep timeout                      */ {"Lepotilan", "viive"},
  /* (<= 10) Shutdown timeout                   */ {"Sammutuksen", "viive"}, // Too long
  /* (<= 13) Motion sensitivity level           */ {"Liikkeen", "herkkyys"},
  /* (<= 13) Temperature in F and C             */ {"Lämpötilan", "yksikkö"},
  /* (<= 13) Advanced idle display mode enabled */ {"Tarkemmat tiedot", "lepotilassa"}, // Too long
  /* (<= 13) Display rotation mode              */ {"Näytön", "esityssuunta"},
  /* (<= 13) Boost enabled                      */ {"Tehostus", "käytössä"},
[... cut ...]
// SettingsMenuEntries lengths <= 13 per line (\n starts second line)
const char* SettingsMenuEntries[4] = {
  /* Soldering Menu    */ "Juotos-\nasetukset" ,//"Juotos\nasetukset",
  /* Power Saving Menu */ "Lepotilan\nasetukset", // "Lepo\ntila",
  /* UI Menu           */ "Käyttö-\nliittymä", //"Käyttö\nliittymä",
  /* Advanced Menu     */ "Lisäasetukset", // "Lisä\nasetukset",
};

const char* SettingsMenuEntriesDescriptions[4] = {
  "Juotosasetukset",
  "Virransäästöasetukset",
  "Käyttöliittymän asetukset",
  "Lisäasetukset"
};
#endif
```

JohnEdwa, 2018-04-13:
Edited it some more and now everything should at least fit the allocated length limits, even if the translations aren't as perfect as they should be.

Now I'm off to bed, it's 4:30 in the morning here.

```C
#ifdef LANG_FI

const char* SettingsDescriptions[17] = {
  // These are all the help text for all the settings.
  // No requirements on spacing or length.
  /* Power source (DC or batt)          */ "Käytettävä virtalähde. Asettaa katkaisujänniteen. <DC 10V, 3S=9.9V, 4S=13.2V, 5S=16.5V, 6S=19.8V>",
  /* Sleep temperature                  */ "Lepotilan lämpötila. <C>", //"Lepolämpötila <C>",
  /* Sleep timeout                      */ "Lepotilan viive. <minuuttia/sekuntia>",//"Lepoviive <minuuttia/sekuntia>",
  /* Shutdown timeout                   */ "Automaattisen sammutuksen aikaviive. <minuuttia>",
  /* Motion sensitivity level           */ "Liikkeentunnistuksen herkkyys. <0=pois, 1=epäherkin, 9=herkin>", //"Liikeherkkyys <0=pois, 1=epäherkin, 9=herkin>",
  /* Temperature in F and C             */ "Lämpötilan yksikkö. <C=celsius, F=fahrenheit>",
  /* Advanced idle display mode enabled */ "Näyttää yksityiskohtaisemmat tiedot lepotilassa.", //"Näyttää tarkemmat tiedot leponäytössä",
  /* Display rotation mode              */ "Näytön kierto. <A=automaattinen O=oikeakätinen V=vasenkätinen>", //"Näytön suunta <A=automaattinen O=oikeakätinen V=vasenkätinen>",
  /* Boost enabled                      */ "Etupainikeella siirrytään juotettaessa tehostustilaan.",
  /* Boost temperature                  */ "Tehostustilan lämpötila.", //"Tehostuksen lämpötila kun käytetään tehostustilaa",
  /* Automatic start mode               */ "Käynnistää virrat kytkettäessä juotostilan automaattisesti. T=juotostila, S=Lepotila, F=Ei käytössä", //"Lämmitys käynnistyy automaattisesti kun virrat kytketään. T=juottamiseen, S=lepotilaan, F=ei käytössä",
  /* Cooldown blink                     */ "Vilkuttaa jäähtyessä juotoskärjen lämpötilaa sen ollessa vielä vaarallisen kuuma.",//"Vilkutetaan näytössä lämpötilaa kun juotoskärki jäähtyy, mutta on yhä kuuma",
  /* Temperature calibration enter menu */ "Kalibroi kärjen lämpötilaeron.", //"Kalibroidaan juotoskärjen lämpötilaero",
  /* Settings reset command             */ "Palauta kaikki asetukset oletusarvoihin.", //"Palautetaan kaikki asetukset alkuperäisiksi",
  /* Calibrate input voltage            */ "Tulojännitten kalibrointi (VIN). Painikkeilla säädetään ja pitkään painamalla poistutaan.",
  /* Advanced soldering screen enabled  */ "Näyttää yksityiskohtaisemmat tiedot juotostilassa.", //"Näytetään tarkemmat tiedot juottamisen aikana",
  /* Description Scroll Speed           */ "Näiden selitetekstien vieritysnopeus." ,//"Tekstin vieritysnopeus näytöllä",
};

const char* SettingsCalibrationWarning = "Varmista että kärki on huoneenlämpöinen ennen jatkamista!"; //"Älä jatka ennen kuin kärki on jäähtynyt!";
const char* SettingsResetWarning = "Haluatko varmasti palauttaa oletusarvot?"; //"Haluatko palauttaa oletusarvot?";
const char* UVLOWarningString = "DC LOW";               // <=8 chars
const char* UndervoltageString = "Alijännite";        // <=16 chars
const char* InputVoltageString = "Jännite: ";           // <=11 chars, preferably end with a space
const char* WarningTipTempString = "Lämpötila: ";        // <=12 chars, preferably end with a space
const char* BadTipString = "VIKATILA";                   // <=8 chars
const char* SleepingSimpleString = "Zzz";              // Must be <= 4 chars
const char* SleepingAdvancedString = "Lepotila...";     // <=16 chars
const char* WarningSimpleString = "HOT";               // Must be <= 4 chars
const char* WarningAdvancedString = " ! KÄRKI KUUMA !"; //"KUUMA KÄRKI";  // <=16 chars
const char* SleepingTipAdvancedString = "Kärki:";         // <=6 chars
const char* IdleTipString = "Kärki:";                     // IdleTipString+IdleSetString <= 10
const char* IdleSetString = " Asetus:";                    // preferably start with a space; IdleTipString+IdleSetString <= 10
const char* TipDisconnectedString = "KÄRKI ON IRTI"; // <=16 chars
const char* SolderingAdvancedPowerPrompt = "Teho: ";   // <=12 chars
const char* OffString ="OFF";                           // 3 chars max

const char SettingRightChar = 'O';
const char SettingLeftChar = 'V';
const char SettingAutoChar = 'A';

const char SettingFastChar = 'N';   //'F';
const char SettingSlowChar = 'H';   //'S';

const enum ShortNameType SettingsShortNameType = SHORT_NAME_DOUBLE_LINE;
const char* SettingsShortNames[17][2] = {
  /* (<= 11) Power source (DC or batt)          */ {"Virtalähde", "DC"}, 
  /* (<=  9) Sleep temperature                  */ {"Lepotilan", "lämpötila"},
  /* (<=  9) Sleep timeout                      */ {"Lepotilan", "viive"},
  /* (<= 10) Shutdown timeout                   */ {"Sammumisen", "viive"}, //{"Sammutuksen", "viive"},
  /* (<= 13) Motion sensitivity level           */ {"Liikkeen", "herkkyys"},
  /* (<= 13) Temperature in F and C             */ {"Lämpötilan", "yksikkö"},
  /* (<= 13) Advanced idle display mode enabled */ {"Tarkempi", "leponäyttö"},//{"Tarkemmat tiedot", "lepotilassa"},
  /* (<= 13) Display rotation mode              */ {"Näytön", "kierto"},//{"Näytön", "esityssuunta"},
  /* (<= 13) Boost enabled                      */ {"Tehostus", "käytössä"},
  /* (<=  9) Boost temperature                  */ {"Tehostus-", "lämpötila"}, //{"Tehostus", "lämpötila"},
[... cut ...]
// SettingsMenuEntries lengths <= 13 per line (\n starts second line)
const char* SettingsMenuEntries[4] = {
  /* Soldering Menu    */ "Juotos-\nasetukset" ,//"Juotos\nasetukset",
  /* Power Saving Menu */ "Lepotilan\nasetukset", // "Lepo\ntila",
  /* UI Menu           */ "Käyttö-\nliittymä", //"Käyttö\nliittymä",
  /* Advanced Menu     */ "Lisä-\nasetukset", // "Lisä\nasetukset",
};

const char* SettingsMenuEntriesDescriptions[4] = {
  "Juotosasetukset",
  "Virransäästöasetukset",
  "Käyttöliittymän asetukset",
  "Lisäasetukset"
};
#endif
```

mikko70, 2018-04-13:
I'm also quite perplexed why you use "Kalibroidaanko" and "Palautetaanko" instead of "Kalibroi" and "Palauta" and so on in the descriptions. It just sounds very odd to me.

because ext is a question.  if  text is without asking , then you are right.
Like: are you drive? or are you driving?

mikko70, 2018-04-13:
/* (<= 10) Shutdown timeout                   */ {"Sammutuksen", "viive"}, // Too long
/* (<= 13) Advanced idle display mode enabled */ {"Tarkemmat tiedot", "lepotilassa"}, // Too long
 /* (<= 13) Automatic start mode               */ {"Automattinen", "käynnistyminen"}, // Too long
tö"},
 /* (<= 13) Advanced soldering screen enabled  */ {"Monipuolisempi", "juotosnäyttö"}, // Too long
 /* (<= 11) Display Help Text Scroll Speed     */ {"Selitteiden","näyttönopeus"}, // Too long

/* (<= 10) Shutdown timeout                   */ {"Sammutus", "viive"},
/* (<= 13) Advanced idle display mode enabled */ {"Tiedot", "lepotilassa"},
 /* (<= 13) Automatic start mode               */ {"Autom.", "käynnistyminen"},
tö"},
 /* (<= 13) Advanced soldering screen enabled  */ {"Tarkempi", "juotosnäyttö"},
 /* (<= 11) Display Help Text Scroll Speed     */ {"Tietojen","näyttönopeus"},

closed 2018-04-27 by Ralim