Ralim/IronOS #268: Custom Battery Levels and Cut-Offs

firereverie, 2018-04-12:
Interested in using alternate battery sources like the Rugged Geek portable car jump-starter (Li-Po power bank with 19v DC out for charging laptops) or Dewalt 20v MAX Li-Po battery packs (they use 10 LG HB4's but have no onboard cut-off as Dewalt opted to put that circuit in the tools) and it would be fantastic to have a way to set custom Min/Max to get a level on the TS100 and protect the packs.

Ralim, 2018-04-12:
Hi,

There has been discussion on this before, and to cut a really long story short neither of these need extra options really. 

The 19V out port can just be used with DC mode since it'd a regulated output you want be able to judge battery SOC from its voltage.

For the 20V drill battery, this is just a 5 Cell battery pack, so using the 5S option should work well.

:)

firereverie, 2018-04-12:
Good point on the Dewalt pack, I hadn't realized that they are 10 cells wired in a two parallel by 5 series config till I went and took a closer look at the internals (working off of teardown pics online that I had misinterpreted the first go). As for the power pack, putting aside that it should have its own cut-off, it would still be nice to be able to set a custom cut-off as a means of setting a reserve limit on the pack so for example there would still be enough juice to charge a phone or jump a car, but I can appreciate that this is a niche and not cost effective related to feature demand. Thank you for your feed back and all your work on this project!

JohnEdwa, 2018-04-12:
> As for the power pack,

It's a powerbank, it outputs a boosted, stable 19V until it runs out of juice and suddenly doesn't output anything anymore. Rght at the very end when you have [less than 10% left](http://www.ibt-power.com/Battery_packs/Li_Ion/Li_Ion_DiscGph.JPG) the voltage might sag just enough to be able to detect it, but there's nothing useful you can do at that point.

closed 2018-04-12 by Ralim