Ralim/IronOS #269: 2.04 Release Candidate

Ashdinalt, 2018-04-13:
* **I'm submitting a ...**
  - [ x ] Bug report

* **What is the current behavior?**
When iron reaches its pre-set soldering temp, it starts jerking temperature on display. 
For example, temp is set to 320C. When it reaches it, the temp readings start jumping up and down around 320C - 326,323,316,318,320,322,327,326,319.... And so on. And the tip heating/not heating icon jumps back and forth also.
Since the temp readings change very quickly, it looks like a bug. 

* **What is the expected behavior?**
It suppose to have some histeresis, when showing working temp.

* **Video of problem if hard to reproduce***
https://youtu.be/8CivkRrJchI

Ralim, 2018-04-13:
That's not a bug, that's the actual temperature inside the tip. The temperature is read at 33Hz, and the jumps look completely valid for most tips.

Most soldering irons normally hide this by lying about the actual temperature.

In the past most people have preferred the screen to show the actual temperature rather than lie about whats actually being measured. One of the fastest complaints I've received was when I implemented this as well, which is why it is not currently done.

Ashdinalt, 2018-04-13:
Yeah, I understand, that it shows the real temp, but it looks so annoying, and keeps dragging attention as if something's wrong. I don't think that in working-soldering mode it suppose to show real temp faster than 1Hz. I don't believe any human can read temp at 33Hz )))

Ashdinalt, 2018-04-13:
https://youtu.be/Xv8fuYdTtdI - that's how it looks in stock firmware.
When it is in working temp and soldering mode, it shouldn't show so accurate as 33Hz. 
Please, make it look smooth. Like, it reaches preset temp and stays so, as if it's holding the temp.
May be its a bit of lie, but looks and feels much better.
PayPal'd you some )). Hope, you could have made it smoother ))

Ashdinalt, 2018-04-13:
Re: [Ralim/ts100] 2.04 RC Temp Jerking (video) (#269)

Hi

Man! It's WAY-WAY better now!!!
Now I like it! That's how I believe looks much better now!

https://youtu.be/QS5Fl-web1k

The last question for now - how to get this version in Russian? ))))

=======::::::::::::::::========
Wishing you the very best,
Ash@26.ru
=======::::::::::::::::========

-----------------------------
Original message 13 апреля 2018 г., 1:20:27:

TS100.zip
Can you test this version for your unit and see how it goes ?
—

-----------------------------

2gnospam, 2018-04-13:
Sorry guys.  It is late for me and I need some sleep.  In my opinion, at this point in the night, is that the actual temp of the tip can not fluctuate "fast".  The sensor can if it is close to the heater, and not representing the actual tip temp.  Thus it is probably good if we "low pass filter" the readout.  Not sure how the temp reading being "fast" impacts the PID adjustment.  Regards, Gary

Ashdinalt, 2018-04-13:
"low pass filter" - good explanation. That is what I was trying to explain )

Ralim, 2018-04-13:
The filter used to be a 64 sample low pass filter already that you were seeing before my above changes.
The temp sensor is actually a thermoelectric junction as part of the heater coil.

This means it tracks the heating coil temperature, not the temperature of the outside of the iron which we cannot sense.

I have changed the filter type to improve response to help dampen out the readings.

Also did some minor changes to synchronization between the PID loop which helped improve its stability as well.

The catch to this is that the PID settings work really well for the common tips, but it oscillates badly for the smaller tips with less thermal mass.

I'm going to add an adjustment for different tip types.

@Ashdinalt Use 2.04 RC2 😁

2gnospam, 2018-04-13:
Cool.  I hope, at some point, I can get up to speed with this system and actually help.  I have this gut feeling that the sensor needs to have a low pass filter applied before going to the PID (meaning not just for display purposes).  Not sure.  Let me put the brain into it tomorrow.

Ashdinalt, 2018-04-13:
Thank you again!!! Fastest response I've ever seen )))

closed 2018-04-13 by Ashdinalt

Ralim, 2018-04-13:
@2gnospam I agree it would be more logical, however I just have had bad performance when adding in any sort of filtering before the PID. (PID is now lockstep to PWM at 33Hz)

@Ashdinalt You just caught me when I was already looking into it because of #261 😁 
I'm going to close this issue and leave 261 to continue the discussion