Ralim/IronOS #270: Make + and - buttons not switch places when the screen is rotated

simas1017, 2018-04-17:
* **I'm submitting a ...**

  - [ ] Feature request

* **What is the current behavior?**
I have auto screen orientation enabled. Sometimes I want to adjust the temperature quickly and It annoys me that the + and - button functions invert when the screen changes orientation. That seems counter-intuitive.

* **What is the expected behavior?**
I'd like to choose what kind of +,- button behavior I like in the settings. (ex. Normal, Inverted, Auto)
(for me normal is A+ and B-)

* **What is the motivation / use case for changing the behavior?**
I think that I'm not the only one that has this problem

* **What are you running:**

  - Firmware Version: V2.04 RC2
  - PCB Version: 1

JohnEdwa, 2018-04-17:
Heh.

It used to just show "< 350 >" before which confused the heck out of me and I opened issue #213 asking this same thing. A middle ground was found where they still swap but now say + and - on them.  

The idea behind it is that it's a list (or a dial) of temperatures: 330 - 340 - 350 - 360 etc, where 'increment' is always to the right, and 'decrement' to the left. So while they switch physical buttons, the actual direction stays the same when you flip the iron. It basically boils down to this:
> You look at it as being "Left/Right" which changes depending on the orientation of the iron, while I see it as being "Up/Down" where up is always towards the tip.

I feel mirroring the Display Orientation and having "Button Orientation Auto/Left/Right/Inverted" would be the most logical - either it automatically rotates with the screen (flipped or not), or it is always locked to the one it is in Left or Right orientation.

federck, 2018-04-20:
Hi there,
I think an option showing "buttons orientation" (normal/inverted) in soldering menu would be good as well.

JohnEdwa, 2018-04-20:
@Ralim Hey, I'm just throwing ideas at you, it's your choice if you decide to catch them or not ;)

LarsSimonsen, 2018-06-22:
I personally don't care if it's configurable, I can adapt to the UI. The important thing is that the UI never ever changes [modes](https://en.wikipedia.org/wiki/Mode_(computer_interface)) unless the user actively and explicitly tells it to.

simas1017, 2018-06-22:
@LarsSimonsen I wouldn't care too if the button placement was consistent. Buttons change places when the screen is rotated. I'am using auto-rotation and I have to double check which button to press everytime I want to change the temperature. It is a bit annoying.

commit f7165781 (2018-07-29): Stop adjust buttons auto-rotating

Closes #270
closed 2018-07-29 by Ralim