Ralim/IronOS #274: Sleep mode temperature regulation faulty / blank screen

blitzcode, 2018-04-24:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

  - Firmware Version: V2.04 Release Candidate 2
  - PCB Version: Unknown
  - Power Supply (Voltage and Current Rating) : 20V 3.5A

Sleep mode doesn't work properly.

- I left the sleep mode temperature at the default 120C, but the iron never actually drops there. The temperature goes down to under 200C, and then bounces back up a bit before going down again. So it'll drop do 190C, then quickly rise to 199C, only to go down to 190C again. Over time it'll slowly lose temperature and eventually it goes down to somewhere like 160-170C.

- After a few minutes, the screen just goes blank. The shut-off feature has been either set to 60min or disabled.

- During normal operation there seem to be no issues with temperature regulation or the screen going blank. I did a few hours of soldering work with this iron & firmware version and did not experience any issue like this at 320C.

blitzcode, 2018-04-27:
Sorry, I should've been more precise in my bug report.

I initially noted this issue because the TS100 seemed to turn itself off when I left it in the stand for a while. I always had set the shutdown time to a large value or disabled, so I was a bit confused. Moving the iron or pressing buttons did not bring the display / heater back to life.

I did some experiments to see if something funky was going on when the iron goes into sleep. That's when I noticed the control loop issue when cooling down to idle temperature. I repeated the experiment ~5-10x and had the control loop issue every time, about half of the time the screen would also eventually turn off.

In this screen-off state the heater was also off. I did not measure power consumption, but here's why I assume this. I use the TS100 with a laptop power brick behind a switchable outlet. When I encountered the blank screen issue I turned the outlet off & on, but that would not power cycle the iron. The remaining energy in the power brick was enough to prevent the iron from power cycling. With both heater & screen turned off, the power consumption must've been very low.

As another data point, there's a user on EEVBlog reporting the same problem, reverting back to 2.02 fixed it for him.

JohnEdwa, 2018-04-28:
The "goes to sleep and doesn't wake up" is a known issue ( #223 ) that popped up in 2.03 after Ralim added support for the new accelerometer used in the version 2 PCB. Pretty sure it should be fixed in the final 2.04 build.

Hopefully at least, can't know without long time testing which I'm currently doing :)

blitzcode, 2018-04-28:
Ah, OK, the title of the issue did not sound like the issue I was having at all, so I did not look into it. But this describes pretty much exactly what I'm seeing:

> Sorry Ben, RC3 is still locking up in the same way, times out to blank display then can't be roused after a while.
> I've also noticed that the Sleep temp bug is back - or at least a display bug. Sleep is set at 180'C but display hovers between about 250 and 230 varying rapidly over a couple of seconds so I don't think the actual tip temperature can be tracking that fast. Tip BC2 supply 24V.

So this should've been fixed in the just released V2.04?

Ralim, 2018-04-28:
Yeah sometimes the titles make it a pain to figure out.

Give the new release a go, I don't think it's perfectly fixed, but should be less commonly occurring. I need to do some work to shrink the ram footprint of the system at this point I think.

blitzcode, 2018-04-28:
I just gave it a shot. With 2.04 the temperature drop is no longer erratic, but it slowly converges to 143C (still 120C default settings). The temperature drops very slowly towards the end and obviously never cools down to actual 120C. I tried this twice so far, the second time the TS100 froze with the screen & heater running after ~5min. I checked the info screen and I have PCB v2 and had the TS-BC2 tip in the iron when I ran the test.

JohnEdwa, 2018-04-28:
@Ralim 
Sadly I have to report that the sleep-issue still isn't fixed in 2.04, stopped responding after the night :/

Kulturnilpferd, 2018-05-01:
Thats very sad same issue here, beside that its a beautiful firmware ^^

Ralim, 2018-05-03:
@freetushkan this has been reported in #223.
Please keep discussions there. 
It's an issue that takes time that I don't have at the moment to diagnose.

closed 2018-05-03 by Ralim

freetushkan, 2018-05-07:
Also I noticed, that this issue exist on simple USB-connection too, not only while soldering or idle on external power. May be it'll be useful. I hope.