Ralim/IronOS #279: menu unresponsive pcb1

notwired, 2018-04-29:
Just received (april 2018) TS100 from banggood and is version 1 pcb. Put the test fw and is reporting 029 060 (also opened and looks like pcb1). **Device freezes after few minutes.** Tip temp goes down. Tried on DC. Also connected only microusb and also froze after few minutes. Put the original firmware and works just fine, no freeze after half a day. Maybe the manufacturing/parts quality has gone bad. 
Also with your fw found some strange behavior - after wake-up, the working temp was set much higher like 520c, the idle temp was also higher than set. But this was one time, i couldn`t replicate this beahavior.

Ralim, 2018-04-29:
Hi,

Please use the template provided in future. It's there for a reason. 

There is currently an ongoing investigation into the freezing problem in other issues, where it occurs easily on some units and really slowly on others. This appears to be memory allocation, and I just haven't had time to keep chasing it at this point in time. As you are using pcb 1, try the 2.02 and 2.03 releases.

The 520C is because of filter catchup and is normal and will be fixed as time permits.

:) I'm working on it, but as this is not a paying job time is always the hard part :)

IAmOrion, 2018-05-04:
Fantastic work with this firmware! Just adding a "+1" to this.  My TS100 arrived the other day, after some research I found your work, and 'flashed' this firmware, which is very nice btw.  However, on 2 different occasions now, it has frozen and needed rebooting.  I have the PCB2 according to your version screen.  There was no rhyme nor reason, it just happened.  I tried to recreate the freeze but couldn't.

Ralim, 2018-05-07:
Hi,
Can you test if 2.04.1 has fixed this issue for you ?

notwired, 2018-05-07:
Hi,
after some tests (30minutes or so) no freezes ! But some pixels on bottom right column are flickering pretty fast when is text or graphics in that area. 
The B button function tested fine.
Also new data (pressing A) on FW menu looks interesting; can you please document the info showed ?

Congrats for finding the solution, thank you !

blitzcode, 2018-05-08:
Just to add another datapoint, it seems so far that 2.04.1 has fixed the freezes for me as well. I'll report back if I encounter another display or crash issue. Flashing pixels are a small price to pay ;-)

Ralim, 2018-05-10:
@notwired 
A brief documentation for the extra menu data : 
*Page one is the compile time data
* Heap : Free uint32_t words in the rtos heap
* HWMG : Free uint32_t words left in the allocated stack for the GUI
* HWMP : Free uint32_t words left in the allocated stack for the PID control
* HWMM : Free uint32_t words left in the allocated stack for the Movement detection
* Time: Seconds since boot
* Move: Timestamp of last movement detection (Seconds)

closed 2018-05-10 by Ralim

JohnEdwa, 2018-05-10:
Time could use with more digits as right now it can only show up to 99999 seconds or ~28 hours, so I don't *for sure* know if the 26015 means it rebooted on its own during the night or if it has stayed on for the whole ~62 hours so far.

And just in case you happen to have any use for these:
```
V2.05 PCB1 
May 7 2018
Heap:   01416
HWMG:   00624
HWMP:   00478
HWMM:   00424
Time:   26015
Move:   26001
```