Ralim/IronOS #283: Long press "B" while soldering shows firmware version when in reality you get in sleep mode

GnFlala, 2018-05-02:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Feature request

* **What is the current behavior?**
Long press the "B" button while in soldering mode to see firmware version and have sleep mode kick in manually

* **What is the expected behavior?**
That is the expected behavior but do we have to see the firmware version?
My suggestion is to move the firmware version screen in the menu as a menu item or a submenu item and when the user is in soldering mode and long press the "B" button he will see something like "SLEEP MODE" on screen

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Be in Soldering mode
2. Long Press "B" button
3. firmware version screen kicks in when in reality you are in sleep mode

* **What is the motivation / use case for changing the behavior?**
User is able to see on screen what is really happening, in this case gets in sleep mode.

* **What are you running:**

  - Firmware Version: 2.04
  - PCB Version: (2)
  - Power Supply (21v 40A) :

* **Other information**

Cpt-Huck, 2018-05-03:
(y) want to see this in the future! Hate pressing Button A + B to go to sleep mode (voll down completely after soldering operations are done)

Mrkvozrout, 2018-05-03:
For me when in soldering mode, `Long Press B` quits soldering and return to welcome screen like it should, no firmware version screen or sleep mode.

- FW: latest snapshot
- PCB: v1

provalinf, 2018-05-03:
Same problem with my soldering iron 
- PCB: V2
- FW: v2.04 release

Thanks for your job :)

Ralim, 2018-05-06:
This was fixed in a recent patch, and is already dealt with (as @Mrkvozrout mentioned).
This will be fixed in any releases coming out.

closed 2018-05-06 by Ralim