Ralim/IronOS #285: Quick change of temp

Cpt-Huck, 2018-05-03:
Ability to hold Button A / B in soldering screen to quickly change temperature of the tip.

Cpt-Huck, 2018-05-03:
I'm submitting a ... 
Feature request

Do you want to request a feature or report a bug?
▪️ Feature 

What is the current behavior?
▪️ While soldering, pressing and holding Button A (or B) does nothing. 

What is the expected behavior?
▪️ Temperature should go up:
▪️ Display 260 °C
▪️ press and hold "Button A" 
▪️ Display 270 - 280 - 290 °C etc 

What is the motivation / use case for changing the behavior? 
▪️ Beeing able to adjust temperature quickly. 

What are you running:

Firmware Version: 2.04

PCB Version: 1

Power Supply: 
▪️ 24 Volt 2.71 Ah (65 Watt step up Booster - 3 Amps max.)

Cpt-Huck, 2018-05-04:
Yes exactly!

GnFlala, 2018-05-06:
I think he means 
When you enter change temperature mode to actually change it you have to multipress (many short presses) A or B button. 
You cannot long press one of them to be able to get the temperature where you want.

Cpt-Huck, 2018-05-06:
@GnFlala exactly!

commit 48cc1ff8 (2018-05-11): Add auto-repeat to temp adjust

Fixes #285
closed 2018-05-11 by Ralim