Ralim/IronOS #286: Donation

IAmOrion, 2018-05-05:
Apologies for entering this as an "issue" but I wanted to check something with you - I sent a donation (albeit not much - £10) via the link in the readme.md file (https://paypal.me/RalimTek) but it's been a day and still says "Pending". 

Just wondered is Ralim has received it or at least a notification of it?

IAmOrion, 2018-05-05:
Literally as I wrote this, I had an email saying donation accepted! Perhaps just a time-zone thing! Sorry!!

Thanks again for the great firmware

closed 2018-05-05 by IAmOrion

Ralim, 2018-05-05:
Totally on me :)
Because my paypal is in AUD, any donations that come in from other currencies require me to "accept" them and I managed to lose the notification email in my inbox :/ I normally login and check once a week to clear any that I missed emails so no-one needs to panic :)

I really appreciate the donation though :) It really, really helps with supporting the time spent on this :)