Ralim/IronOS #287: Heats up after "deep sleep" if i press "A" button, but screen will not come on

GnFlala, 2018-05-06:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
If i l forget the solder iron for a couple of hours without using it, then by pressing "A" button the iron starts heats up to the same temperature i had it to heat before but screen is completly off and i cannot find a way to turn it on again unless i unplug the power and replug it (reboot)

* **What is the expected behavior?**
Screen to come on by pressing one of the buttons? Maybe i am mistaken on this one?

***Steps to reproduce:***

1.  Let solder iron alone for couple of hours
2.  try to use it by pressing "A" button
3.  Iron heats up to the temperature you where using it before but screen is completely off 

* **What is the motivation / use case for changing the behavior?**
Being able to wake up the solder iron from deep sleep and have working lcd/menu

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.4
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : I am using 14.5V supply right now but i can change that if that is the case.

Ralim, 2018-05-07:
This is an issue that is related to the lockup issues reported elsewhere. I have found the current cause of this and will be pushing out a bugfix release asap for this.

Ralim, 2018-05-07:
Can you confirm fix in 2.04.1

GnFlala, 2018-05-08:
I confirm that with 2.0.4.1 wake up from deep sleep works normally.
Sorry it took a while to respond but I needed a few hours to test that again and again. 
What I did
I let the solder iron to rest for 6 hours. Then pressed the A or B button.
Screen turns on normally.

Thank you

closed 2018-05-08 by GnFlala