Ralim/IronOS #290: Screen issues

federck, 2018-05-08:
* **I'm submitting a ...**
  - [x] Bug report

* **Do you want to request a *feature* or report a *bug*?**
Generic OLED issues

* **What is the current behavior?**
1) The first five pixels of the screen (starting from bottom right side) flickers continuosly
2) Misalignment in scrolling text seems to have increased

* **What is the expected behavior?**
More fluid scrolling

***Steps to reproduce:***
1. Go to main menu
2. Wait for a description

  - Firmware Version: 2.04.1
  - PCB Version: 1
  - Power Supply (Voltage and Current Rating): Irrilevant (5V/24.4V)

Ralim, 2018-05-08:
Hi,

I will fix the glitching pixels shortly. 

Scrolling can't be made more fluid without slowing down. 

Because running the i2c bus faster than 100khz is the cause of all of the lock up issues so far as the oled doesn't handle it correctly and will randomly get into an unknown state. This then causes lock ups in the stm peripheral. If I find a workaround for this I will bump the speed up again, but it isn't looking likely.

federck, 2018-05-08:
Very complete as always, thanks

commit b6b20756 (2018-05-11): Delay I2C DMA thread on oled slightly

The flickering on the LCD screen was caused by the OLED DMA taking slightly longer than the delays, so the tail end would flicker if the buffer was cleared before it finished writing.

In future may want to double buffer the LCD.

Fixes #290
closed 2018-05-11 by Ralim