Ralim/IronOS #291: Stop hold-to-increment a setting when you hit the last item.

JohnEdwa, 2018-05-08:
* **I'm submitting a ...**
  - [x] Feature request

For example, when setting the shutdown time once you hit 60M you fly right past it before you can react. It would be better to stop on '60M' to indicate it's the last item.

Ralim, 2018-05-09:
Would you prefer stop on last or stop on first ?

Just thinking it might be better to cancel the auto-scrolling when the user wraps back to the beginning?

JohnEdwa, 2018-05-09:
Last, as we don't have any indication of how close you are to the end and you can't go back.    
Setting something to the max: hold down until it stops, done. Wanted the lowest? Press once more, done. Somewhere between? You now know the highest and lowest, so you have a grasp of when to stop.

If it stops on the first, you'd have to notice and remember the split second it shows the highest number, scroll through again stopping just short of it on the next run as to not overshoot and click a few times more to get it. Not nearly as useful.

Ralim, 2018-05-11:
Moving this out of 2.05 as I want to get 2.05 out sooner, and this will take a bit of work. Currently the repeat handler cant find out when the last one occurs nicely.

Ralim, 2020-09-14:
This has actually been finally checked off 😅

closed 2020-09-14 by Ralim