Ralim/IronOS #292: Boot Logo Options

Ralim, 2018-05-09:
* **I'm submitting a ...**
 - [x] Feature request

* **Do you want to request a *feature* or report a *bug*?**
Feature
* **What is the current behavior?**

Nil

* **What is the expected behavior?**

Add menu options for : 

* Turn on/off bootup logo
* Adjust display time [0-inf] 

I'm thinking this could be a single option for boot-up logo display.
With the options being [Off,1,5,10,30,Inf]

JohnEdwa, 2018-05-10:
* **I'm submitting a ...**
  - [x] Feature request

Add menu option for :
* Show logo on: Boot/Wakeup/Both

Eldenroot, 2018-06-01:
@Ralim - will you release v2.05 soon? I am asking because I have there 3 pcs of units from my friends to flash them to your fw from the stock one. Would be nice to use the latest version as possible. Thank you!

PR #1241 (2022-04-02): Bootup logo settings
@Ralim
Tried my best to implement *bootup logo settings* by digging through other PRs.
But since I am not absolutely sure what is missing / went wrong, I need help / advice on what to do to get this going.

resolves #292
closes #1253

Thanks in advance.
closed 2022-04-02 by Ralim