Ralim/IronOS #295: Iron doesn't boot after attempting to upload boot logo.

TerminalSaint, 2018-05-10:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Iron doesn't boot after attempting to upload boot logo.

* **What is the expected behavior?**
Iron should boot and display logo.

***Steps to reproduce:***
1. Connect iron to PC in DFU mode.
2. Transfer current firmware .hex file to iron to install firmware.
3. File change to .RDY format indicates successful installation.
4. Disconnect iron from PC and connect to power supply to confirm successful installation and proper operation.
5. Disconnect iron from power supply and reconnect to PC in DFU mode.
6. Transfer .hex logo file (created using TS100 logo editor) to iron to install logo.
7. File change to .RDY format suggests successful installation.
8. Disconnect iron from PC and connect to power supply.
9. Iron display stays blank and iron does not respond.

* **What are you running:**
  - Firmware Version: 2.05
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 24V 4.5A
  - PC OS: Windows 10

* **Other information** 
Perhaps I'm doing something wrong, but I believe I've followed all instructions properly. Attached is the logo file.
[TS100_LS_LOGO.zip](https://github.com/Ralim/ts100/files/1990052/TS100_LS_LOGO.zip)

Ralim, 2018-05-10:
Hi,
How did you generate the logo file?

TerminalSaint, 2018-05-10:
Using the logo editor program.

aetherbot1872, 2018-05-10:
I have the same issue. Using latest firmware 2.04.1

Ralim, 2018-05-11:
Hi,
Looking at your uploaded logo file, it was made with an old version of the logo editor program.

A while back the program was updated (as well as the python script) to change the address the images are stored at. 

Can you try re-generating your logo on a newer build please? 😄 
Sorry :(

TerminalSaint, 2018-05-11:
That's fixed it. 
Thanks for your time!

closed 2018-05-11 by Ralim