Ralim/IronOS #297: Show a message if settings/calibrations were reset on last FW update.

JohnEdwa, 2018-05-10:
* **I'm submitting a ...**
  - [*] Feature request

* **What is the current behavior?**
When changes are made to the settings between firmware versions they are reset, but there is no indication if this has happened to the user except stuff doesn't work like they used to. Most will notice the setting reset on their own, but I've for example completely forgotten to redo the temp and voltage calibrations. I last did them on v2.02, IIRC.

* **What is the expected behavior?**
On boot, pop up a message instructing users to re-calibrate and go through their settings again. Attaching this check to the tip temperature calibration having been completed might be a good idea, as at least mine is out of whack by default (it's 22C in here but the iron says 39C)

csdpm, 2018-05-10:
Wouldn't it be simpler just to go through and check the settings whenever you upgrade the f/w. It is, after all, a very deliberate physical process.

JohnEdwa, 2018-05-10:
They don't get reset just because you flash an update, they are only set to reset by Ralim if there are changes that demand it, like in commit https://github.com/Ralim/ts100/commit/f2f20111b6c7e3ca6263737aed1bd5dad5d4ae86 . If there are no changes that require the reset, it won't happen.   

And like I said, I did go through the settings every time they changed, but completely forgot the calibrations even exist as by default there is no indication for them - you need to enable "Detailed idle screen" to notice that the idle temp and voltage are out of whack. That's why it would be a good idea to pipe up that "Hey, this update you flashed also wiped your calibrations and settings, you should probably go check those again, mmkay?"

PR #535 (2019-12-28): Settings reset should warn
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [] The commit message makes sense
- [] The changes have been tested locally
- [] Are there any breaking changes

* **What kind of change does this PR introduce?**
This introduces a warning screen on startup if settings have been reset due to a firmware update (or corrupt flash).

* **What is the current behavior?**
nil

* **What is the new behavior (if this is a feature change)?**
Adds warning screen

* **Do [... cut]
closed 2019-12-28 by Ralim