Ralim/IronOS #298: TS100 bricked

zZ-JayBuck-Zz, 2018-05-10:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Booting with button closest to tip goes to DFU 3.43, C4D3BC1A (E:) drive opens. 
[image] These files are there, dunno if relevant.

When copying TS100 v2.04 to drive this occurs and the drive disappears.
[image]

After a minute or two E drive reappears as this.

[image]

TS100 does not power when plugged in now, and firmware from manufacture, returns with first error message.

TS100 is bricked, any help or should I just RMA it to Amazon?

b0bje, 2018-05-11:
You seem to have a trash folder on the flash side ?! Also looks like a
hidden osx folder .fseventd ? Delete those and retry the flash.
Also if you use a tool like terracopy, it could mess with the flash, so I
personally copy the hex file in a DOS cmd session.

JohnEdwa, 2018-05-11:
As for the "Please insert disk error", mine ended up just being a dodgy USB cable that provided power but didn't have a solid connection for the data pins, so try with a different cable first.

@b0bje Those files and folders are there by default always, not really sure why.

Ralim, 2018-05-11:
@b0bje Those folders are hard coded into their bootloader to appear, because otherwise both OS's (Windows and OSX) try and make them at mount time. Which doesnt work for the way they are trying to intercept writes and re-route to the decoder for HEX files.

@zZ-JayBuck-Zz  Can you please test with a new USB cable first (to rule it out).

When you have my firmware loaded on the TS100 it will not enumerate as a USB device, however the bootloader prevents writes from overwriting it normally so it should not have managed to have been damaged. 

If the bootloader is rejecting both official as well as my firmware that something is going odd with the bootloader code. If you have a ST programmer you can easily re-program the IC to re-load the bootloader code onto it.

Also a tiny point, I did once have this issue where the internal connector was going intermittent as the I forgot to put the little foam block into a new case I put on.

karpilin, 2018-05-13:
Bought a brand new TS100 from amazon, and on first boot has the same issue.
So far I have tried different usb cable and different usb port.

Ideally would like to have an advice how to get it working instead of just returning it to seller.

krepa098, 2018-06-01:
I'm having exactly the same issue. Flashing it successfully on Windows10 seems to be impossible as it always fails with "not enough space", followed by the wired behaviour in picture 3 and of course the iron becomes unusable thereafter as the firmware seems to be gone. Also copying the file from CMD fails with the not enough space message.

I had some success using the Linux flash script though. It also fails most of the time creating a TS100.ERR file, but after a couple of tries (felt like a 100) on different USB ports it succeeds eventually showing a TS100.rdy file. I'm using a pretty long USB cable of doubtful quality. That might be part of the issue.

Ralim, 2018-06-16:
I have since hear a report of the above happening to people who have *any* anti-virus that intercepts writes. Most of these also keep their hook in the OS even when turned off. This causes issues as the TS100 is hard coded to expect writes to the FAT partition in a very particular order, and any deviation from this can cause issues.

I have in the past recommended people to try an entirely different computer with different software (such as a friends) just to try a totally different OS base for testing.

Performance under Linux sometimes goes odd, but generally is fairly reliable if the file is copied over itself a couple of times.

If you are having repeating issues with this, one alternative is to flash the replacement boot loader [From Here](https://github.com/eDesignOSS/ts100-bl-flasher/releases).

This will change the boot loader for an open source DFU compatible one that is much easier to work with. (This can be an idea if you keep having issues on your machine, but it works fine on a friends).

durd, 2018-08-04:
Trying the alternative boot loader made things worse :(
It flashes, but after connecting the ts100 back to the computer no disk appears. I hear drivers being loaded, the chime from windows. But nothing :/

Ralim, 2018-08-04:
The alternate bootloader is USB-DFU compliant, so it won't show up as a USB drive, but will work with dfu until and other programs.

durd, 2018-08-04:
Ah ok, thanks for the tip.
Downloading to the ts100 fails at 41%, tried manufacturers which fails too.

Ralim, 2018-08-05:
If you are using dfu-util you will need a bin, not a .hex file.

durd, 2018-08-05:
Hm, ok. Well, I'm back to the beginning anyway. Used the bin for the stock bootloader as I couldn't find any else.
Anyway, guess I'm forced to stick around with the manufacturers firmware.

Ralim, 2018-08-05:
@durd Happy to try and help further, but telling me it just doesnt boot doesnt give much information to go on. Give the (now) updated 2.05.01 a try, but if it fails usually it comes down to a hardware defect in your unit would be my guess (if it flashes correctly).

durd, 2018-08-05:
@Ralim I'd be happy to help you more, but seeing as I wasn't getting any other errors than OP, there's not much else to post or mention. I tried as you suggested a different bootloader (no one else had tried so I posted that I had more trouble), which wasn't much easier to work with except with more help from you, which I'm of course grateful for!

I just now tried your re-uploaded 2.05.01-version, this worked perfectly with stock bootloader.
I actually did have a pre-order of the TS80 in the cart at banggood, but couldn't order because the site became unreachable right when I was ready to order. So yeah, you actually saved me a bunch of money rescuing me from dfu-util-hell! :)

@krepa098 as Ralim mentioned, you need to use a .bin-file. Grab the stock one from the same link and apply that. Then download the 2.05.01 again (even though it says it hasn't been updated) and try again.
Mine worked with this.

krepa098, 2018-08-05:
Sorry, my bad. I uploaded the .HEX instead of the .bin file. It is now working flawlessly. Thank you very much for providing the replacement boot loader. It seems to be absolutely reliable.

```
> dfu-util -D TS100_EN.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [=========================] 100%        40108 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
```

0xygen8, 2018-08-12:
I seem to have bricked my TS100 with my OSX computer
I can still enter DFU mode by hand and reflash the original 2.18 fw
but when press a lot of buttons in normal mode the screen locks up 

I tried to correct the problem by using a W10 computer and reflash the original FW
but every time I enter the DFU mode I see the troubling files and dirs

```
drwxrwxrwx  1 0xygen8   staff    512 Nov 25  2014 .Trashes
-rwxrwxrwx  1 0xygen8   staff   4096 Nov 25  2014 ._.Trashes
drwxrwxrwx  1 0xygen8   staff    512 Aug 12 18:00 .fseventsd
drwxrwxrwx  1 0xygen8   staff    512 Oct 29  2014 System Volume Information
```
When I remove these files they seem to reappear and rebrick [sic] the device
Is there a way to fix this?

JohnEdwa, 2018-08-12:
In this [exact same issue](https://github.com/Ralim/ts100/issues/298#issuecomment-388243348), it was explained that those files need to be there.

0xygen8, 2018-08-12:
Ok -- I seem to have skipped that post, but still I can flash the latest firmware to the 3.4.3 Bootloader (result RDY) but when I reboot it it does nothing (sometimes just random pixels on screen) the only way I can us it when I reflash the original 2.18 fw  and then 1 out of 10 boots the screen works and I can use it
puzzled... (before I flashed it with this firmware it worked fine)

Ralim, 2018-08-13:
Hi,
Can you please confirm exactly which firmware you have flashed? What hardware revision is your iron ?

It is almost impossible for this firmware to break anything, how long did you use your iron with the original firmware for? I have noticed a recent uptick in units with screen issues, which show up as random dots. This firmware can make these occur more often as it runs the i2c bus faster than the old stock firmwares (such as 2.12) however, even the newer miniware firmwares run the bus faster.

0xygen8, 2018-08-13:
I have tested your version v2.05.01 and v2.01 as said the bootloader is 3.4.3
How do I get the hardware revision of the board?
going back to 2.18 (original) somewhat fixes the dots [image]
but I used the iron for a day with the original fw for a simple project (1 hour without problems)
if I flash your FW (and get a RDY) the display never works (all black)
when flashing back to 2.18 and pressing the B button a lot the screen now goes haywire (showing text mirrored or several pixels shifted up)

Ralim, 2018-08-13:
The only real way to get the hardware rev is to be able to boot the firmware, however by the looks of that, something on your I2C bus is not playing fair, and is locking up the device and preventing memory writes to the lcd. This is *usually* a hardware issue, especially considering how hard it is to replicate. 

This usually shows up when the screen is re-drawn and an error occurs on the I2C bus. Which would explain why the original firmware also shows this on rapid screen changes. 
Any chance you have taken the back cover off your unit and checked for damage?

0xygen8, 2018-08-13:
I'll open it and report here -- thank you for your help 👍

0xygen8, 2018-08-13:
I've let my daughter open it up hardware revision is 2.85

0xygen8, 2018-08-13:
I've ordered a new one and hope that one has less issues :)

Ralim, 2018-10-18:
@0xygen8  Any updates to this ?

zZ-JayBuck-Zz, 2018-10-18:
Hey Guy sorry I gave up on this. I have returned it. Thank you for all you
help. If I try again I will let you know

0xygen8, 2018-10-18:
I have received a new ts100 and to be sure I'll use it for several months to be sure it has no issues before I'll re-flash it -- the strange thing is that (old ts100 with issues) It does not take the new firmware (yes I see a RDY file) but when I power it with 20v it does not do anything...

Ralim, 2018-10-19:
What happens when you power it with USB ?

0xygen8, 2018-10-19:
Screen is an stays black -- only when when I put it back in DFU mode the screen comes back

centuryrehab, 2018-12-05:
Hi. I just received my TS-100. I tried to flash it with a few computers and I get the same dots on mine or nothing at all. 

I can fix it by flashing the default firmware from their website.

I wonder if new TS-100's all have this issue? Is mine a clone?

dhiltonp, 2018-12-07:
@centuryrehab: dapboot isn't very common, are you following [the instructions in the wiki](https://github.com/Ralim/ts100/wiki/Upgrading-Firmware)?

0xygen8, 2018-12-07:
did you make the slower I2C version? so I can test if that fixes the issue

0xygen8, 2018-12-27:
is there anything else I can do to salvage this unit? debug something, flash in a different way upgrade DFU...

ExtraThiccDave, 2019-02-01:
Just got my TS100 today and i wanted to flash the ralim FW according to some ytb videos. I dragged over the file to the usb drive and when i wanted to copy it over, the flash drive was empty, it showed an error "not enough space for this file system", the disconnected and showed up again as E drive which says "insert E drive". The iron is dead, doesnt power up with 19V source or USB in config mode. Tried flashing back the stock firmware, ends with the same error. Any tips?

Ralim, 2019-02-02:
@ExtraThiccDave If you flash dapboot then its no longer a usb drive but instead a usb DFU device (read the documentation on what your flashing).

@FFY00 If you do get a ST-link it might be worth checking for anything odd / writing the image directly. (you can just program the hex).

ExtraThiccDave, 2019-02-02:
@Ralim Thanks! I discovered it like 20 minutes after writing that comment.
I downloaded dfu-util, connected the iron, but when i run the command for DFU device list, it shows as no devices detected.
Tried different USB cables (i had good quality cable to begin with), nothing.
Same  "BL flasher" screen shows when i only connect power as well, doesnt need to be usb.
Do you have any ideas? Seems to me like my unit is defective.
Thanks a lot!

Ralim, 2019-02-02:
@ExtraThiccDave Try a different usb port / usb controller. Miniware didnt include a crystal so its clocks typically just out of spec (especially if room temp isnt 25c). Often a different controller helps. Intel ones tend to be better.

ExtraThiccDave, 2019-02-02:
> @ExtraThiccDave Try a different usb port / usb controller. Miniware didnt include a crystal so its clocks typically just out of spec (especially if room temp isnt 25c). Often a different controller helps. Intel ones tend to be better.

I tried all USB ports on my Ryzen system and my Intel laptop. No luck. Room temp is around 21°C. I'm really out of ideas at this point.

ExtraThiccDave, 2019-02-02:
Anyone got any ideas? This is how it behaves atm. I flashed the dapboot.hex as last resort, hoping i could flash 2.05 bin file.
https://photos.app.goo.gl/p8quXhyZNQCQKP667

ExtraThiccDave, 2019-02-02:
@Ralim would it be possible to replace the STM32 daughterboard and fix the issue? Or is the firmware and bootloader on the main PCB.

Ralim, 2019-02-03:
Once you have flashed the dap boot software, it replaces the miniware one, so now you need to boot it up while holding the front button to enter the dap boot bootloader.

But yeah its totally possible to reprogram the riser board, all code is on there. Also if you cant get it working, im happy to re-flash the daughter board for you at the cost of postage if you wanted (I'm in Australia though). You can also use any st-link etc to re-program it as well.

ExtraThiccDave, 2019-02-03:
@Ralim I just spent 3 minutes banging my head against the table XD I forgot to hold down the front button. Now it's being detected by DFU-UTIL.
I'll try to use web DFU and flash 2.05.
If i wanted to put the stock bootloader back, would i have to reflash the ROM afterwards?
In the worst case i'd attempt to ship it to you or get a replacement from miniware, i don't like the idea of returning the whole iron when only the daughterboard is bad.

FFY00, 2019-02-03:
@Ralim can you tell me what to change to make a firmware version with a slowed down I2C bus like you were talking above? I can compile it myself. It might be a while for my ST LINK to arrive since I ordered it from China.

whitehoose, 2019-02-03:
That's won me an extra week's pocket money.  Not having a ts100 I didn't know the process in detail, and I've no wish to try replacing bootloaders.  So I've been biting my tongue, but I had that feeling that what you were describing a negligent discharge, own goal or whatever you want to call it ... was what most likely what you were experiencing.

My advice would be when the st-link arrives, go somewhere with deep water and throw the st-link as far out into it as you can manage.

If you bought from banggood, they ask you to give them video evidence of the whatever it is either doing, or not doing whatever it is you're complaining about.  If you convince them they repost the entire item as it comes in the box.  (My ts80 died  and that was what they did for me)  seemed a shame moboard faulty 
 rest of h/W hardly used, so I then contacted mini to see if they would sell me the bits to resurrect the dead iron.
I told my tale of woe and the guy in tech support said he would like to see what the damage was and if I posted out the motherboard he would like to try and fix it.  I did, he did and within a couple of weeks I was the proud owner of a second, working Iron.
The follow up from both seller and manufacturer was excellent.

dskleingeld, 2019-06-23:
I had the exact symptoms as @zZ-JayBuck-Zz but then on a TS80. Switching to a USB 2.0 port did the trick for me. This was on a Ryzen B350 motherboard.

whitehoose, 2019-06-23:
USB is a weird beast.  Windows sees it either as a network port or a SCSI device.  In both cases it assigns a  unique device ID - just how that's arrived at involves date and time plus the USB port id.
Problem is it stores it forever! Using windows 10's idiot proof restore/rebuild option that means everything you've ever plugged in, for non tech types that means the caches will probably never be cleared for the machine's life.
Google "reset usb ports windows xx" xx=your flavour of win.

Not something to do regularly, but if you know a device is good else were worth a bash.
The downside is you'll have to reinstall all your USB devices, not the end of the world in my book - but some folk don't realise.

ppisljar, 2019-06-29:
i got into the same issue.

i wanted to install alternative bootloader, copying that file 'kind of' works: 
- file starts copying
- i get windows sound that usb device was disconected before it finishes, but everything seems to be ok otherwise
- drive reapears with filename changed to .RDY
- however screen still says DFU 2.45

tried rebooting, its still DFU 2.45, so it seems flashing the alternative bootloader didn't work.
am i missing any steps to installing the alternative bootloader ?

is there a place i could find original firmware so i can try flashing that one over usb drive ?

ppisljar, 2019-06-29:
never mind, trying on an old computer with usb 2.0 solved my problem

Jasonstewart81, 2019-10-15:
Hello everyone. New here. Really could use some help. Ive had my sainsmart ts100 for maybe 3 years now and absolutely love it, up until I found out about this upgraded firmware and tried flashing it over by following the instructions with no success. I was having the same issues others were having where it wouldn't go through, the serial numbered drive window that would pop up would just revert to a USB drive D: and either say try again, skip, or cancel. Then I tried following the instructions about the bootloader I see mentioned in here but was extremely confused about how to do the process. So now my ts100 is completely screwed and stuck on the "bl flash" when powered by the DC barrel jack or via the USB cable. I'm heart broken. I love my ts100 and now I can't use it. How do I fix this?

Ralim, 2019-10-16:
@Jasonstewart81 
If you have flashed the new bootloader and done the "confirm" stage, so it has replaced the original firmware. You now have to use a usb-dfu compliant tool to flash the firmware, rather than it showing up as a virtual USB drive.

If you have flashed the alternate bootloader, just use a USB-DFU tool for your operating system. As you have not stated your operating system, I'm going to assume you are using a Windows based OS. Back when I used Windows I was using http://dfu-programmer.github.io/ to do DFU flashing.
Just use that program to load the .bin file into the flash at the correct offset, or use the HEX file (if its suported)

If you have not actually replaced your bootloader, then you should still be able to hold the front button (A) and have it show up as a USB drive on your pc. In this case you just need to flash the firmware like normal.

Jasonstewart81, 2019-10-16:
Sorry, I'm using Windows 10. I'm not savvy with programming and stuff so I'm a bit lost when trying to figure this stuff out. So basically where I'm at with my ts100 is whether plugged in via the micro USB cord or DC barrel jack all I get is the "BL FLASH" screen where you confirm it and it says "done!".

Jasonstewart81, 2019-10-16:
So what all do I need to download to try this out? I know I'll need your TS100-EN.HEX file for the firmware. What are the other files do I need to get my iron back online? 😔

Jasonstewart81, 2019-10-16:
> Weird. When I flashed dapboot ( [`ts100-dapboot.hex`](https://github.com/eDesignOSS/ts100-bl-flasher/releases/tag/1.0.0)) it did not replace DFU, it replace the firmware. When I start the iron normally it just shows `BL Flash`, if I start it pressing the front button it goes to DFU.

This is EXACTLY my issue as of right now and i have absolutely no idea how to do anything else. If i press the A button while inserting the USB cable it does pop up in my device manager as "STM device in DFU mode" under the Universal Serial Bus controllers. I just have no idea what to do from here.

Ralim, 2019-10-17:
If your device is showing up as "STM device in DFU mode" that means that you **have** replaced your bootloader with the DFU one instead.

As i mentioned earlier, this means you must use a DFU tool to load the firmware. You could try the one i linked earlier, i normally use dfu-util to load firmware on my devices.

Jasonstewart81, 2019-10-17:
I'll try again, but I've never done this before so I have no idea what I'm doing. That why I came here looking for guidance on how to do it.

Jasonstewart81, 2019-10-17:
[image]
This is what I get trying to use the DFU-UTIL you linked to me.

Jasonstewart81, 2019-10-17:
All i wanted to do was upgrade my firmware by following the very simple steps and it didn't work. Now i have a useless soldering iron. About ready to throw it in the trash. I'm pissed because there was nothing wrong with it prior to me trying to upgrade firmware.

Ralim, 2019-10-17:
Hey man, don't panic. This will be fixed fine, if you haven't used normal USB-DFU updaters before this is just a bit of teething as it's the first time your doing it.
That error message indicates your missing libusb, so check the libusb wiki for how to setup the libusb driver https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows

You could also try following this guide : https://www.hanselman.com/blog/HowToFixDfuutilSTMWinUSBZadigBootloadersAndOtherFirmwareFlashingIssuesOnWindows.aspx

I don't have any windows machines currently so I cannot give you step by step instructions, but generally with windows it's just a case of getting drivers sorted.out and then it works.
Try not to panic, this isn't going to suddenly brick your device, you just chose to flash a different bootloader to the unit, so now you have to use that bootloaders method of updating (which, is generally much, much more reliable than the miniware one).

Jasonstewart81, 2019-10-17:
I really do appreciate the help. I just honestly don't know what i'm supposed to do. I've tried to read and follow the instructions, but honestly they aren't very helpful. It's worded in such ways that only people who already understand it and have programming experience will know what to do. I give up on it. I'd rather have a useless ts100 than brick my computer I recently built myself. I've built several computers and I am a knowledgeable guy when it comes to many things, but programming isn't one of them and I really just can't get a grasp or understanding of how to do what you guys are telling me to do. I'll just stick to using my Hakko FX-888D for now I guess.

Ralim, 2019-10-17:
Try using
https://devanlai.github.io/webdfu/dfu-util/
I haven't used it in ages, but it should work. Give it the .bin file from
the releases page.

If the TS100 doesn't show up straight away, you may need to fill in the
vendor ID. This can be found by looking in device manager ->details ->
hardware details. Then it will be VEN=XXXX and you need the four X's.

Jasonstewart81, 2019-10-17:
See, what's weird is sometimes in the device manager it shows up as STM device in DFU mode and other times I get DAPBoot something bootloader.

[52 more replies]

closed 2022-05-08 by Ralim