Ralim/IronOS #301: Wrong temperature coefficient? with Hakko T7/T12/T15 tips

patapovich, 2018-05-12:
* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

Request a feature

* **What is the current behavior?**

There is a big error in set temp., temp. shown on display and real temp. when using Hakko T15 tip instead of original.

| Set temp.| Disp. temp. | "Real"* Temp |
| ------------- | ------------- | ------------- |
| 100  | 152 | 75  |
| 150 | 192 | 150 |
| 200 | 240 | 195|
| 250 | 284  |  |
| 350 | 370  |  |

*My measuring equipment is not quite up to the task

Also current temperature shown on display jumps a lot, while real temp. is staying reasonably stable.

* **What is the expected behavior?**

Reasonably stable temperature regulation and display with Hakko tips.

* **What is the motivation / use case for changing the behavior?**

Using Hakko tips and Handle with our iron.
[Adapter](https://www.thingiverse.com/thing:2739632)

* **What are you running:**

  - Firmware Version: 2.05
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 24.1V 6.5A

* **Other information** 

From what I found out, the T12 temperature coefficient is 20-21uV/C.

 I'am not not really sure what going on, because I would expected the error be the same for set and shown temperature.

Eldenroot, 2018-05-12:
It is because you dont use the "original" tips. Maybe this could fix this: https://github.com/Ralim/ts100/issues/275

patapovich, 2018-05-12:
I'm not sure the PID tuning by it self will resolve the problem.

The T12 tip thermocouple type is apparently different from "original" the is a need for correction. 
What surprised me  is that real temp is pretty close to set, at least for temperatures I can measure.
Real-time temperature reading is off and not by specific offset.

Ralim, 2018-05-14:
> Reasonably stable temperature regulation

This will be improved by #275 hopefully.

The temperature offset is because you are not using official tips with the unit, so naturally these would have a different temperature response. I can get my hands on some Hakko tips and test this as well, but it might be a bit of time until this is implemented as a feature, as the Hakko tips have also led to some damaged units. Not sure exact cause, but I know two people who were using Hakko T tips and both ended up breaking the internal mosfet in their irons. (Works fine after upgrade though).

This should be fixable in future, but I dont want to encourage using Hakko Tips.

Eldenroot, 2018-05-14:
Same here, I read many reports from users whose used non original tips with TS100. Many of them reported that it is not a good idea, after some time TS100 were broken.

JohnEdwa, 2018-05-14:
>as the Hakko tips have also led to some damaged units. Not sure exact cause, but I know two people who were using Hakko T tips and both ended up breaking the internal mosfet in their irons. (Works fine after upgrade though).

Can't really break a mosfet by any other way than overdriving it, one datasheet gives the TS100 mosfet (CJQ4953) a max rating of T<10s 3.7A while the TS100 and Hakko tips are supposed to be exactly 8.3ohm to give the rated 70W, 2.9A@24V. 
Get a crappy one-hung-low tip that is 15% off the spec and you'll cook the mosfet - seeing as one of my official TS100 tips measures 9.39 ohms (13.1% high), I don't think it's that unlikely.

Ralim, 2018-07-27:
Going to close this, and count it as part of #303

commit 3d1d47ac (2018-07-27): Long overdue additions

Closes #329
Closes #324
Closes #331
Should Close #321
Should Close #301
closed 2018-07-27 by Ralim