Ralim/IronOS #302: PID integralCount is not static in startPIDTask

ornea, 2018-05-12:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
integralCount is not cumulative

* **What is the expected behavior?**
integralCount should be cumulative for better PID performance

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

integralCount is currently declared as

  int32_t integralCount = 0;

and in the code is used as 
   integralCount += ierror;

But as integralCount is not static and reset to 0 each time the function is executed so has the same effect as
   integralCount = ierror

I think PID performance will improve by declaring integralCount as static.

My First Issue submitted so treat accordingly.

Ralim, 2018-05-12:
Hi,

If you look [here](https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/main.cpp#L831)

As the system is running under and RToS the code runs in a forever loop so there is no need for it to be static as that loop never exits :)

ornea, 2018-05-12:
Thanks Ralim. 

My limited understanding of RToS has now slightly improved.

 I guess I close the request.

closed 2018-05-12 by ornea