Ralim/IronOS #303: Simple high precision temperature calibration

ruslan-khudyakov, 2018-05-14:
* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Setting up of tip reference temperature as 100°C (temperature of boiling water)**

The simplest way to calibrate is dip the tip into the boiling water.
(water boils at 100 °C (212 °F)

The currently used method (ambient temp) expects ≈30°С and is not accurate.

***Procedure:***

1. Heat the water to a boil;
2. Dip tip in the water;
3. Press "Calibrate"
4. Done! )))

Ralim, 2018-05-14:
Hi,
This has been suggested before, and I might impliment this as a second stage calibration.

However, "Water" does not boil at 100C. Pure `H20` does, when its at 1 atmosphere (and thus standard pressure). Any deviation from this can cause errors in the measurement.

At the moment, the temperature sensor in the handle is used (Thus the warning to ensure the handle is the same temp as the tip). This provides a temperature in 0.1C increments that is used for the calibration.

ruslan-khudyakov, 2018-05-14:
Hi, thanks for answer!

Right now in my room is 23°C degrees and after calibration soldering iron show me tip how 33°C.
(Was in standby mode etc.)

However, this is so far from 0.1°C…

Yes, idial conditions for 100°C it sea level and 1 atmosphere but even at 1000 meters water does boil at 97°C.
How many people living on 0.5 miles above sea level? ))) 

P.S. as a second way it would be cool!

yschaeff, 2018-05-14:
What about calibrating at 37 degrees Celsius? More practical than boiling water and more accurate than "room temperature". Also Come to think of it the iron can double as a thermometer. ;)

ruslan-khudyakov, 2018-05-14:
@yschaeff, yeah, "rectal calibration mode", LOL!!!

JohnEdwa, 2018-05-14:
The temp calibration should simply just be user-settable offset like the voltage, so it doesn't matter if you do it at 15C, 37C or 100C. Everyone owns a thermometer of some sort.

Eldenroot, 2018-05-14:
Anyway we need a better way how to calibrate properly - we need at least two points calibration - two temperatures - 30 and 100 C? What do you think

ruslan-khudyakov, 2018-05-14:
The best way in my opinion is 100°C plus Manual User-setting like a voltage.

Eldenroot, 2018-05-14:
37 and 100 for two points calibration would be the best. One point calibration is not precise.

Ralim, 2018-05-15:
@Eldenroot I'm not against a multiple point calibration option. But generally it will just provide a means for the user to mess up the calibration of the iron.

As to go further would involve then calibrating the thermocouple and op-amp, and to calibrate out the tolerance of the op-amp circuitry etc would require two stages of calibration, and two different ambient temperatures as well to allow for a change of the cold junction.

At the moment the firmware is assuming an average value for the temperature gain for the system. And the calibration is there to calibrate the offset in the system that is unaccounted for after cold junction correction.

The actual correction required for cold junction can vary tip to tip, along with both the tolerance of the op-amp and the temperature curve for the tip.

So the calibrations required for this would ultimately inflate to a 4 step process, done for two different tips to profile the unit fully. Which still will only improve the tip actual temperature slightly.

Note that there is currently an issue with the screen showing a higher temperature than is actually at the tip due to the filtering done for the UI. This is most pronounced at the low end of the scale, and dimishes when the iron is operating. This offset is a bug that will be handled in future.

Eldenroot, 2018-05-15:
Ok, you are right. I am looking forward to see this bug fixed, keep good work, perfect fw.

ruslan-khudyakov, 2018-05-15:
We have two available constants —
**1**. Boiling water: 100°C;
**2**. Melting Ice: 0°C.

The best for home use.

blitzcode, 2018-05-16:
Using melting ice & boiling water seems a bit weird to me. The more obvious and relevant calibration point would be the melting point of some eutectic solder alloy like normal 63/37. Besides, a multi meter with a thermocouple is pretty common equipment and cheap Hakko tip thermometer clones are available for under 10 bucks.

I noticed on my TS100 that the temperature is completely off:

http://www.minidso.com/forum.php?mod=viewthread&tid=3168&extra=page%3D1

This is also not a constant offset and increases with the set temperature, so a single bias value would not be enough. Calibration with the second temperature sensor at room temperature does not seem useful as the error increases with temperature, so at 20C or so it is negligible.

I would really appreciate a feature in the custom firmware that allows this to be mitigated as with the current official & custom firmware the difference between actual vs displayed temperature >30C.

blitzcode, 2018-05-16:
Yes, I think so. It's going to be tricky / messy / unreliable / impossible to heat the iron to exactly 100C with boiling water or cool it to exactly 0C with a bunch of ice cubes. Slowly increasing the set temperature till you reach the well-defined melting point of 63/37 solder (which you likely have) seems much simpler and more reliable. The melting point of solder is also closer to the temperatures which you'll actually be using the iron at, making the calibration more likely to be useful in the presence of any non-linearities like the ones I observed when measuring my TS100. Besides, if you have an external reference like a DMM or tip thermometer you can also save yourself from messing around with kitchen supplies to calibrate your iron ;-)

But I'm not an expert on measuring & calibrating soldering irons, I'm happy with any solution that would allow me to get my TS100 to heat up with <10C error from the set temperature.

ruslan-khudyakov, 2018-05-16:
@blitzcode, the iron in the boiling water will have exactly 100°C and we will get maximum accuracy ±1°C (may be better)

Ralim, 2018-05-16:
Hi,
I'll try and get something a bit better in a coming firmware.

Can I ask what tips everyone is using? I only have offset issues on the non production tip I was sent and hakko tips.

It is looking like some models of tips have a much higher error compared to some.

I own two of the clone hakko fg-100 units, and neither gives the same result (difference of about 10C).

When testing my older BC2 tips and C1 tips they come in real close to the set point, however the newer BC2 tip is about 15C.

I have noticed significant offset can occur if the offset error cancellation is done with a warm handle, Or warm tip. 

At the least I will try and get an updated firmware that lists the temp offset out when I'm next working on this.

ruslan-khudyakov, 2018-05-16:
@JohnEdwa Clone does not guarantee accuracy.
(Original HAKKO FG-100 tolerance of ±3°C)

blitzcode, 2018-05-16:
@JohnEdwa 

Exactly, this stuff is not expensive. Nice graphs! ;-)

Here are my measurements from the other thread:

```
Set  TipTherm Fluke
300  267      269
350  308      314
400  355      358
```

TipTherm = FG-100 clone, Fluke is a Fluke DMM with a K-type thermocouple. I used a D24 tip. I also have a BC2 tip, also very inaccurate.

It seems the FG-100 clone and my DMM agree reasonably close. If I place the DMM thermocouple on the D24 tip the measurements are a few C below the FG-100, if I place it a bit higher on the tip they're a few degrees above.

I also found this video:

https://www.youtube.com/watch?v=DEEaLMv6dog&feature=youtu.be&t=12m49s

Some users claim they get very accurate temperatures with their TS100, I certainly have an error > 30C. Official firmware is no better. Both my tips have this huge error.

ruslan-khudyakov, 2018-05-16:
@Ralim After replacement tip we need recalibration.
All tips a little different. (thermocouple position, weight, volume, etc)

JohnEdwa, 2018-05-16:
@ruslan-khudyakov 
> Clone does not guarantee accuracy.
(Original HAKKO FG-100 tolerance of ±3°C)

Of course not, you pay $250 for that promise of accoracy, but the device itself is just a K-type thermocouple reader, and the sensor is a K-type thermocouple with a piece of metal crimped on top - both super simple things for a clone to do for a fraction of the price of the 'name brand'.

And seeing that my $15 clone, the device itself is within 1.7C from my Brymen BM869 (0.3% +-1.5C) and the testing the same temp with a regular K-type the difference is just 0.8C, I'd say it's accurate enough, seeing it's just 16% the cost, wouldn't you say?

[image]
[image]

ruslan-khudyakov, 2018-05-16:
@JohnEdwa 1.7°C between Clone and BM869 doesn't mean anything.
Accuracy - it's tolerance from Real Correct Temperature, not comparative cheap gadgets measurements.

Here we don't know temperature you really have here.
We just know that THIS measurements are close.
For assess of the accuracy we need a temperature standard (etalon).

JohnEdwa, 2018-05-17:
@ruslan-khudyakov 

Sure, we have no idea what the actual spec for the FG-100 clone is, or how accurate and precise it actually is, or if the measurements are linear at all.
But we do know it for the BM869 - it's has a 0.3% +- 1.5C spec on temperature. The 294.5C I measured, the real temperature should be between 292C and 297C. And when that same temperature was tested with a different thermocouple, the BM869 got 293.7C, meaning it should be between 291C to 296C.

And as in this instance, the FG-100 measured 292C which fits. If I wanted to do it, I could see if my FG-100 clone also fit the 0.3% +-1.5C spec - all I would have to do is do a bucketload of measurements with both, and check that they all fit within the range of the BM869 spec.

blitzcode, 2018-05-17:
I got a KU tip today, quick test at 320C set shows ~290C with the tip thermometer, seems like the temperature is completely wrong with this one as well. I don't think there's much need to discuss the finer details of measuring temperature and calibrating soldering irons, the problem appears to be in a different order of magnitude ;-)

LarsSimonsen, 2018-06-22:
> Can I ask what tips everyone is using?
BC2

Ralim, 2018-08-02:
FYI:
I'm working on this at the moment slowly as I have time / patience. 
@JohnEdwa Since you have a few tips at your disposal, could you give me a hand by getting a spread of 2-4 temps over the temp range per tip and recording the "Rtip" measurement that is now shown in the debug menu (long hold the rear button on idle screen). Doesn't need to be perfect temps, but even just some rough values to get an idea of the different tip curves.

JohnEdwa, 2018-08-03:
@Ralim Sure, though these tips cool off really fast and even more so if I actually try to measure the temperature at the same time, so a way to see Rtip while the iron is on would be immensely helpful.

Repled, 2018-08-05:
Hi All,

I bought two TS100 and one with the full kit of tips. I have now made some tests and this is what I have come up with. I have a Fluke 87 mkV with a thermocouple. I have used all tips un/calibrated and the latest Ralim release (Maj 7:th 2018).

1. It is very important to wet the tip with solder when measuring the tip temperature and give it time. Otherwise smaller tips give much lower values than larger tips and the time is needed to heat the external Fluke measurement thermocouple.  Theses measurements gave that the tips where in the range 10-20 degC below the set temperature, at approx. 200 degC.

2. When using a 63/37 Tin/Lead Solder with a melting point if about 183 degC, all my tips melt the Solder in the range 190 - 200 degC. The important thing here as well is to overheat the tip and wet it with solder before going down in temperature to test the melting point. Otherwise the same thing here, smaller tips give higher temperatures.

I have a Weller TS80 and that one is off by about 30 degC (just an analog turning knob though).

The problem I see when going up i temperature is that the tips start to deviate more and more the higher they get, this is also confirmed by JohnEdwa's measurements.

Thinking out loudly, I am wondering if this is partly due to the design of the tips where the thermocouple and the heater is integrated. If you have a fixed temp you will have a more or less fixed PWM dutycycle for that temp. The higher you go the higher will the dutycycle be. At higer temps, if the wire (heater + thermo) within the tip does not reach equilibrium you measure a higher temp
than what the outer part of the tip actually has. The temperature gradient, cooling of the tip also gets steeper the higher you go in temp.

Does this make any sense?

)) P

P.S. I would try to push the ADC measurement as far away from the PWM-pulse as possible.
(I have not looked at the exact implementation so this is perhaps already the case?)

Ralim, 2018-08-05:
@johnedwa
Good point, I'll try and get you a testing build soon.

@repled

I agree with your two points completely, and when measuring I usually try to ensure the top of the thermocouple is inside of the solder glob.

The deviation is a combination of non-linearity as well as when above 400C the tips starts to run out of headroom of the ADC/op-amp.

The design comes into this as Heating and measuring the tip temperature is mutually exclusive, and there is a recovery time after the end of the heating pulse train for the sensor to stabilise and the op amp to desaturate. The tip appears to recover faster than the op-amp.

In the system at the moment the hardware is setup to trigger the ADC automatically after a delay at the end of the pwm period. This delay is not perfect,and on some devices the first ADC sample will catch the tail end of this recovery,which will lead to a slight overreading, however this only occurs during full pwm duty (usually only the main heat up), which is mostly hidden by the pid and thermal reaction time of the tip 

https://ralimtek.com/stm32_double_pwm/

That is the small article on how I setup the timers to gain nicely scheduled ADC readings.

Cherenkov11, 2018-08-06:
Just an idea, due to the variety of tips (types), I think it would be a good idea add presets, to store the calibration setting for each tip.

Repled, 2018-08-06:
Thank you for the link, very enlightening, and your comments !
You have done your homework.

It's very tempting to start modifying the hardware, but that is perhaps to go one step too far ;o)
(Even though it's a very tweak-friendly design).
- DC couple the PWM-signal, remove, short C10.
- Remove the Cap parallel to Q2.
- Get the OpAmp out of saturation faster (diodes).
- Lower the Gain of the OpAmp slightly, make it more linear at high temp.

In practice, when soldering you probably want to be in the range 330-380 degC to heat and wet the joint quickly. Just using a fixed offset for each tip would perhaps be the simplest solution. (This is what I personally would prefer). This on the other hand requires that you have same means to measure the tip temperature.

The other way is to try to get a model of the non linear behaviors in combination with some sort
of calibration, but it might be a tough road to walk, or it's not. If you get the temperature right within
+/- 20 degrees, that is probably good enough.

The current calibration does a fairly good job at getting the temperature right at 200 degC.

On the other hand, I have lived with my 'analog' Weller for almost two decades and have learned to
set the dial between 350 and 420 degC depending on situation and tip, and I just now found out that
the temperature is off by 30 degC.

Just curious, how do you compile and debug this design? Eval board?

Ralim, 2018-08-06:
I would not modify the hardware for two reasons; Safety and consistancy.
Having two firmware builds for drastically different hardware is a pain. The cap (C10) actually provides a means of safety, where a stuck pin state will not cause heating.

The calibration in the unit already does an offset calibration, but as each model of tip has slightly different gain values, some are closer than others. This is all the stock firmware did which is why its what I carried forward, however going to an option to either do single point + select tip model OR two point calibration would be ideal. This allows a fast cal option for most users, with odd ball tips supporting a two stage cal.

I measure the tip temperature during the cal by asking the user to use a cool iron, so the handle sensor and tip should be in alignment.

To debug: Solder the SWDIO & SWCLK wires to the small mcu board inside the iron :)

Ralim, 2018-08-08:
@JohnEdwa 
I have made the attached firmware
It's accuracy will be worse than stock most likely as I'm partway through enabling the second ADC to get a bit more resolution in temp measurements, and haven't calibrated this in at all.

But, if you load this firmware and enable the detailed soldering screen, the top line will show you the raw tip temp. This will be as noisy as the thermocouple is (quite), so you may need to eyeball figures a bit i'm afraid. Generally its more stable the bigger the tip.

[TS100A.zip](https://github.com/Ralim/ts100/files/2270081/TS100A.zip)

Would love a dataset of Measured temp Vs the raw temp (RTemp) for the different tips, with as much detail as you can be bothered to do. At the least I'll try to push out some updated curves while I try and nail down why my two stage cal isnt working yet ( I think mostly as boiling water isnt that hot doesnt help).

Cherenkov11, 2018-08-08:
I saw a video of the T12 soldering station where the calibration it is done by three point, for each tip, I do not know if it is applicable to the TS100, due to some limitations on the hardware.

[3 Point Calibration](https://youtu.be/zCKr8ese4mY?t=409)

JohnEdwa, 2018-08-08:
@Ralim I didn't manage to test them all yet, but as we are on exactly the opposite timezones I might as well link the BC2 and C1 I did so far before going to bed. 
https://docs.google.com/spreadsheets/d/1CJ1jZ5H-96_IjHNsdhNipEQ9NROWWIWTls93KeOXh8Y/edit?usp=sharing   

The C1 does feel suspiciously low though, but, it is linear, and it did start melting the solder just before the 200C mark like all the others (60/40 melting point is 188C), so it might be true.
If you do add that curve to the firmware and someone configs the iron for a C1, but puts a B2 in it and cranks that to 450C, it would heat it up to around 560C.

Repled, 2018-08-09:
@Ralim 
> I would not modify the hardware for two reasons; Safety and consistancy.
> Having two firmware builds for drastically different hardware is a pain. The cap (C10) actually provides a means of safety, where a stuck pin state will not cause heating.

Yes, as I said, probably one step to far :o) and yes I fully understand the reason for C10.

If doing two point, I would personally prefer a well defined solder, though the ideal would be something
higher up in the 300-400 degC range.

> To debug: Solder the SWDIO & SWCLK wires to the small mcu board inside the iron :)
Simple and effective ! Thank you.

If I get some time and is back home again I will see if I can contribute with some tip measurements.

Ralim, 2018-08-10:
@JohnEdwa 
I've build a slightly nicer firmware that is much, much easier to read (attached)
The hardware will cap out at a maximum of somewhere between 430 - 455 C depending on the tip curve, so nothing will detonate but cal will be as bad as it is now with the wrong tip :P 

If you could also record the H: reading for each time (handle temp) it would save me an extra step too!
[TS100A.zip](https://github.com/Ralim/ts100/files/2277578/TS100A.zip)

@Repled 
Extra readings wont hurt :) Going to basically try and find average curve fits for each tip type.
This will be faster to push out than support for a dual point calibration, but i'm slowly poking that along too.

A 2 point calibration with as high of a temp separation as possible would be ideal, since it improves accuracy. The dream would be to have two styles of calibration, one where you can use ambient + 100C water. And another where it heats up to what it thinks 300C is, and you can trim its gain until it lines up.

But more time to implement :(

Ralim, 2018-08-13:
Hi All,

Pushing up here a new build, It's not a finished build, but it has new gain values from @JohnEdwa 's numbers.

Added to the advanced menu is a tip type selection.
Note that the "Custom Tip" is not functional yet (needs next set of calibrations).
I highly recommend you also do the normal calibration routine on your handle the first time, this makes a large difference in the accuracy.
[TS100A.zip](https://github.com/Ralim/ts100/files/2282529/TS100A.zip)

Build coming when i get time to finish the coding for a more advanced calibration option.
This will be performed using a two point reference to derive the gain value for your tip.
This will have two options, one using ambient and hot water, and another using a tip temp measurement tool.

Repled, 2018-08-15:
I have not had the time yet to measure the tips but I have been experimenting with a themocouple made out of really thin wire, approx 0.07 mm. This seems to give much better, more accurate results that settles more or less at once.  Will hopefully have some time to measure tomorrow.

Ralim, 2018-08-16:
@Repled Thank you,
If you could use the above firmware in my last post, would be nice to see how close it tracks for some of the tips as well. :)

Repled, 2018-08-16:
It's not easy to measure Tip temperatures ! The wires in the very fine Thermocouple that found are not really good to solder on, they don't really go well with the solder and they got full of solder waste and oxides and more or less stopped working at higher temps due to that. A fix, squeezing on a very thin piece of nickel sheet metal (1.5x1.5mm) at the top of the Thermocouple made a huge difference.

Here is a preliminary peek on my findings so far. Will upload the data tomorrow when I hopefully have made some more measurements.

[image]

Repled, 2018-08-17:
Hi All,

Here are the complete measurements. These values are 'Raw-values' so no calibration is added. In the graph a Linear Approximation has been added. This is used in the second graph to remove the linear part and magnify the differences.

>Graph 1
[image]

In graph 2 below the Y-axis scale is approximately 15 to 16 degC per 1000 units. 
* As can be seen for all curves but two at 350 degC is that they are well within 15 degC from each other, uncalibrated ! At 400 degC some of tips diverge significantly.
* There are two TS-I tips and they are the most linear of them all with just an offset between them. From this the question arises if this is a pure coincidence or if each type of tip has a typical 'curve'. Why is that then? 
* The 2 'thinnest' tips TS-C1 and TS-C4 have the largest offset.

>Graph 2
[image]

Graph 3. These where the initial measurements with the latest FW 2.04 with only room temperature calibration. This Calibration does a fairly good job at getting it right at around 200 degC.

>Graph 3
[image]

Attached are the measurements as a .ods-file (Open Document Format Spreadsheet). Please, let me know if you want some other format.

[TS100_Tip_Measurements_2018-08-17.zip](https://github.com/Ralim/ts100/files/2297276/TS100_Tip_Measurements_2018-08-17.zip)

Repled, 2018-09-04:
Good to see some WIP  even if slowly :o)

Ralim, 2018-09-04:
It's getting there.
The hot water cal appears to be working the few times I have tested it. Haven't implemented and tested the two point hot cal though (advanced cal)

Repled, 2018-09-12:
Will try to get some time over the weekend doing some measurements.

dhiltonp, 2018-09-12:
Excellent!

One thing to play with on that firmware - if you use the advanced display, you can see the watts going into the tip. The temp hits its target pretty fast, then the wattage slowly drops to a base wattage to maintain that temp.

That slow decay must be due to the tip slowly achieving thermal equilibrium.  I look forward to your observations!

dhiltonp, 2018-09-13:
There is a thermal gradient along the tip at all times. The base dissipates heat slowest, having the smallest surface area/volume, and the tip has the most area/volume. That gradient may not be linear, but it'll be good to have temp data from when the watts have stabilized.

Hitting equilibrium fastest probably requires overshoot, avoiding oscillation is going to be the annoyance...

Repled, 2018-09-15:
Below are the latest measurements with observations.  This is starting to look quite good. I could live with these results. The TS-ILS was off by 50 C before at 420 C, now It's more like 12 C excluding the offset.

- TC = ThermoCouple
- All Tips Factory Reset before usage and after that calibrated att Room Temp. Though I get a feeling one calibration for all is enough.
- More accurate calibration ? TS-I Handle Temp went from 32degC to 21 degC (TC measured 21.5). There was a larger offset before between handle temp and actual tip temp after calibration.
- When cooling down the Tip temp on the display and the Thermocouple temp tracks well.
- The regulated temp is almost always showing 1 degC below set temp (Minor ;o))
- The Soldering iron reaches its temperature in seconds, when jumping 10 degC steps, but the Tip TC takes 15-30 seconds to stabilize.
- Done tests adding removing TC from Tip to get an idea of the response time of the TC. It is very fast, - within a second.
- Oscillations, almost non existent, within the last degC.
- As for the Thermal Gradient (see data in attached document) the hottest part with the largest area is where the heating element is, the part above the Tip. The tip shape does not seem to be important for the wattage.
- The shape of the smaller longer Tips like the TS-ILS will make it harder to get the heat to where it should be. In a perfect world the TC should be at the outermost part of the Tip and not part of the heating element.
- Why does the TS-KU track really well to 350 C and after that the temp diverges ??

Sorry, missed to check the C4 tip... might do that later.

[image]

[TS100_Tip_Measurements_2018-09-15.zip](https://github.com/Ralim/ts100/files/2385790/TS100_Tip_Measurements_2018-09-15.zip)

P.S. I have a really hard time loading the FW. Does not work at all with Win10. Seems to work well with an old Win7-computer but out of the two TS100, one is working and the other one looks to work, but nothing changes, still the same FW afterwards.

dhiltonp, 2018-09-15:
That's great!

TS-ILS went from being almost 30*C off of a linear fit at max to being 5*C off, @Ralim's 2 point calibration plus the improved PID will really improve performance.

Now to hit equilibrium faster...

Repled, 2018-09-15:
Equilibrium: I don't see anything wrong in overshooting a bit. Let's say you set it for 350 *C but you target 360 *C or higher and after going past 350 *C you set the final  target of 350 *C. For sure, with the right parameters for the PID you can make it overshoot, but how repeatable will it be for all the different tips ? Just an idea....

Let me know if you want me to confirm any new ideas, developments !

Berre1959, 2018-10-29:
I noticed that callibration can only be done reasonable accurate when tip and handle are at equal (room) temperature.  Holding the handle in (warm) hands for a couple of minutes raises the temperature of the handle internally up to 32°C, which may differ from the tip by more than 15 °C in a cool environment.  Prior to calibration one should not manipulate the handle with warm hands for at least 15 minutes.

Eldenroot, 2019-04-22:
When? :)

Maartenwut, 2019-06-30:
Just calibrated my TS-I using the simple calibration method, wondering if it could be added in the firmware list of tips? What information would you need?

whitehoose, 2019-06-30:
Reading this thread shows a great deal of ingenuity and lateral thinking.  I cant help thinking however that it's being wasted on a tiny molehill.
I've been soldering since I was 8 years old (now 63).  Copper slug with handle on a stove (we had it hard in them days!) 50v and 240v elements heating various weights of copper (the only regulation was in good hardware design), 3v NiCad portable.  And an assortment of gas apparatus in various configurations.  Its only in the last couple of years I got a Solder re-work station which happened to have temperature regulation/display.  1 iron now fits all (most) requirements.

In the past I'd look at a job and select the appropriate sized iron based on the area I intended to cover in 60/40 or 63/37.  can't say I've ever looked at a job and thought that that needs 326.87 deg c.  My internal conversation was more like "That's a hefty lump - better go for the big 'un".  Or, "better be careful and use the mini".

With 63/37 (eutectic) the melting point is pretty certain, but with reputable brands - the datasheet states the melting point.  The aim is to transfer solder to a surface and raise it's temperature sufficiently to fuse the alloy to the to the surface at a molecular level.  If you don't achieve that bond - you're going to have problems.

The devil in the detail is in danger of eclipsing the real aim.  With temp control I just want consistency, when I set my ideal - I want to be certain that that tip will always be where I set it.  The thermostat provides the thermal muscle rather than tip mass. 
This magic happens at a set temperature - but just how or how long it takes to achieve varies quite considerably depending on physical and environmental circumstances.  That's where I come in. 

If you know how - knowing the actual tip temp +/- 0.5C is I'll admit nice to know - especially for us mildly aspergers / obsessive compulsive engineer types, but knowing doesn't actually affect the quality of the job.
In fact if you blindly assume the join is good based only on temperature - you are in for a world of grief.

You can see the solder break and melt, that moment when the solder fuses to the part is unmistakable.  doesn't matter what the temperature says (even if it's spot on).  If its not taken, or if it won't take because of impurities or the insulation or casing is dripping onto the floor something isn't right.
If a gourmet cook can settle for a 1-10 (or 11) heat scale, experience and/or a separate instrument (meat, sugar thermometer) fills in the gaps and confirms the concept.  The rest is just dressing, if you put too much faith in one detail you'll miss the bigger point.

The firmware does a fair job.  I like the feel better than the stock.
 From a temp perspective I can live with 5 degrees or so, as long as the gradient is consistent I'll fill in the blanks.

jjakob, 2020-02-28:
So I performed the simple room temperature calibration on 2 TS100 irons today. Both were loaded with the latest 2.09.1 firmware. The temperatures of both irons when operating is ~17% high, when set to 300C they both read ~350C. The 2 thermocouples I measured the tip temps with were checked at 0C and 100C and they are both within 1%.

I also noticed a weird behaviour with the tip calibration: after I do the calibration on a iron that is cold (cooled down for >1h) the standby tip temperature reads ~30C, even though the room temp is ~22C. Is the calibration assuming the room temp is 30C?

I also noticed that after leaving the iron plugged in for a few minutes in standby (the tip was cold and wasn't heated), the handle gets noticeably warm. I assume the circuitry is generating the heat (voltage regulator, mcu or display). If the software reads the internal temperature and assumes the tip is at that same temperature during calibration, the calibration will be off, turning the tip gain up higher to match the sensor in the handle.

IMO the most accurate cal would be with the 0C/100C two point method, the second option having manually adjustable coefficients.

Firebie, 2020-02-28:
In the code is used const 24.9 uV/C, but, if TS100 tip is similar to T12 - which has Type C, should be used coeff 21 uV/C.

jjakob, 2020-02-28:
These are the stock B2 tips that came with the iron, from the original packaging. I wonder if the coefficient changed, and what it was before that yielded the more accurate calibration with previous firmwares.

I looked at the current calibration routine, and it just saves the raw measured (filtered) value to CalibrationOffset https://github.com/Ralim/ts100/blob/44e5ceeedfc8ceaedf4c374d0e1fe44e6d392e21/workspace/TS100/Core/Src/gui.cpp#L638

https://github.com/Ralim/ts100/blob/e725e63b3f86e4aafe5faf62d6be6f0506fd5234/workspace/TS100/Core/Src/TipThermoModel.cpp#L42-L61
This doesn't make sense to me, it takes the measured thermocouple voltage and subtracts the tip offset. But the tip offset was measured at room temp (293K) so the room temp is subtracted here too. The result is the current TC voltage minus TC voltage @293K minus offset (ADC). It doesn't take the cold junction compensation correctly into account too?
Of course a consequence of this is that the resulting corrected voltage can dip below 0 so it has to catch that, which seems weird.

https://github.com/Ralim/ts100/blob/e725e63b3f86e4aafe5faf62d6be6f0506fd5234/workspace/TS100/Core/Src/TipThermoModel.cpp#L82-L88
The resulting voltage is then converted to a temperature by a single factor, which again seems odd. The room temp should be taken into account in one of these steps, ideally subtracted during the offset calibration. Otherwise if the room temp drops below the room temp during calibration, the measured value will be stuck at 0 and the temperature stuck at its minimum too - I doubt this does good for the PID as it then goes open loop? It shouldn't do anything catastrophic since at or below room temp we're blasting the power at 100% anyways, it's just wrong IMO.

jjakob, 2020-02-28:
Reading some thermocouple theory made me realize:
a) the output voltage is dependent on the temperature difference between the hot and cold junctions
b) the cold junction, if it's just a junction to a different metal and not a real thermocouple junction, as is the case here, needs to be compensated by measuring its temperature and subtracting the resulting TC voltage that would occur at that temperature.

This means that as the handle-to-tip contacts temperature change, the offset should change by the same amount too. This is typically done by a separate sensor thermally coupled to the connection.
https://www.maximintegrated.com/en/design/technical-documents/app-notes/4/4026.html

So if I understand this correctly, the current routine does factor out the cold junction temperature with the offset calibration (the room temperature doesn't matter, just that both junctions are at the same temperature), but doesn't account for changes in the cold junction temperature during use with changing iron handle temperature (conducted from the tip). Unless the iron has a temperature sensor inside the handle (maybe even inside the MCU?), this is impossible to solve. Of course we don't really care about the absolute temperature during either calibration or use, just about the temperature difference between the tip TC and the handle-to-tip connection.

A workaround would be to calibrate with the 2- or multi-point method at room temp (cold junction/offset cal) and at operating temp (I don't think 100C would be enough, as the tip temperature affects the cold junction temperature, a higher difference would be better, preferably normal operating temperature). This would calibrate the thermocouple coefficient too. It would still be unoptimal, with the tip temp with a cold handle undershooting and hot hande overshooting.

Firebie, 2020-02-28:
https://hackaday.io/project/94905-hakko-revenge/log/144548-hakko-t12-thermocouple-is-not-type-k/discussion-119449

jjakob, 2020-02-28:
So at the very least, the constant needs to be corrected to 21uV/C to match the type C thermocouple, even if the advanced calibration isn't implemented.
@Ralim

Ralim, 2020-02-28:
My understanding here is that the miniware tips are _not_ type C, at least from my testing of the ~5 tips i have available. I get a similar over-reading if i use Hakko tips, but very close to on point reading when using the miniware tips here. One concern here though is that some of my tips are getting on four years old, so it could be them aging.

The concept behind the current code is to use the handle temperature sensor as cold junction temp (its as thermally bonded to the contacts as it can be - it was designed for this use case).

The concept behind the current code is to just model the signal path itself, so it converts the ADC readings into uV. Then in later code the cold junction temperature is subtracted out to give the relative reading.

Advanced calibration has been implemented twice, two different ways, and it was _less_ accurate than the current method, which is why i pulled it out (got sick of emails about it not working). I'm happy for someone else to write one, but i think the concept of it will never really work unless the person doing it *has* a high accuracy thermocouple anyway, and at that point it should really be just a cold junction calibration + ~350C one.

There is a temp sensor inside the MCU, but its less accurate than the one in the handle (due to calibration).

One thing I am aware of that should be fixed first is the 3.3V calibration should be done, as the current code assumes its 3.3V but it usually never is, and if that is off by X percent, all temps will be off by that same percentage (roughly).

The 24.9uV constant came from my 5 tips, and I'm happy to adjust that to suit the majority of users :)

My thoughts here are that the order to fix would be:

1. Fix 3.3V rail so its measured to compensate for internal inaccuracies
2. Change constant to 21uV
3. Get a varied mix of tips tested
4. Evaluate what constant _should_ be

Firebie, 2020-02-29:
3.3V - it is about this line?
uint32_t vddRailmVX10 = 33000;	//TODO use ADC Vref to calculate this

[13 more replies]